openwrt/packages #6335: luci-app-acme and nginx

lucize, 2018-06-23:
with new LUCI support for nginx one also needs to have luci-app-uhttpd if luci-app-acme is needed, maybe doing a config for nginx also and depend on what webserver is selected ?

Ansuel, 2018-06-23:
think i will have to make a matepackage for nginx...

diizzyy, 2018-07-17:
@Ansuel 
This is fixed now as far as I can tell?

Ansuel, 2018-07-17:
ummm no... we still need to create e metapackage for acme on nginx... 

also i can't understand, acme needs uhttpd to validate the certs or just to autoinstall it? @lucize can you give me some info about it... 

(Create a configuration to use nginx to validate the cert is not so easy i think....)

Ansuel, 2018-07-17:
think you were referring to ariang package?

lucize, 2018-07-18:
right now the dependency is more like a theoretical one, you need to define a folder for key validation so the dependency makes it easy for you to do that but is not automatically done if you have it installed (luci-app-uhttpd) so not really needed. I can't test right now, I'm on holidays

closed 2018-09-20 by lucize