openwrt/packages #6339: unbound: Error when build custom firmware using imagebuilder "unbound.sh: No such file or directory"

zero13th, 2018-06-24:
Maintainer: @EricLuehrsen 
Environment: imagebuilder ar71xx, imagebuilder ramips-mt7620 

Description:
Build custom firmware using imagebuilder:
make image PROFILE="DEVICE_tl-wr1043nd-v1"      PACKAGES="unbound"

there is error line 19: /usr/lib/unbound/unbound.sh: No such file or directory

```
~/openwrt-imagebuilder-18.06.0-rc1-ar71xx-generic.Linux-x86_64$ make image PROFILE="DEVICE_tl-wr1043nd-v1"      PACKAGES="unbound"
Building images for ar71xx - TP-LINK TL-WR1043N/ND v1
Packages: unbound base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ledtrig-usbport kmod-usb2 libc libgcc logd mtd netifd odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch wpad-mini

-------------------- omitted ------------------------------

Configuring ip6tables.
Configuring kmod-usb-ledtrig-usbport.
Configuring ppp-mod-pppoe.

Finalizing root filesystem...
/home/openwrt/openwrt-imagebuilder-18.06.0-rc1-ar71xx-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ar71xx/etc/init.d/unbound: line 19: /usr/lib/unbound/unbound.sh: No such file or directory
./etc/init.d/unbound: line 19: /usr/lib/unbound/unbound.sh: No such file or directory

Building images...

-------------------- omitted ------------------------------

```

EricLuehrsen, 2018-06-24:
I haven't used `imagebuilder` in a long time. I build from source and file is placed in the correct location. Both configured build and in-use package install work. I am not sure how to debug this part of `imagebuilder`.

EricLuehrsen, 2018-06-24:
@hnyman @jow- any hints?

I haven't used imagebuilder in a long time. I build from source and file is placed in the correct location. Both configured build and in-use package install work. I am not sure how to debug this part of imagebuilder.

EricLuehrsen, 2018-06-25:
There are stop and start functions which reuse other content. This makes sourcing or including convenient and from what I see normal. My problem is figuring out where or why image builder trips on this.

EricLuehrsen, 2018-06-26:
Please try PR #6352 and see if this resolves the problem.

zero13th, 2018-06-27:
Hi Eric,

just tested on latest trunk imagebuilder. there is no such error.
Thank you

closed 2018-06-27 by zero13th