openwrt/packages #6361: busybox: conflict betwwen busybox and diffutils, both provide cmp

pieska, 2018-06-26:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

building an image with imagebuilder lede-imagebuilder-18.06.0-rc1-ar71xx-generic.Linux-x86_64.tar.xz fails with the following error:

```
 * check_data_file_clashes: Package busybox wants to install file /home/me/Projekte/LEDE/18.06.0-rc1/openwrt-imagebuilder-18.06.0-rc1-ar71xx-generic.Linux-x86_64/build_dir/target-mips_24kc_musl/root-ar71xx/usr/bin/cmp
	But that file is already provided by package  * diffutils
 * opkg_install_cmd: Cannot install package busybox.
```

sjjgo, 2018-06-27:
Does it build if you unselect:
Base system -> Busybox -> Editors -> cmp

pieska, 2018-06-27:
I do not build from scratch, i tried to build an image with image builder, thus i'm using the prebuilt packages.

diizzyy, 2018-07-05:
It's always a trade off between naming and convenience. For instance, BSD's usually use g(nameofutility) to avoid clashes with the BSD licensed counterparts.

fantom-x, 2019-06-11:
Just hit this issue myself when using the image builder. A workaround is to make sure that busybox is listed before diffutils: “-busybox busybox diffutils” or just “busybox diffutils”.

fantom-x, 2019-06-12:
Should the *base* packages be listed *before* the packages that are specified in *PACKAGES*? The current output is below and *diffutils* is before *busybox*. If the *base* packages are listed first, then my workaround above would not be required.

```
Building images for ipq806x - Netgear Nighthawk X4S R7800
Packages: iftop iptraf-ng ethtool openssh-server dnscrypt-proxy dnscrypt-proxy-resolvers htop nano ccrypt curl wget patch diffutils tree hostapd-utils wpad-openssl ca-certificates luci-ssl-openssl luci-mod-admin-full luci-app-ddns luci-app-statistics collectd-mod-conntrack collectd-mod-cpufreq collectd-mod-entropy collectd-mod-ping collectd-mod-thermal collectd-mod-uptime luci-app-sqm luci-theme-bootstrap luci-theme-material luci-theme-rosy luci-app-bcp38 luci-app-adblock luci-app-banip luci-app-nlbwmon ath10k-firmware-qca9984-ct base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ata-ahci kmod-ata-ahci-platform kmod-ata-core kmod-ath10k-ct kmod-gpio-button-hotplug kmod-ipt-offload kmod-leds-gpio kmod-usb-core kmod-usb-dwc3-of-simple kmod-usb-ledtrig-usbport kmod-usb-ohci kmod-usb-phy-qcom-dwc3 kmod-usb2 kmod-usb3 libc libgcc logd mtd netifd odhcp6c odhcpd-ipv6only opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch urandom-seed urngd

Package list missing or not up-to-date, generating it.

```

yousong, 2019-06-12:
This can be fixed by adding `PKG_ALTERNATIVE` spec to both busybox and diffutils.  For now, installing diffutils after busybox should workaround the issue.  I will try to find time to see if it's viable we special-case busybox in the opkg alternatives code.

liaoya, 2019-06-12:
@fantom-x 
Thanks! I verified and it works!

closed 2019-06-15 by yousong