openwrt/packages #6382: isc-dhcp-server-ipv4: service starts too early and exist because no network is available

justdave, 2018-06-29:
Maintainer: @antonyantony 
Environment: ar71xx - WZR-HP-G300NH - Reboot (17.01.4, r3560-79f57e422d)

Description:

isc-dhcp-server-ipv4 fails at boot because the network isn't available yet when it starts.

```
Thu Jun 28 02:18:15 2018 daemon.err dhcpd: Not configured to listen on any interfaces!
Thu Jun 28 02:18:15 2018 daemon.err dhcpd:
Thu Jun 28 02:18:15 2018 daemon.err dhcpd: If you think you have received this message due to a bug rather
Thu Jun 28 02:18:15 2018 daemon.err dhcpd: than a configuration issue please read the section on submitting
Thu Jun 28 02:18:15 2018 daemon.err dhcpd: bugs on either our web page at www.isc.org or in the README file
Thu Jun 28 02:18:15 2018 daemon.err dhcpd: before submitting a bug.  These pages explain the proper
Thu Jun 28 02:18:15 2018 daemon.err dhcpd: process and the information we find helpful for debugging..
Thu Jun 28 02:18:15 2018 daemon.err dhcpd:
Thu Jun 28 02:18:15 2018 daemon.err dhcpd: exiting.
```

If I log into the router and run `/etc/init.d/dhcpd start` after it boots up, it starts fine.

diizzyy, 2018-06-30:
@pprindeville 
Probably related to this https://github.com/openwrt/openwrt/pull/1072 (see comments)

justdave, 2018-06-30:
Yeah, that seems likely.

There's a START=60 in the init script which based on seeing it in other scripts like like it's supposed to be startup order, but network has START=20 so apparently that's not used for the order so much anymore?

pprindeville, 2018-07-31:
60 seems awfully late.

diizzyy, 2018-07-31:
@pprindeville 
It seems like it takes at least a few secs to get network interfaces up (both switches and separate nics) on several platforms (ar71xx, octeon etc). I think the last comments in 1072 will give you an idea.

pprindeville, 2018-07-31:
@nbd168 @EricLuehrsen @diizzyy @jow- Looking at the comments in https://github.com/openwrt/openwrt/pull/1072, made me wonder if we yet have a well-understood fix for this issue.

pprindeville, 2018-07-31:
> It seems like it takes at least a few secs to get network interfaces up (both switches and separate nics) on several platforms (ar71xx, octeon etc).

And that happens as `S20network`?  So changing the priority to 21 should be enough, right? Because all the services at priority `n` have to complete before `n+1` gets started...

EricLuehrsen, 2018-08-01:
dnsmasq, Unbound, and Adblock use the `boot()` hook to load `service_triggers()` but avoid the bulk of `start_service()` function. They are called at `START=19` and will wait until a few seconds after a procd network trigger. You will certainly want `service_triggers()` to use `/etc/config/dhcp` to only trigger on DHCP interfaces. WAN6 for example can initiate rapid restarts due to netifd issuing ifup like signals to procd for route life timers in IP6 land.

```
boot() {
  BOOT=1
  start "$@"
}

start_service() {
  if [ -n "$BOOT" ] ; then
    return 0
  fi

  ... procd stuff ...
}

service_triggers() {
  local trigger triggers

  ... some uci to fill in $triggers with DHCP / LAN only ...

  PROCD_RELOAD_DELAY=2000
  procd_add_reload_trigger "dhcp"

  for trigger in $triggers ; do
      # due to some netifd/procd interactions with IP6, limit interfaces
      procd_add_reload_interface_trigger "$trigger"
  done
}
```

diizzyy, 2018-08-01:
@pprindeville 
I'm by far no expert, but it seems to work. I know FreeBSD uses a different approach however...

pprindeville, 2018-08-01:
Hi @EricLuehrsen:

I've taken an initial stab as:

https://github.com/openwrt/packages/compare/master...pprindeville:issue/6382

but it seems like I'm still missing a bit, at least the `stop_service()`, but `dnsmasq`'s seems a bit complicated... like the `dnsmask_stop()` function, for instance.  I don't understand when/why there would exist multiple instances, for example.

Can you please walk me through this?

Thanks.

EricLuehrsen, 2018-08-02:
It seems I didn't read all the ISC-dhcp scripts. I clicked on a sample and missed DHCPv4 server. All others don't have and need procd. DHCPv4 was close to a reasonable state. `boot()` hook did not need to change. Looping through DHCP interface triggers seems right. Also you can use `procd_add_reload_interface_trigger` as a shortcut but `PROCD_RELOAD_DELAY=` needs to be changed (optional) first. Delay may need some experiments to see what the server likes. Instances shouldn't apply. dnsmasq doesn't have "views" in DNS so whatever is in cache everyone gets. dnsmasq has tags for DHCP, as a side note. Consider DNS isolation from home LAN/WiFi and guest WiFi.

EricLuehrsen, 2018-08-03:
Some changes to Unbound may also help here. See PR #6665.

pprindeville, 2018-08-07:
@EricLuehrsen Can you please markup PR #6709?  That might be clearer than reading your description of changes...

pprindeville, 2018-09-02:
@dedeckeh What would the equivalent change look like to parallel FS #1542?

You gated:

```
config_list_foreach "$cfg" "interface" append_interface
```

based on `$BOOT=` but I don't have an equivalent iterator.

neheb, 2020-01-17:
https://github.com/pprindeville/packages/commit/bffd2aecb6667d3c1231c29a084d970ffc69e3a0 was merged. Closing.

closed 2020-01-17 by neheb