openwrt/packages #6398: ppp-mod-pptp for the pptp is not working in version 18.06.0-rc1

thankeyes, 2018-07-04:
The version 18.06.0-rc1 which I install is from here.
https://downloads.openwrt.org/releases/18.06.0-rc1/targets/ramips/mt7620/openwrt-18.06.0-rc1-ramips-mt7620-miwifi-mini-squashfs-sysupgrade.bin

The step what I did is
1.opkg update
2.opkg install ppp-mod-pptp
3.opkg install kmod-nf-nathelper-extra
4.reboot
5.setup the PPTP in the LUCI
##################################
the erro log
##################################
Using interface pptp-vpn
Connect: pptp-vpn <--> pptp (66.212.x.x)
read returned zero, peer has closed
read returned zero, peer has closed
LCP: timeout sending Config-Requests
Connection terminated.
Modem hangup
Using interface pptp-vpn
Connect: pptp-vpn <--> pptp (66.212.x.x)
##############################
And all the PC in the LAN can not client to the PPTP server. 

If I install the version 17.01.4
https://downloads.openwrt.org/releases/17.01.4/targets/ramips/mt7620/lede-17.01.4-ramips-mt7620-miwifi-mini-squashfs-sysupgrade.bin

And follow the step that I wrote upstair,the PPTP is working fine.
And all the PC in the LAN can client to the PPTP server.

thankeyes, 2018-07-05:
@diizzyy 
Yes,it is still broken. I made a snapshots with ppp-mod-pptp and kmod-nf-nathelper-extra.
OpenWrt SNAPSHOT r7402-0f54388 / LuCI Master (git-18.185.66236-26d90f0)
The PPTP client still can not work and all the PC in the LAN can not client to the PPTP server.

flfq, 2018-07-06:
the new SNAPSHOT not work
the 18.06.0-rc1 not work 

The PPTP client still can not work and all the PC in the LAN can not client to the PPTP server.

diizzyy, 2018-07-06:
Create a bug report over at https://bugs.openwrt.org and provide as much information as possible including that you've tested both 18.X and trunk.

yousong, 2018-07-06:
As suggested by @diizzyy , these packages are part of the base system and issue should be opened in the bugs.openwrt.org.  Please also note that it's unlikely that people will go to the length of setting up all the test environment to debug this issue for you, you may need to provide more detailed and specific information to narrow things down.

closed 2018-07-06 by yousong