openwrt/packages #6409: lxc-checkconfig: Checks for non existing modules in openwrt-18.06

lukasmrtvy, 2018-07-07:
lxc-checkconfig in openwrt-18.06 is checking for modules 8021q,bridge,nf_tables. But these are not available as modules. 

It seems that these are removed from 3.18 kernel https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=7096ed58fdc908144d2c52bf416e9b3b0e48795a

Can anyone update lxc-checkconfig package?

lukasmrtvy, 2018-07-09:
@lperkov ping to package maintainer

zxdavb, 2018-10-13:
FWIW: These are the changes I make to **lxc-checkconfig** to fix various issues that are resolved in a later release than that currently used by OpenWrt:
```
    sed -i '/SETCOLOR_FAILURE/   s/\"\$SET/\$SET/'  /usr/bin/lxc-checkconfig
    sed -i '/SETCOLOR_FAILURE/   s/ILURE\"/ILURE/'  /usr/bin/lxc-checkconfig
    sed -i '/xt_CHECKSUM$/a echo'                   /usr/bin/lxc-checkconfig
    sed -i '/E_INSTANCES$/a echo'                   /usr/bin/lxc-checkconfig
    sed -i '/ROUP_NS yes$/a echo'                   /usr/bin/lxc-checkconfig
    sed -i '/CAPABILITIES/  s/ES )/ES; echo \)/'    /usr/bin/lxc-checkconfig
```

neheb, 2020-02-09:
@zxdavb can you make a PR?

zxdavb, 2020-02-09:
no, sorry - I have too much git on my plate as it is

cotequeiroz, 2021-03-18:
18.06 is EOL now, so I'm closing this.

closed 2021-03-18 by cotequeiroz