openwrt/packages #6410: usbip: remove obsolete patches, adapt package to new kernel/libudev

damnms, 2018-07-07:
I still can not build openwrt 18.06

related: https://github.com/openwrt/packages/pull/6062

```
arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/tcp_wrappers_7.6:tcp_wrappers_7.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -DFACILITY=LOG_DAEMON -DHOSTS_ACCESS      -DDAEMON_UMASK=022 -DREAL_DAEMON_DIR=\"/usr/sbin\" -DPROCESS_OPTIONS -DKILL_IP_OPTIONS -DSEVERITY=LOG_INFO -DRFC931_TIMEOUT=10  -DHOSTS_DENY=\"/etc/hosts.deny\" -DHOSTS_ALLOW=\"/etc/hosts.allow\"   -DSYS_ERRLIST_DEFINED -DHAVE_STRERROR -DHAVE_WEAKSYMS -D_REENTRANT -DINET6=1 -Dss_family=__ss_family -Dss_len=__ss_len    -o tcpdchk tcpdchk.o fakelog.o inetcf.o scaffold.o -Lshared -lwrap
arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/tcp_wrappers_7.6:tcp_wrappers_7.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -DFACILITY=LOG_DAEMON -DHOSTS_ACCESS      -DDAEMON_UMASK=022 -DREAL_DAEMON_DIR=\"/usr/sbin\" -DPROCESS_OPTIONS -DKILL_IP_OPTIONS -DSEVERITY=LOG_INFO -DRFC931_TIMEOUT=10  -DHOSTS_DENY=\"/etc/hosts.deny\" -DHOSTS_ALLOW=\"/etc/hosts.allow\"   -DSYS_ERRLIST_DEFINED -DHAVE_STRERROR -DHAVE_WEAKSYMS -D_REENTRANT -DINET6=1 -Dss_family=__ss_family -Dss_len=__ss_len    -o hosts_access.o -c hosts_access.c
arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/tcp_wrappers_7.6:tcp_wrappers_7.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -DFACILITY=LOG_DAEMON -DHOSTS_ACCESS      -DDAEMON_UMASK=022 -DREAL_DAEMON_DIR=\"/usr/sbin\" -DPROCESS_OPTIONS -DKILL_IP_OPTIONS -DSEVERITY=LOG_INFO -DRFC931_TIMEOUT=10  -DHOSTS_DENY=\"/etc/hosts.deny\" -DHOSTS_ALLOW=\"/etc/hosts.allow\"   -DSYS_ERRLIST_DEFINED -DHAVE_STRERROR -DHAVE_WEAKSYMS -D_REENTRANT -DINET6=1 -Dss_family=__ss_family -Dss_len=__ss_len    -o options.o -c options.c
options.c: In function 'banners_option':
options.c:230:6: warning: implicit declaration of function 'write'; did you mean 'fwrite'? [-Wimplicit-function-declaration]
      write(request->fd, "", 1);
      ^~~~~
      fwrite
options.c: In function 'group_option':
options.c:259:25: warning: implicit declaration of function 'setgid'; did you mean 'setbit'? [-Wimplicit-function-declaration]
     if (dry_run == 0 && setgid(grp->gr_gid))
                         ^~~~~~
                         setbit
options.c: In function 'user_option':
options.c:281:25: warning: implicit declaration of function 'setuid'; did you mean 'setbit'? [-Wimplicit-function-declaration]
     if (dry_run == 0 && setuid(pwd->pw_uid))
                         ^~~~~~
                         setbit
options.c: In function 'nice_option':
options.c:363:25: warning: implicit declaration of function 'nice' [-Wimplicit-function-declaration]
     if (dry_run == 0 && nice(niceval) < 0)
                         ^~~~
options.c: In function 'twist_option':
options.c:385:52: warning: implicit declaration of function 'close'; did you mean 'pclose'? [-Wimplicit-function-declaration]
 #define maybe_dup2(from, to) ((from == to) ? to : (close(to), dup(from)))
                                                    ^
options.c:387:6: note: in expansion of macro 'maybe_dup2'
  if (maybe_dup2(request->fd, 0) != 0 ||
      ^~~~~~~~~~
options.c:385:63: warning: implicit declaration of function 'dup'; did you mean 'strdup'? [-Wimplicit-function-declaration]
 #define maybe_dup2(from, to) ((from == to) ? to : (close(to), dup(from)))
                                                               ^
options.c:387:6: note: in expansion of macro 'maybe_dup2'
  if (maybe_dup2(request->fd, 0) != 0 ||
      ^~~~~~~~~~
options.c:394:13: warning: implicit declaration of function 'execl' [-Wimplicit-function-declaration]
      (void) execl("/bin/sh", "sh", "-c", value, (char *) 0);
             ^~~~~
options.c:394:13: warning: incompatible implicit declaration of built-in function 'execl'
options.c: In function 'setenv_option':
options.c:434:9: warning: implicit declaration of function 'setenv'; did you mean 'setjmp'? [-Wimplicit-function-declaration]
     if (setenv(chop_string(value), chop_string(var_value), 1))
         ^~~~~~
         setjmp
arm-openwrt-linux-muslgnueabi-gcc -Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/tcp_wrappers_7.6:tcp_wrappers_7.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -DFACILITY=LOG_DAEMON -DHOSTS_ACCESS      -DDAEMON_UMASK=022 -DREAL_DAEMON_DIR=\"/usr/sbin\" -DPROCESS_OPTIONS -DKILL_IP_OPTIONS -DSEVERITY=LOG_INFO -DRFC931_TIMEOUT=10  -DHOSTS_DENY=\"/etc/hosts.deny\" -DHOSTS_ALLOW=\"/etc/hosts.allow\"   -DSYS_ERRLIST_DEFINED -DHAVE_STRERROR -DHAVE_WEAKSYMS -D_REENTRANT -DINET6=1 -Dss_family=__ss_family -Dss_len=__ss_len    -o shell_cmd.o -c shell_cmd.c
shell_cmd.c: In function 'shell_cmd':
shell_cmd.c:47:25: warning: implicit declaration of function 'fork' [-Wimplicit-function-declaration]
     switch (child_pid = fork()) {
                         ^~~~
shell_cmd.c:55:21: warning: implicit declaration of function 'wait'; did you mean 'exit'? [-Wimplicit-function-declaration]
  while ((wait_pid = wait((int *) 0)) != -1 && wait_pid != child_pid)
                     ^~~~
                     exit
shell_cmd.c: In function 'do_child':
shell_cmd.c:78:9: warning: implicit declaration of function 'close'; did you mean 'pclose'? [-Wimplicit-function-declaration]
  (void) close(tmp_fd);
         ^~~~~
         pclose
shell_cmd.c:79:9: warning: implicit declaration of function 'open'; did you mean 'popen'? [-Wimplicit-function-declaration]
     if (open("/dev/null", 2) != 0) {
         ^~~~
         popen
shell_cmd.c:81:16: warning: implicit declaration of function 'dup'; did you mean 'strdup'? [-Wimplicit-function-declaration]
     } else if (dup(0) != 1 || dup(0) != 2) {
                ^~~
[... 118 lines cut ...]
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  577k  100  577k    0     0   712k      0 --:--:-- --:--:-- --:--:--  711k
touch /home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/usbip-2.0/.prepared_e4e36e6a025aea34eefcbba7663a83e5_6664517399ebbbc92a37c5bb081b5c53_check
rm -rf /home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/usbip-2.0
cp -fpR /home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/linux-4.14.52/tools/usb/usbip /home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/usbip-2.0
true
[ ! -d ./src/ ] || cp -fpR ./src/. /home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/usbip-2.0

Applying ./patches-2.0/100-musl-compat.patch using plaintext: 
patching file src/usbipd.c
Hunk #1 FAILED at 453.
1 out of 1 hunk FAILED -- saving rejects to file src/usbipd.c.rej
Patch failed!  Please fix ./patches-2.0/100-musl-compat.patch!
make[3]: *** [Makefile:117: /home/oli/lede/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/linux-mvebu_cortexa9/usbip-2.0/.prepared_e4e36e6a025aea34eefcbba7663a83e5_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/oli/lede/feeds/packages/net/usbip'
Command exited with non-zero status 2
time: package/feeds/packages/usbip/compile#0.09#0.06#0.94
make[2]: *** [package/Makefile:109: package/feeds/packages/usbip/compile] Error 2
make[2]: Leaving directory '/home/oli/lede'
make[1]: *** [package/Makefile:103: /home/oli/lede/staging_dir/target-arm_cortex-a9+vfpv3_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/oli/lede'
make: *** [/home/oli/lede/include/toplevel.mk:218: world] Fehler 2
[oli@lucy lede]$
```

arrmo, 2018-07-15:
Hi,

I am also seeing similar issues, building for a mips74k device (WNDR3700v3) => building packages fails, also at usbip.

Any workaround?

Thanks!

damnms, 2018-07-15:
i guess the workaround is to delete the patch files, but for me this is not a usable environment when i have to delete stuff - feels somehow alpha stage, not really release candidate

damnms, 2018-08-01:
Thats very bad. 18.06 is released but its still not compileable because of that bug. Is no one responsible for that?!

dibdot, 2018-08-17:
Seems to be fixed - close this by now.

closed 2018-08-17 by dibdot