openwrt/packages #6446: lxc-create: Failed to download rootfs.tar.xz

alex9434, 2018-07-09:
Environment: amd64, OpenWRT 18.06-RC1
Description:
Creation of LXC container fails:
`lxc-create -t download -n u1 -- --dist ubuntu --release bionic --arch amd64`

`Setting up the GPG keyring
Downloading the image index
Downloading the rootfs
ERROR: Failed to download http://images.linuxcontainers.org//images/ubuntu/bionic/amd64/default/20180709_07:42//rootfs.tar.xz
lxc-create: u1: lxccontainer.c: create_run_template: 1473 container creation template for u1 failed
lxc-create: u1: tools/lxc_create.c: main: 329 Error creating container u1`

dibdot, 2018-07-09:
Works for me:

```
root@blackhole:~# lxc-create -t download -n u1 -- --dist ubuntu --release bionic --arch amd64 --flush-cache
Flushing the cache...
Setting up the GPG keyring
Downloading the image index
Downloading the rootfs
Downloading the metadata
The image cache is now ready
Unpacking the rootfs

---
You just created an Ubuntu bionic amd64 (20180709_07:42) container.
```
Probably your setup is not complete, check that you've installed the following packages: 'kernel cgroups', 'kernel namespaces' and 'miscellaneous LXC related options' plus 'kmod-veth' for optional network support ... for SSL support you need 'wget', 'ca-certificates', 'gnupg' and 'gnupg-utils' as well.

closed 2018-07-09 by dibdot

dibdot, 2018-07-10:
Please post the output of:
```
wget -T 30 -q http://images.linuxcontainers.org//images/ubuntu/bionic/amd64/default/20180709_07:42//rootfs.tar.xz -O /tmp/rootfs.tar.xz;echo $?

and opkg list-installed
```

Still, this is a configuration issue and not something we could fix within the package. Therefore it would be much better to ask in the forum (https://forum.lede-project.org/) for this ...

alex9434, 2018-07-10:
'root@OpenWrt:~# wget -T 30 -q http://images.linuxcontainers.org//images/ubuntu/b
ionic/amd64/default/20180709_07:42//rootfs.tar.xz -O /tmp/rootfs.tar.xz;echo $?
3
root@OpenWrt:~# opkg list-installed
ar - 2.27-1
base-files - 190-r0-5745bbe
bash - 4.4.18-2
busybox - 1.28.3-4
bzip2 - 1.0.6-3
ca-certificates - 20180409
coreutils - 8.23-3
coreutils-chroot - 8.23-3
coreutils-sha1sum - 8.23-3
curl - 7.60.0-1
debootstrap - 1.0.92-1
dnsmasq - 2.80test2-3
dropbear - 2017.75-5
e2fsprogs - 1.44.1-1
firewall - 2018-05-14-b45e162e-1
flock - 2.32-2
fstools - 2018-04-16-e2436836-1
fwtool - 1
getopt - 2.32-2
gnupg - 1.4.21-1
gnupg-utils - 1.4.21-1
gpgv - 1.4.21-1
ip6tables - 1.6.2-1
iptables - 1.6.2-1
jshn - 2018-04-12-6eff829d-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 4.14.52-1-4358fdf549c21b2571cb1146df13392a
kmod-button-hotplug - 4.14.52-3
kmod-e1000 - 4.14.52-1
kmod-e1000e - 4.14.52-1
kmod-hwmon-core - 4.14.52-1
kmod-i2c-algo-bit - 4.14.52-1
kmod-i2c-core - 4.14.52-1
kmod-igb - 4.14.52-1
kmod-input-core - 4.14.52-1
kmod-ip6tables - 4.14.52-1
kmod-ipt-conntrack - 4.14.52-1
kmod-ipt-core - 4.14.52-1
kmod-ipt-nat - 4.14.52-1
kmod-ipt-offload - 4.14.52-1
kmod-lib-crc-ccitt - 4.14.52-1
kmod-mii - 4.14.52-1
kmod-nf-conntrack - 4.14.52-1
kmod-nf-conntrack6 - 4.14.52-1
kmod-nf-flow - 4.14.52-1
kmod-nf-ipt - 4.14.52-1
kmod-nf-ipt6 - 4.14.52-1
kmod-nf-nat - 4.14.52-1
kmod-nf-reject - 4.14.52-1
kmod-nf-reject6 - 4.14.52-1
kmod-ppp - 4.14.52-1
kmod-pppoe - 4.14.52-1
kmod-pppox - 4.14.52-1
kmod-pps - 4.14.52-1
kmod-ptp - 4.14.52-1
kmod-r8169 - 4.14.52-1
kmod-slhc - 4.14.52-1
libbfd - 2.27-1
libblkid - 2.32-2
libblobmsg-json - 2018-04-12-6eff829d-1
libbz2 - 1.0.6-3
libc - 1.1.19-1
libcap - 2.25-2
libcomerr - 1.44.1-1
libcurl - 7.60.0-1
libext2fs - 1.44.1-1
libf2fs - 1.10.0-1
libgcc - 7.3.0-1
libip4tc - 1.6.2-1
libip6tc - 1.6.2-1
libiwinfo - 2018-02-15-223e09bf-1
libiwinfo-lua - 2018-02-15-223e09bf-1
libjson-c - 0.12.1-1
libjson-script - 2018-04-12-6eff829d-1
liblua - 5.1.5-1
liblucihttp - 2018-04-22-ccc685e5-1
liblucihttp-lua - 2018-04-22-ccc685e5-1
liblxc - 2.1.1-1
liblzma - 5.2.4-1
libmbedtls - 2.8.0-1
libncurses - 6.1-1
libnl-tiny - 0.1-5
libopenssl - 1.0.2o-1
libpcre - 8.41-2
libpthread - 1.1.19-1
libreadline - 7.0-1
librt - 1.1.19-1
libsmartcols - 2.32-2
libss - 1.44.1-1
libubox - 2018-04-12-6eff829d-1
libubus - 2018-01-16-5bae22eb-1
libubus-lua - 2018-01-16-5bae22eb-1
libuci - 2018-03-24-5d2bf09e-1
libuclient - 2017-11-02-4b87d831-1
libuuid - 2.32-2
libxtables - 1.6.2-1
logd - 2018-02-14-128bc35f-2
lua - 5.1.5-1
luci - git-18.180.55209-4ba85e3-1
luci-app-firewall - git-18.180.55209-4ba85e3-1
luci-app-lxc - git-18.180.55209-4ba85e3-1
luci-base - git-18.180.55209-4ba85e3-1
luci-lib-ip - git-18.180.55209-4ba85e3-1
luci-lib-jsonc - git-18.180.55209-4ba85e3-1
luci-lib-nixio - git-18.180.55209-4ba85e3-1
luci-mod-admin-full - git-18.180.55209-4ba85e3-1
luci-proto-ipv6 - git-18.180.55209-4ba85e3-1
luci-proto-ppp - git-18.180.55209-4ba85e3-1
luci-theme-bootstrap - git-18.180.55209-4ba85e3-1
lxc - 2.1.1-1
lxc-attach - 2.1.1-1
lxc-autostart - 2.1.1-1
lxc-cgroup - 2.1.1-1
lxc-checkconfig - 2.1.1-1
lxc-common - 2.1.1-1
lxc-config - 2.1.1-1
lxc-configs - 2.1.1-1
lxc-console - 2.1.1-1
lxc-copy - 2.1.1-1
lxc-create - 2.1.1-1
lxc-destroy - 2.1.1-1
lxc-device - 2.1.1-1
lxc-execute - 2.1.1-1
lxc-freeze - 2.1.1-1
lxc-hooks - 2.1.1-1
lxc-info - 2.1.1-1
lxc-init - 2.1.1-1
lxc-ls - 2.1.1-1
lxc-lua - 2.1.1-1
lxc-monitor - 2.1.1-1
lxc-monitord - 2.1.1-1
lxc-snapshot - 2.1.1-1
lxc-start - 2.1.1-1
lxc-stop - 2.1.1-1
lxc-templates - 2.1.1-1
lxc-top - 2.1.1-1
lxc-unfreeze - 2.1.1-1
lxc-unprivileged - 2.1.1-1
lxc-unshare - 2.1.1-1
lxc-user-nic - 2.1.1-1
lxc-usernsexec - 2.1.1-1
lxc-wait - 2.1.1-1
mkf2fs - 1.10.0-1
mtd - 21
netifd - 2018-05-30-a580028d-1
odhcp6c - 2018-04-29-5316d7f8-12
odhcpd-ipv6only - 1.7-1
openwrt-keyring - 2018-05-18-103a32e9-1
opkg - 2017-12-07-3b417b9f-2
partx-utils - 2.32-2
ppp - 2.4.7-12
ppp-mod-pppoe - 2.4.7-12
procd - 2018-03-28-dfb68f85-1
r8169-firmware - 2017-09-06-a61ac5cf-1
rpcd - 2018-05-13-82062195-1
rpcd-mod-lxc - 20171206
rpcd-mod-rrdns - 20170710
shadow-common - 4.2.1-8
shadow-newgidmap - 4.2.1-8
shadow-newuidmap - 4.2.1-8
shadow-utils - 4.2.1-8
tar - 1.30-1
terminfo - 6.1-1
ubox - 2018-02-14-128bc35f-2
ubus - 2018-01-16-5bae22eb-1
ubusd - 2018-01-16-5bae22eb-1
uci - 2018-03-24-5d2bf09e-1
uclient-fetch - 2017-11-02-4b87d831-1
uhttpd - 2017-11-04-a235636a-1
uhttpd-mod-ubus - 2017-11-04-a235636a-1
usign - 2015-07-04-ef641914-1
wget - 1.19.5-1
xz - 5.2.4-1
xz-utils - 5.2.4-1
zlib - 1.2.11-2'

dibdot, 2018-07-10:
Thanks. Root cause is the file i/o error you get in wget (return code 3). Now repeat the slightly modified wget command to see more ...

```
wget -T 30 http://images.linuxcontainers.org//images/ubuntu/bionic/amd64/default/20180709_07:42//rootfs.tar.xz -O /tmp/rootfs.tar.xz
```

alex9434, 2018-07-11:
root@OpenWrt:~# wget -T 30 http://images.linuxcontainers.org//images/ubuntu/bion
ic/amd64/default/20180709_07:42//rootfs.tar.xz -O /tmp/rootfs.tar.xz
--2018-07-11 20:05:51--  http://images.linuxcontainers.org//images/ubuntu/bionic/amd64/default/20180709_07:42//rootfs.tar.xz
Resolving images.linuxcontainers.org... 91.189.91.21, 91.189.88.37, 2001:67c:1560:8001::21, ...
Connecting to images.linuxcontainers.org|91.189.91.21|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://uk.images.linuxcontainers.org/images/ubuntu/bionic/amd64/default/20180709_07:42//rootfs.tar.xz [following]
--2018-07-11 20:05:51--  http://uk.images.linuxcontainers.org/images/ubuntu/bionic/amd64/default/20180709_07:42//rootfs.tar.xz
Resolving uk.images.linuxcontainers.org... 91.189.88.37, 2001:67c:1560:8001::21
Connecting to uk.images.linuxcontainers.org|91.189.88.37|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 83714996 (80M) [application/x-tar]
Saving to: '/tmp/rootfs.tar.xz'

/tmp/rootfs.tar.xz  100%[===================>]  79.84M  5.75MB/s    in 14s     

2018-07-11 20:06:06 (5.64 MB/s) - '/tmp/rootfs.tar.xz' saved [83714996/83714996]

dibdot, 2018-07-12:
Great! Thanks for your feedback.