openwrt/packages #6457: net-snmp: Extend syntax / usage

arrmo, 2018-07-11:
Maintainer: @dedeckeh 
Environment: WNDR3700v3, OpenWrt 18.06.0-rc1, r7090-d2aa3a1b62

Description:
Similar to Issue #4085, trying to use extend functionality of net-snmp. The package is "new enough" (5.7.3-9), but can't get extend to work. It may be the syntax I'm trying to use in /etc/config/snmpd, but any pointers would be appreciated! Trying to effectively add a line like below,

```
extend .1.3.6.1.4.1.2021.7890.1 distro /usr/bin/distro
```

Thanks!

karlp, 2018-07-11:
well, perhaps share the config you were trying?  UCI support for extend was added in  https://github.com/openwrt/packages/commit/a565ef8eaaa27e586b8759743f2b9188b7b3b42f  #4085 is irrelevant for this, unless you're creating your own config files.

arrmo, 2018-07-12:
Sure, NP! I only made a change right near the top of the file, as below (the rest is unchanged). And of course, this matters ... it's in /etc/config/snmpd
```
config agent
        option extend distro /usr/bin/distro
        option agentaddress UDP:161,UDP6:161
```

I have also tried the line noted above (with the mboid added, same result).

Thanks!

karlp, 2018-07-12:
it's not a magic single option.  that's not how UCI works, for anything, anywhere.  You should try copying how "exec" configs work.  I'd imagine from looking at the code, that you want something like
```
config extend
     option miboid blah
     option name blah
     option prog blah
    option args "blah asdf asdf"
```

arrmo, 2018-07-12:
That makes sense, and sorry - I should have mentioned ... I first tried what you explained here. That didn't work either ... :(. I tried to run in debug mode, but it won't really do that ... it runs the daemon instead, with no logs (was hoping the log would tell me what I did wrong in my syntax ... :()

karlp, 2018-07-12:
have a look at the generated config file CONFIGFILE="/var/run/snmpd.conf"
and see what was put in.  name and prog are both required, as you can see from the commit I linked.

arrmo, 2018-07-13:
That's the odd part - nothing is added to the /var/run/snmpd.conf file (or at least, I mean nothing for the extend line). It's like it's not happy with something there, but no log output to see? I checked logread, nothing there - perhaps the wrong place?

Thanks!

karlp, 2018-07-13:
you're still not sharing what you're actually tried.

arrmo, 2018-07-13:
Sorry! Here is what I added. Thougts?

```
config extend
       option miboid   .1.3.6.1.4.1.2021.7890.1
        option name     distro
        option prog     /usr/bin/distro
        option args
```

I have also commented out args, and tried adding ''. No joy in any case ... :-(.

Thanks!

arrmo, 2018-07-14:
OK, some more debugging, and a couple interesting findings,
1) If I manually edit /var/run/snmpd.conf, and then kill / manually run snmpd (not from init.d, but just execute /usr/sbin/snmpd -Lf /dev/null -f) => extend does work! So it seems the issue is generating /var/run/snmpd.conf from /etc/config/snmpd. I even stripped the above down to only name and prog, but it's still not happy.
2) If I enter a value for miboid in /var/run/snmpd.conf, that seems to break extend. But that may be due to the MIB files that are incorporated (vs. the miboid I use).

I can live with 2) above, but 1) is blocking me. Any idea what code processes the UCI file?

Thanks!

arrmo, 2018-07-14:
FYI, also tried to add extend using the commands outlined here (but exec => extend),
https://openwrt.org/docs/guide-user/services/snmp/snmpd

Seems to add the info to /etc/config/snmpd, but it still doesn't show up in /var/run/snmpd.conf. Something seems to be messing up processing of the UCI file / code.

arrmo, 2018-07-14:
OK, I think I found it ... :-). The script /etc/init.d/snmpd doesn't seem to include extend support. OK, just checked the repository - it's in the master, but not 18.06 ... :-(. That's why I don't see it, and why this isn't working. Unfortunately - I can build the master, but it fails to boot on my router (just hangs trying to boot the kernel).

I'm open to suggestions!

karlp, 2018-07-19:
building master or running master is a different problem.  _this_ bug appears to be resolved though.  (you're also free to backport patches yourself if you want to build something based on 18.06 too)

closed 2018-07-19 by karlp