openwrt/packages #6470: Problem with latest PPP commit

ibnbd, 2018-07-14:
Since I pulled the new commit for packages, there were some changes for PPP. I can't compile anymore, every time I try I am getting this error.

```
make[3]: Entering directory '/home/linaro/openwrt/package/network/services/ppp'
mkdir -p /home/linaro/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /home/linaro/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/linaro/openwrt/build_dir/target-mipsel_24kc_musl/ppp-default/ppp-2.4.7/.pkgdir/ppp/. /home/linaro/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /home/linaro/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.ppp_installed
mkdir -p /home/linaro/openwrt/bin/targets/ramips/mt7621/packages /home/linaro/openwrt/build_dir/target-mipsel_24kc_musl/ppp-default/ppp-2.4.7/ipkg-mipsel_24kc/ppp-mod-pppoe/CONTROL /home/linaro/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo
install -d -m0755 /home/linaro/openwrt/build_dir/target-mipsel_24kc_musl/ppp-default/ppp-2.4.7/ipkg-mipsel_24kc/ppp-mod-pppoe/usr/lib/pppd/2.4.7
install -m0755 /home/linaro/openwrt/build_dir/target-mipsel_24kc_musl/ppp-default/ppp-2.4.7/ipkg-install/lib/pppd/2.4.7/rp-pppoe.so /home/linaro/openwrt/build_dir/target-mipsel_24kc_musl/ppp-default/ppp-2.4.7/ipkg-mipsel_24kc/ppp-mod-pppoe/usr/lib/pppd/2.4.7/
install: cannot stat '/home/linaro/openwrt/build_dir/target-mipsel_24kc_musl/ppp-default/ppp-2.4.7/ipkg-install/lib/pppd/2.4.7/rp-pppoe.so': No such file or directory
Makefile:302: recipe for target '/home/linaro/openwrt/bin/packages/mipsel_24kc/base/ppp-mod-pppoe_2.4.7-12_mipsel_24kc.ipk' failed
make[3]: *** [/home/linaro/openwrt/bin/packages/mipsel_24kc/base/ppp-mod-pppoe_2.4.7-12_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/home/linaro/openwrt/package/network/services/ppp'
Command exited with non-zero status 2
time: package/network/services/ppp/default/compile#0.30#0.02#0.66
package/Makefile:107: recipe for target 'package/network/services/ppp/compile' failed
make[2]: *** [package/network/services/ppp/compile] Error 2
make[2]: Leaving directory '/home/linaro/openwrt'
package/Makefile:103: recipe for target '/home/linaro/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/linaro/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/linaro/openwrt'
/home/linaro/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

airend, 2018-07-14:
Seems related to e7397eef69.

ibnbd, 2018-07-14:
@airend I guess too, seems this was the last commit.

airend, 2018-07-14:
I meant to add that reverting the LTO build flags fixes the issue. Still, pretty odd bug.

ibnbd, 2018-07-14:
@airend I modified the `Makefile` and I was able to compile again. Hopefully author will change it to something.

nbd168, 2018-07-14:
Fixed in 888a15ff8312cd157bc968b67c33d2f738883b79

ibnbd, 2018-07-14:
@nbd168 thanks, it worked

closed 2018-07-18 by hnyman