openwrt/packages #6473: Python3-setuptools: request update package from 39.0.1 .

Kirya2k, 2018-07-14:
Maintainer: @commodo 
Environment: Openwrt-18.06 and Master.

Description: Modules pycryptodome and uvloop don't build pip3 with python3-setupttols 39.0.1.
Update package plz.
Thanks a lot.

commodo, 2018-07-15:
Hey,

The version of this package is controlled by the python3 package.
Whenever we update python3 we get the version that Python3 has built-in.

Can you update this package via pip3 ?
Would that work ?

Something like: ‘pip3 install —upgrade setuptools ‘
 It could be a good idea to also upgrade pip3 in a similar fashion, if upgrading setuptools.

Thanks
Alex

Kirya2k, 2018-07-15:
build-in pip3 make update setuptools normally.
But me also have to install the gcc-toolchain, which is inconvenient.
So I ask you to update setuptools in ipk of python-stream Openwrt.
pip3 I haven't tried to update.

With best regards, Kirya.

commodo, 2018-07-16:
i will take a look at a solution for this :)
maybe there is something that works in-between for this

jefferyto, 2019-01-08:
Does #7849 (and #7854) fix this issue?

commodo, 2019-01-09:
I think @Kirya2k needs to be take a look closer.
`python3-setuptools` did upgrade ; hopefully the version that is shipped now is good enough to meet the requiremenets here.
But if a newer version is desired, then this will be a permanent/ongoing issue, as the version of setuptools will always be tied to the version that comes shipped with Python/Python3.

Kirya2k, 2019-01-11:
Yes, version 40.6.2 is enough for build.
Thx a lot !

closed 2019-01-11 by Kirya2k