openwrt/packages #6479: openwrt 18.06 runs in netgear r6220 , shadowsocks-libev cannot work well

kevinqiu0098, 2018-07-15:
Description:
openwrt new install version is 18.06
rourter model :netgear r6220
problem : shadowsocks works failed
[image]
[image]
can someone show me how to troubleshoot with it ? thanks for your help @yousong

kevinqiu0098, 2018-07-15:
[image]
@yousong  i try restart  shadowsocks but got nothing  it looks  didnt install right (btw i install from sofeware )

kevinqiu0098, 2018-07-15:
@yousong 
first thanks for the reply it mean so much for me   
it's strange I can open google.com and get the search result page once(youtube twitter etc. cannot open) but now it looks like being block again 
does dns setting need additional setting ?
I am current live in china I am expecting local website visit directly and the block website just take the ss server

yousong, 2018-07-15:
@kevinmehall , you may want to enable a ss_tunnel instance and point dhcp server to the local endpoint of the tunnel.

kevinqiu0098, 2018-07-15:
sorry I am not quite familiar with openwrt, kindly check below 
[image]
instance won't enable correctly
and the red mark I dont konw what to do with it

yousong, 2018-07-15:
Tunnel address can be `8.8.8.8:53`.  You can read an introduction to how each shadowsocks-libev component works at https://github.com/openwrt/packages/tree/master/net/shadowsocks-libev

kevinqiu0098, 2018-07-15:
ss_tunnel instance still can not running back to the began...
it 's late thanks for your help

yousong, 2018-07-17:
@kevinqiu0098 , any updates on this one?

kevinqiu0098, 2018-07-20:
Nope ...back to beginning   I will figure another  Way thanks anyway

yousong, 2018-07-21:
No problem.  I am closing this one for now.  Feel free to post here in case anything new pops up.

closed 2018-07-21 by yousong