openwrt/packages #6494: tvheadend 4.3 support

erotavlas85, 2018-07-16:
Hi,
is there any plan to create a package with tvheadend 4.3? The current version 4.0 does not support a lot of feature [roadmap](https://tvheadend.org/projects/tvheadend/roadmap).
I found this interesting thread on [tvheadend forum](https://tvheadend.org/boards/4/topics/30019?r=33478#message-33478), it seems that an user was able to compile and create tvheadend package for openwrt. Do you know where to find it or how to compile and create a package?
Thank you

hnyman, 2018-07-18:
There is a proposal for going with 4.2.6, but even that is somewhat problematic.
Read #6386

closed 2018-07-18 by hnyman

benteam10, 2018-08-27:
Compiled for 4.2.6 fine for ar71xx and 4.3 for ar71xx tplink archer c7 V2, and Cortex a9 Linksys WRT3200ACM works like a charm.

https://1drv.ms/u/s!Ar_ONyLrmjE06zzKPkoHGdxQzLRk
https://1drv.ms/u/s!Ar_ONyLrmjE06zvaWyMIpJeh-Y72

if you guys have problems compiling it and making this available let me know i can share you how to cross compile it and avoid all problems

erotavlas85, 2018-08-27:
Hi,
it would be very useful a brief guide about how to cross compile for lantiq xrx200 for bt homehub v5a.
Thank you

benteam10, 2018-08-27:
Cross compile tvheadend for Openwer 18

this guide is based in my knowledge and my own practices doing this, if you know a better way to cross-compile for opwenwrt let me know.

Everything is done in UBUNTU 18 however is the exact same if you do it in UBUNTU 16 or Debian 9.5 .

    1. Install the openwrt buildroot
       https://wiki.openwrt.org/doc/howto/buildroot.exigence
       
    2. Once the buildroot is installed and you run the “make menuconfig” to configure the buildroot make sure you configure correctly the first 3 options of the menu : Target System, Subtarget and Target Profile, for your router if you dont know read openwrt wifi for your router, leave everything else, save changes and exit.
    3. Before you compile the toolchain if you are doing this as root like me set this variable to avoid errors.  
       set FORCE_UNSAFE_CONFIGURE=1
       export set FORCE_UNSAFE_CONFIGURE=1 
    4. compile : make -j4 V=s   ( if anything fails use -j1 to see what the problem is modify -j? To fit your processors cores, read before setting it wrong ) this takes 1 hr aprox to create the toolchain.
    5. Once completed, modify the tvheadend makefile to disable ffmpeg issues and compile ver 4.3.
       FILE location: openwrt/feeds/packages/multimedia/tvheadend/Makefile
       change PKG_VERSION:=4.3 , 
       comment #PKG_MIRROR_HASH:=afe42bcf683e8576ca8ccd4755ea308f79195e92f7c30fc5961973e238f49105
    6. add the 6 last lines to the CONFIGURE_ARGS part of the file.
CONFIGURE_ARGS += \
        --arch=$(ARCH) \
        --disable-dbus_1 \
        --disable-libav \
        --enable-bundle \
        --disable-ffmpeg_static \
        --disable-cccam \
        --disable-libx264 \
        --disable-cwc \
        --disable-capmt \
        --disable-constcw \
       
    7.  edit the file “openwrt/package/feeds/packages/tvheadend/files/tvheadend.init” and add -C at the end of this line  “procd_set_param command "$PROG" -B” it should look like this         “procd_set_param command "$PROG" -B -C”
       
    8. Download and create the Tvheadend 4.3 package manually to force the cross compiler to use this file instead of download some crappy version.

Download the Commit to compile version from github (I use the Rpcameron version since it has the Cablecard support I need. However you can use the official tvheadend.
https://github.com/rpcameron/tvheadend/archive/cablecard-4.3.zip
Unzip the file, rename the folder from “tvheadend-cablecard-4.3” to “tvheadend” , 
 to avoid the error “ERROR] main: iconv() routine is not working properly ((null)), aborting!” 
Remove this patch https://tvheadend.org/projects/tvheadend/repository/revisions/4535a2cf6ccf5209d220971890869390e1567889/diff/src/intlconv.c?utf8=%E2%9C%93&type=sbs
you can manually edit the file “src/intlconv.c” and remove the patch lines.

create a Tar with the tvheadend folder  “tar -cvf tvheadend-v4.3.tar tvheadend” and create an xz file with the Tar  “xz tvheadend-v4.3.tar” this will create a file tvheadend-v4.3.tar.xz.

    9. Copy the file tvheadend-v4.3.tar.xz to the folder openwrt/dl/ 
    10. run “make menuconfig” on and go to the Multimedia folder and select the (“M”) to make the compiler create an ipk file.
    11.  Run the compiler again  “make -j4 V=s” ( if anything fails use -j1 to see what the problem is modify -j? To fit your processors cores, read before setting it wrong ) this takes 20 min aprox to compile the files.
    12. At the you will see the tvheadend file in the list of ipks, just copy it and install it in your router.
    13. Thveadend is big, it requires more than 10 mb available to install its prereqs and itself if your router is not a wrt version or has limited capacity in the /overlay directory please use a USB stick formatted in ext4 and follow the extroot guide to move the /overlay to the USB stick and have enough space to install anything.  https://wiki.openwrt.org/doc/howto/extroot
    14. once installed open http:Ipaddress:9981
    15. Enjoy

just2mad, 2019-03-11:
> Cross compile tvheadend for Openwrt 18
> 
> this guide is based in my knowledge and my own practices doing this, if you know a better way to cross-compile for opwenwrt let me know.
> 
> Everything is done in UBUNTU 18 however is the exact same if you do it in UBUNTU 16 or Debian 9.5 .
> 
> ```
> 1. Install the openwrt buildroot
>    https://wiki.openwrt.org/doc/howto/buildroot.exigence
>    
> 2. Once the buildroot is installed and you run the “make menuconfig” to configure the buildroot make sure you configure correctly the first 3 options of the menu : Target System, Subtarget and Target Profile, for your router if you dont know read openwrt wifi for your router, leave everything else, save changes and exit.
> 3. Before you compile the toolchain if you are doing this as root like me set this variable to avoid errors.  
>    set FORCE_UNSAFE_CONFIGURE=1
>    export set FORCE_UNSAFE_CONFIGURE=1 
> 4. compile : make -j4 V=s   ( if anything fails use -j1 to see what the problem is modify -j? To fit your processors cores, read before setting it wrong ) this takes 1 hr aprox to create the toolchain.
> 5. Once completed, modify the tvheadend makefile to disable ffmpeg issues and compile ver 4.3.
>    FILE location: openwrt/feeds/packages/multimedia/tvheadend/Makefile
>    change PKG_VERSION:=4.3 , 
>    comment #PKG_MIRROR_HASH:=afe42bcf683e8576ca8ccd4755ea308f79195e92f7c30fc5961973e238f49105
> 6. add the 6 last lines to the CONFIGURE_ARGS part of the file.
> ```
> CONFIGURE_ARGS += 
> --arch=$(ARCH) 
> --disable-dbus_1 
> --disable-libav 
> --enable-bundle 
> --disable-ffmpeg_static 
> --disable-cccam 
> --disable-libx264 
> --disable-cwc 
> --disable-capmt 
> --disable-constcw
> 
> ```
> 7.  edit the file “openwrt/package/feeds/packages/tvheadend/files/tvheadend.init” and add -C at the end of this line  “procd_set_param command "$PROG" -B” it should look like this         “procd_set_param command "$PROG" -B -C”
>    
> 8. Download and create the Tvheadend 4.3 package manually to force the cross compiler to use this file instead of download some crappy version.
> ```
> Download the Commit to compile version from github (I use the Rpcameron version since it has the Cablecard support I need. However you can use the official tvheadend.
> https://github.com/rpcameron/tvheadend/archive/cablecard-4.3.zip
> Unzip the file, rename the folder from “tvheadend-cablecard-4.3” to “tvheadend” ,
> to avoid the error “ERROR] main: iconv() routine is not working properly ((null)), aborting!”
> Remove this patch https://tvheadend.org/projects/tvheadend/repository/revisions/4535a2cf6ccf5209d220971890869390e1567889/diff/src/intlconv.c?utf8=%E2%9C%93&type=sbs
> you can manually edit the file “src/intlconv.c” and remove the patch lines.
> 
> create a Tar with the tvheadend folder “tar -cvf tvheadend-v4.3.tar tvheadend” and create an xz file with the Tar “xz tvheadend-v4.3.tar” this will create a file tvheadend-v4.3.tar.xz.
> 
> ```
> 9. Copy the file tvheadend-v4.3.tar.xz to the folder openwrt/dl/ 
> 10. run “make menuconfig” on and go to the Multimedia folder and select the (“M”) to make the compiler create an ipk file.
> 11.  Run the compiler again  “make -j4 V=s” ( if anything fails use -j1 to see what the problem is modify -j? To fit your processors cores, read before setting it wrong ) this takes 20 min aprox to compile the files.
> 12. At the you will see the tvheadend file in the list of ipks, just copy it and install it in your router.
> 13. Thveadend is big, it requires more than 10 mb available to install its prereqs and itself if your router is not a wrt version or has limited capacity in the /overlay directory please use a USB stick formatted in ext4 and follow the extroot guide to move the /overlay to the USB stick and have enough space to install anything.  https://wiki.openwrt.org/doc/howto/extroot
> 14. once installed open http://Ipaddress:9981
> 15. Enjoy
> ```

Hey,

I can't get it compiled, don't know whats wrong. I get the following error(s):

> time: package/libs/libiconv/compile#0.10#0.03#0.34
make[3]: Entering directory `/home/freetz/openwrt/feeds/packages/multimedia/tvheadend'
touch /home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3/.prepared_a1a77111c6dab98baa182ec8e6987762_6664                 517399ebbbc92a37c5bb081b5c53_check
. /home/freetz/openwrt/include/shell.sh; gzip -dc /home/freetz/openwrt/dl/tvheadend-4.3.tar.gz | tar -C /home/freetz/openwrt/build_dir/targe                 t-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3

Applying ./patches/010-openssl-deprecated.patch using plaintext:
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/src/main.c
|+++ b/src/main.c
--------------------------
No file to patch.  Skipping patch.
3 out of 3 hunks ignored
Patch failed!  Please fix ./patches/010-openssl-deprecated.patch!
make[3]: *** [/home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3/.prepared_a1a77111c6dab98baa182ec8e6987                 762_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory `/home/freetz/openwrt/feeds/packages/multimedia/tvheadend'
time: package/feeds/packages/tvheadend/compile#1.58#2.61#4.39
make[2]: *** [package/feeds/packages/tvheadend/compile] Error 2
make[2]: Leaving directory `/home/freetz/openwrt'
make[1]: *** [/home/freetz/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/freetz/openwrt'
make: *** [world] Fehler 2

just2mad, 2019-03-11:
> can you build openwrt withouth the tvheadend package?>

Hi,
Thanks for the answer!

Yes, it's getting built without tvheadend! 
I make every step as you described.

Whats the 010-openssl-deprecated.patch for? 
And do you is this really all changes in the Makefile?

just2mad, 2019-03-11:
PS: When I remove the patch, I'm getting the following error:

`. /home/freetz/openwrt/include/shell.sh; xzcat /home/freetz/openwrt/dl/tvheadend-v4.3.tar.xz | tar -C /home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3/.. -xf -

[ ! -d ./src/ ] || cp -fpR ./src/. /home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3
echo 'Tvheadend ~openwrt' > /home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3/debian/changelog
bash: /home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3/debian/changelog: No such file or directory
make[3]: *** [/home/freetz/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/tvheadend-4.3/.prepared_83718f9f06914b3b03589b13de4c23d7_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory `/home/freetz/openwrt/package/multimedia/tvheadend'
time: package/multimedia/tvheadend/compile#1.20#1.95#3.22
make[2]: *** [package/multimedia/tvheadend/compile] Error 2
make[2]: Leaving directory `/home/freetz/openwrt'
make[1]: *** [/home/freetz/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/freetz/openwrt'
make: *** [world] Fehler 2
root@freetz-linux:/home/freetz/openwrt#
`

benteam10, 2019-03-11:
not sure what it is. 

i recall having a similar problem using debian to build openwrt, i'm not sure what it was but i had to switch to ubuntu, i think it was something with the openssl packages. 

i still have my build environment intact if you need something i can look at it.

just2mad, 2019-03-11:
> not sure what it is.
> 
> i recall having a similar problem using debian to build openwrt, i'm not sure what it was but i had to switch to ubuntu, i think it was something with the openssl packages.
> 
> i still have my build environment intact if you need something i can look at it.

ok, thanks!

I'd need a fritzbox 7530 tvheadend package. soc is ipq40xx

thanks!

benteam10, 2019-03-20:
sorry i have been very busy with a lot of work. 

let me get the makefile and upload it here.

benteam10, 2019-03-20:
#
# Copyright (C) 2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=tvheadend
PKG_VERSION:=4.3
PKG_RELEASE:=1

PKG_SOURCE_SUBDIR:=$(PKG_NAME)
PKG_SOURCE_VERSION:=v$(PKG_VERSION)
PKG_SOURCE:=$(PKG_NAME)-$(PKG_SOURCE_VERSION).tar.xz
#PKG_MIRROR_HASH:=afe42bcf683e8576ca8ccd4755ea308f79195e92f7c30fc5961973e238f49105
PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/tvheadend/tvheadend.git
PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)

PKG_LICENSE:=GPL-3.0
PKG_LICENSE_FILES:=LICENSE.md

PKG_FIXUP:=autoreconf

PKG_USE_MIPS16:=0

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/nls.mk

define Package/tvheadend
  SECTION:=multimedia
  CATEGORY:=Multimedia
  TITLE:=Tvheadend is a TV streaming server for Linux
  DEPENDS:=+libopenssl +librt +zlib +TVHEADEND_AVAHI_SUPPORT:libavahi-client $(ICONV_DEPENDS)
  URL:=https://tvheadend.org
  MAINTAINER:=Jan Čermák <jan.cermak@nic.cz>
endef

define Package/tvheadend/description
  Tvheadend is a TV streaming server and recorder for Linux, FreeBSD and Android
  supporting DVB-S, DVB-S2, DVB-C, DVB-T, ATSC, IPTV, SAT>IP and HDHomeRun as input sources.

  Tvheadend offers the HTTP (VLC, MPlayer), HTSP (Kodi, Movian) and SAT>IP streaming.
endef

define Package/tvheadend/config
  menu "Configuration"
  depends on PACKAGE_tvheadend
  source "$(SOURCE)/Config.in"
  endmenu
endef

ifeq ($(CONFIG_TVHEADEND_CWC_SUPPORT),)
  CONFIGURE_ARGS += --disable-cwc
endif

ifeq ($(CONFIG_TVHEADEND_LINUXDVB_SUPPORT),)
  CONFIGURE_ARGS += --disable-linuxdvb
endif

ifeq ($(CONFIG_TVHEADEND_DVBSCAN_SUPPORT),)
  CONFIGURE_ARGS += --disable-dvbscan
endif

ifeq ($(CONFIG_TVHEADEND_AVAHI_SUPPORT),)
  CONFIGURE_ARGS += --disable-avahi
else
  CONFIGURE_ARGS += --enable-avahi
endif

CONFIGURE_ARGS += \
	--arch=$(ARCH) \
	--disable-dbus_1 \
	--disable-libav \
	--enable-bundle \
	--disable-ffmpeg_static \
	--disable-cccam \
	--disable-libx264 \
	--disable-cwc \
	--disable-capmt \
	--disable-constcw \

#TARGET_CFLAGS += -Wno-error=pointer-compare

define Build/Prepare
	$(call Build/Prepare/Default)
	echo 'Tvheadend $(shell echo $(PKG_SOURCE_VERSION) | sed "s/^v//")~openwrt$(PKG_RELEASE)' \
		> $(PKG_BUILD_DIR)/debian/changelog
endef

define Package/conffiles
/etc/config/tvheadend
endef

define Package/tvheadend/install
	$(INSTALL_DIR) $(1)/etc/init.d
	$(INSTALL_BIN) ./files/tvheadend.init $(1)/etc/init.d/tvheadend
	$(INSTALL_DIR) $(1)/etc/config
	$(INSTALL_CONF) ./files/tvheadend.config $(1)/etc/config/tvheadend

	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/build.linux/tvheadend $(1)/usr/bin/
endef

$(eval $(call BuildPackage,tvheadend))

benteam10, 2019-03-20:
Makefile:

_______________________________________________________________________

#
# Copyright (C) 2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=tvheadend
PKG_VERSION:=4.3
PKG_RELEASE:=1

PKG_SOURCE_SUBDIR:=$(PKG_NAME)
PKG_SOURCE_VERSION:=v$(PKG_VERSION)
PKG_SOURCE:=$(PKG_NAME)-$(PKG_SOURCE_VERSION).tar.xz
#PKG_MIRROR_HASH:=afe42bcf683e8576ca8ccd4755ea308f79195e92f7c30fc5961973e238f49105
PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/tvheadend/tvheadend.git
PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)

PKG_LICENSE:=GPL-3.0
PKG_LICENSE_FILES:=LICENSE.md

PKG_FIXUP:=autoreconf

PKG_USE_MIPS16:=0

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/nls.mk

define Package/tvheadend
  SECTION:=multimedia
  CATEGORY:=Multimedia
  TITLE:=Tvheadend is a TV streaming server for Linux
  DEPENDS:=+libopenssl +librt +zlib +TVHEADEND_AVAHI_SUPPORT:libavahi-client $(ICONV_DEPENDS)
  URL:=https://tvheadend.org
  MAINTAINER:=Jan Čermák <jan.cermak@nic.cz>
endef

define Package/tvheadend/description
  Tvheadend is a TV streaming server and recorder for Linux, FreeBSD and Android
  supporting DVB-S, DVB-S2, DVB-C, DVB-T, ATSC, IPTV, SAT>IP and HDHomeRun as input sources.

  Tvheadend offers the HTTP (VLC, MPlayer), HTSP (Kodi, Movian) and SAT>IP streaming.
endef

define Package/tvheadend/config
  menu "Configuration"
  depends on PACKAGE_tvheadend
  source "$(SOURCE)/Config.in"
  endmenu
endef

ifeq ($(CONFIG_TVHEADEND_CWC_SUPPORT),)
  CONFIGURE_ARGS += --disable-cwc
endif

ifeq ($(CONFIG_TVHEADEND_LINUXDVB_SUPPORT),)
  CONFIGURE_ARGS += --disable-linuxdvb
endif

ifeq ($(CONFIG_TVHEADEND_DVBSCAN_SUPPORT),)
  CONFIGURE_ARGS += --disable-dvbscan
endif

ifeq ($(CONFIG_TVHEADEND_AVAHI_SUPPORT),)
  CONFIGURE_ARGS += --disable-avahi
else
  CONFIGURE_ARGS += --enable-avahi
endif

CONFIGURE_ARGS += \
	--arch=$(ARCH) \
	--disable-dbus_1 \
	--disable-libav \
	--enable-bundle \
	--disable-ffmpeg_static \
	--disable-cccam \
	--disable-libx264 \
	--disable-cwc \
	--disable-capmt \
	--disable-constcw \

#TARGET_CFLAGS += -Wno-error=pointer-compare

define Build/Prepare
	$(call Build/Prepare/Default)
	echo 'Tvheadend $(shell echo $(PKG_SOURCE_VERSION) | sed "s/^v//")~openwrt$(PKG_RELEASE)' \
		> $(PKG_BUILD_DIR)/debian/changelog
endef

define Package/conffiles
/etc/config/tvheadend
endef

define Package/tvheadend/install
	$(INSTALL_DIR) $(1)/etc/init.d
	$(INSTALL_BIN) ./files/tvheadend.init $(1)/etc/init.d/tvheadend
	$(INSTALL_DIR) $(1)/etc/config
	$(INSTALL_CONF) ./files/tvheadend.config $(1)/etc/config/tvheadend

	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/build.linux/tvheadend $(1)/usr/bin/
endef

$(eval $(call BuildPackage,tvheadend))

benteam10, 2019-03-20:
`#
# Copyright (C) 2015 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=tvheadend
PKG_VERSION:=4.3
PKG_RELEASE:=1

PKG_SOURCE_SUBDIR:=$(PKG_NAME)
PKG_SOURCE_VERSION:=v$(PKG_VERSION)
PKG_SOURCE:=$(PKG_NAME)-$(PKG_SOURCE_VERSION).tar.xz
#PKG_MIRROR_HASH:=afe42bcf683e8576ca8ccd4755ea308f79195e92f7c30fc5961973e238f49105
PKG_SOURCE_PROTO:=git
PKG_SOURCE_URL:=https://github.com/tvheadend/tvheadend.git
PKG_BUILD_DIR:=$(BUILD_DIR)/$(PKG_NAME)

PKG_LICENSE:=GPL-3.0
PKG_LICENSE_FILES:=LICENSE.md

PKG_FIXUP:=autoreconf

PKG_USE_MIPS16:=0

include $(INCLUDE_DIR)/package.mk
include $(INCLUDE_DIR)/nls.mk

define Package/tvheadend
  SECTION:=multimedia
  CATEGORY:=Multimedia
  TITLE:=Tvheadend is a TV streaming server for Linux
  DEPENDS:=+libopenssl +librt +zlib +TVHEADEND_AVAHI_SUPPORT:libavahi-client $(ICONV_DEPENDS)
  URL:=https://tvheadend.org
  MAINTAINER:=Jan Čermák <jan.cermak@nic.cz>
endef

define Package/tvheadend/description
  Tvheadend is a TV streaming server and recorder for Linux, FreeBSD and Android
  supporting DVB-S, DVB-S2, DVB-C, DVB-T, ATSC, IPTV, SAT>IP and HDHomeRun as input sources.

  Tvheadend offers the HTTP (VLC, MPlayer), HTSP (Kodi, Movian) and SAT>IP streaming.
endef

define Package/tvheadend/config
  menu "Configuration"
  depends on PACKAGE_tvheadend
  source "$(SOURCE)/Config.in"
  endmenu
endef

ifeq ($(CONFIG_TVHEADEND_CWC_SUPPORT),)
  CONFIGURE_ARGS += --disable-cwc
endif

ifeq ($(CONFIG_TVHEADEND_LINUXDVB_SUPPORT),)
  CONFIGURE_ARGS += --disable-linuxdvb
endif

ifeq ($(CONFIG_TVHEADEND_DVBSCAN_SUPPORT),)
  CONFIGURE_ARGS += --disable-dvbscan
endif

ifeq ($(CONFIG_TVHEADEND_AVAHI_SUPPORT),)
  CONFIGURE_ARGS += --disable-avahi
else
  CONFIGURE_ARGS += --enable-avahi
endif

CONFIGURE_ARGS += \
	--arch=$(ARCH) \
	--disable-dbus_1 \
	--disable-libav \
	--enable-bundle \
	--disable-ffmpeg_static \
	--disable-cccam \
	--disable-libx264 \
	--disable-cwc \
	--disable-capmt \
	--disable-constcw \

#TARGET_CFLAGS += -Wno-error=pointer-compare

define Build/Prepare
	$(call Build/Prepare/Default)
	echo 'Tvheadend $(shell echo $(PKG_SOURCE_VERSION) | sed "s/^v//")~openwrt$(PKG_RELEASE)' \
		> $(PKG_BUILD_DIR)/debian/changelog
endef

define Package/conffiles
/etc/config/tvheadend
endef

define Package/tvheadend/install
	$(INSTALL_DIR) $(1)/etc/init.d
	$(INSTALL_BIN) ./files/tvheadend.init $(1)/etc/init.d/tvheadend
	$(INSTALL_DIR) $(1)/etc/config
	$(INSTALL_CONF) ./files/tvheadend.config $(1)/etc/config/tvheadend

	$(INSTALL_DIR) $(1)/usr/bin
	$(INSTALL_BIN) $(PKG_BUILD_DIR)/build.linux/tvheadend $(1)/usr/bin/
endef

$(eval $(call BuildPackage,tvheadend))
`

kkudielka, 2019-10-15:
In case anybody is interested, I have cast the instructions above (plus a few more tweaks) into a patch:
[https://github.com/kkudielka/packages/commits/playground](https://github.com/kkudielka/packages/commits/playground)
It seems to compile in all configurations, and seems to run on my Turris Omnia (IPTV only).

erotavlas85, 2019-10-16:
@kkudielka very good job. It would be nice to a working package for all openWRT.