openwrt/packages #6517: acme: should check for port-forwards

paour, 2018-07-18:
# Issue template (remove lines from top till here)

Maintainer: @tohojo 
Environment: 
```
root@OverTheBox:~# cat /etc/openwrt_release
DISTRIB_ID='otb'
DISTRIB_RELEASE='v3.4'
DISTRIB_REVISION='r5234-451d0221da'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='otb Reboot v3.4 v0.6.21-1-gc00c79b'
DISTRIB_TAINTS='busybox override'
```

Description:
The acme run script checks for existing listeners on port 80, and it adds an accept rule on the firewall, but what it neglects to do is check for port-forward rules, which cause the underlying acme.sh script to fail, because it's not getting the request – it's forwarded to another server.

tohojo, 2018-07-18:
Pierre-Luc Paour <notifications@github.com> writes:

> The acme run script checks for existing listeners on port 80, and it
> adds an accept rule on the firewall, but what it neglects to do is
> check for port-forward rules, which cause the underlying acme.sh
> script to fail, because it's not getting the request – it's forwarded
> to another server.

Why are you trying to run acme while forwarding the port? And what is
the expected behaviour you are trying to get out of it?

karlp, 2018-07-18:
Personally I don't see any reason to try and mangle the package to support you having http and https going to different servers.  You're in the deep end of your own network configuration at this point.

lucize, 2018-07-18:
> checks for existing listeners on port 80, and it adds an accept rule on the firewall

for me is not working, i'm using ngnix on 80 and uhttpd on 8080, or the last time it didn't

paour, 2018-07-18:
I agree there are many use-cases for port 80 and this package may not be able to account for all of them. Perhaps it should try to curl to localhost:80 just for one last check before invoking acme.sh though.

tohojo, 2018-07-19:
Pierre-Luc Paour <notifications@github.com> writes:

> Perhaps it should try to curl to localhost:80 just for one last check
> before invoking acme.sh though.

But that is likely to produce different results than an outside request,
so don't think it's terribly useful.

I think there are too many corner cases that checking for this case
makes sense. For example, a port forward can be installed for IPv4 while
the router is still accessible via IPv6, in which case we *don't* want
to abort (this is the setup I use to get around the same restriction,
actually).

-Toke

paour, 2018-07-19:
Indeed. I guess one way to address this would be to add a troubleshooting link to the luci app, so that in the case of a failure we could list potential pitfalls.

Thank you for taking the time to respond, and let's close this ticket.

closed 2018-07-19 by tohojo