openwrt/packages #6536: php7-mod-intl: idn converting doesnt work

Tatikoma, 2018-07-21:
Maintainer: @mhei
Environment: MIPS 74Kc V5.0, TP-Link Archer C7 v4, OpenWrt SNAPSHOT, r7531-03fce62

Description:
PHP functions idn_to_ascii and idn_to_utf8 doesnt work at all.

Code example:
```php
<?php

$x = idn_to_ascii('яндекс.ру', IDNA_DEFAULT, INTL_IDNA_VARIANT_UTS46, $info);
var_dump($x, $info);

$x = idn_to_utf8('xn--tst-qla.de', IDNA_DEFAULT, INTL_IDNA_VARIANT_UTS46, $info);
var_dump($x, $info);
```
Expected output (tested on local machine with the same php version):
```
/home/tatikoma/x.php:4:
string(22) "xn--d1acpjx3f.xn--p1ag"
/home/tatikoma/x.php:4:
array(3) {
  'result' =>
  string(22) "xn--d1acpjx3f.xn--p1ag"
  'isTransitionalDifferent' =>
  bool(false)
  'errors' =>
  int(0)
}
/home/tatikoma/x.php:8:
string(8) "täst.de"
/home/tatikoma/x.php:8:
array(3) {
  'result' =>
  string(8) "täst.de"
  'isTransitionalDifferent' =>
  bool(false)
  'errors' =>
  int(0)
}
```

Actual output (logged via ssh to router, executed php-cli myscript.php):
```
bool(false)
array(0) {
}
bool(false)
array(0) {
}
```

mhei, 2018-09-03:
@nxhack: I'm not so familiar with icu. Could you please have a look into this issue? It seems that including ICU data into the library make this library useful. However, I could imagine that there are use-cases where people are shocked by the new size of around 30 MB the new package. So I think we should add an opt-in or opt-out for the data...

nxhack, 2018-09-03:
If you want to process i18n, it will inevitably increase in size. I agree to change icu package to choose options as you say.

nxhack, 2018-09-04:
I tried ICU Data as a separate package. In order to reduce ICU Data, processing such as limiting to the required language is necessary. It can not correspond with general purpose package. So I also made the tools necessary for creating ICU Data in a separate package.

If this does not seem to be a problem, I will send a pull request.

https://github.com/nxhack/packages/commit/89c7e945db303ff2e4df86e658b4727237eafbf8

mhei, 2018-09-04:
@nxhack: Thanks a lot, this looks exactly as I had imagined. I'll give it a test run later this evening...

mhei, 2018-09-04:
@nxhack: As promised, here are my test results:

1. I cherry-picked your commit in a test branch. I recommend also use leading tabs in the Package/icu-tools/install section - looks more consistent.
2. I only installed the icu library - without tools and without data package -> the problem reported above exists.
3. I additionally installed the data package -> the problem is gone.

So this works as expected: installing only the library package allows to compile the software and run it up a certain degree; if you really need a working icu installation, then you have to install the data package, too.

Please add such an explanation to the description of icu / icu-data package, so that users can read the help to understand what to do.

Regarding the new tools package: I recommend to rename it to icu-devtools, because usually these tools are not required on the target, except you are compiling on the target, right?
Also, I think that including icu-config is not good because it contains cross-compiler stuff and paths - and when installed on the target, you are not cross-compiling but running a native build process!
However, I only had a short look into this file - not sure whether I understood all correctly...

nxhack, 2018-09-05:
Modified. I think that it is clear than before.

https://github.com/nxhack/packages/commit/38c150dfa37fc52e4d2b4b897876907560f70f05

mhei, 2018-09-05:
LGTM, thank you very much.

commit 3b6e346a (2018-09-07): php7: mod-intl: add optional dep for icu-full-data (fixes #6536)

As worked out, ICU library needs its data archive to be useful.
Explain the situation in help text and add option to declare/prevent
this dependency in package meta-data.
closed 2018-09-07 by mhei