openwrt/packages #6551: mwan3: Can't connect to openvpn when mwan3 is enabled (18.06.0-rc2)

Per9, 2018-07-23:
Maintainer: @FlorianFieber 
Environment: Asus RT-N56U, RAMIPS/rt3883, 18.06.0-rc2

Description: 
It works fine when mwan3 is disabled and the seemingly same configuration works well in 17.01.5. I used the [OpenVPN server setup guide](https://openwrt.org/docs/guide-user/services/vpn/openvpn/server.setup) from the wiki in both setups.

When I try to connect I get this in the client log (Parts of the external IPs have been replaced by 1 in all the logs)
```
Sat Jul 21 11:29:31 2018 OpenVPN 2.4.4 i686-w64-mingw32 [SSL (OpenSSL)] [LZO] [LZ4] [PKCS11] [AEAD] built on Sep 26 2017
Sat Jul 21 11:29:31 2018 Windows version 6.2 (Windows 8 or greater) 32bit
Sat Jul 21 11:29:31 2018 library versions: OpenSSL 1.0.2l  25 May 2017, LZO 2.10
Enter Management Password:
Sat Jul 21 11:29:31 2018 MANAGEMENT: TCP Socket listening on [AF_INET]127.0.0.1:25341
Sat Jul 21 11:29:31 2018 Need hold release from management interface, waiting...
Sat Jul 21 11:29:32 2018 MANAGEMENT: Client connected from [AF_INET]127.0.0.1:25341
Sat Jul 21 11:29:32 2018 MANAGEMENT: CMD 'state on'
Sat Jul 21 11:29:32 2018 MANAGEMENT: CMD 'log all on'
Sat Jul 21 11:29:32 2018 MANAGEMENT: CMD 'echo all on'
Sat Jul 21 11:29:32 2018 MANAGEMENT: CMD 'hold off'
Sat Jul 21 11:29:32 2018 MANAGEMENT: CMD 'hold release'
Sat Jul 21 11:29:32 2018 NOTE: --fast-io is disabled since we are running on Windows
Sat Jul 21 11:29:32 2018 Outgoing Control Channel Authentication: Using 160 bit message hash 'SHA1' for HMAC authentication
Sat Jul 21 11:29:32 2018 Incoming Control Channel Authentication: Using 160 bit message hash 'SHA1' for HMAC authentication
Sat Jul 21 11:29:32 2018 MANAGEMENT: >STATE:1532165372,RESOLVE,,,,,,
Sat Jul 21 11:29:33 2018 TCP/UDP: Preserving recently used remote address: [AF_INET]1.1.1.1:1194
Sat Jul 21 11:29:33 2018 Socket Buffers: R=[65536->65536] S=[65536->65536]
Sat Jul 21 11:29:33 2018 UDP link local: (not bound)
Sat Jul 21 11:29:33 2018 UDP link remote: [AF_INET]1.1.1.1:1194
Sat Jul 21 11:29:33 2018 MANAGEMENT: >STATE:1532165373,WAIT,,,,,,
Sat Jul 21 11:30:03 2018 SIGTERM[hard,] received, process exiting
Sat Jul 21 11:30:03 2018 MANAGEMENT: >STATE:1532165403,EXITING,SIGTERM,,,,,

```

On the server side it says
```
Sat Jul 21 09:30:33 2018 47.60.41.55:61617 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Sat Jul 21 09:30:33 2018 47.60.41.55:61617 TLS Error: TLS handshake failed

```
Diagnostics from mwan with it is disabled (works):
```
MWAN Status - Troubleshooting
INFO: MWAN not running

Software-Version
-------------------------------------------------
OpenWrt - OpenWrt 18.06.0-rc2 r7141-e4d0ee5af5
LuCI - git-18.196.56128-9112198

Output of "ip a show"
-------------------------------------------------
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether bc:ee:7b:56:58:55 brd ff:ff:ff:ff:ff:ff
    inet6 1::beee:7bff:fe56:5855/64 scope link
       valid_lft forever preferred_lft forever
3: ifb0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 32:1c:61:cd:58:9b brd ff:ff:ff:ff:ff:ff
4: ifb1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether da:62:b1:76:5a:b1 brd ff:ff:ff:ff:ff:ff
7: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether be:ee:7b:56:58:55 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/24 brd 192.168.1.255 scope global br-lan
       valid_lft forever preferred_lft forever
[... 255 lines cut ...]
Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 mwan3_policy_wan_only  tcp  --  *      *       192.168.1.3          0.0.0.0/0            multiport sports 0:65535 multiport dports 119,563 mark match 0x0/0x3f00 /* nntp */
  430 22415 mwan3_rule_https  tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            multiport sports 0:65535 multiport dports 443 mark match 0x0/0x3f00 /* https */
  440 28296 mwan3_policy_wan_wan2  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0x3f00 /* default_rule */

Chain qos_Default (0 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK restore mask 0xf
    0     0 qos_Default_ct  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 length 0:500 MARK xset 0x22/0xff
    0     0 MARK       icmp --  *      *       0.0.0.0/0            0.0.0.0/0            MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 tcp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 udp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff

Chain qos_Default_ct (1 references)
 pkts bytes target     prot opt in     out     source               destination
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 22,53 /* ssh, dns */ MARK xset 0x11/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 22,53 /* ssh, dns */ MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 20,21,25,80,110,443,993,995 /* ftp, smtp, http(s), imap */ MARK xset 0x33/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 5190 /* AOL, iChat, ICQ */ MARK xset 0x22/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 5190 /* AOL, iChat, ICQ */ MARK xset 0x22/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff
```

feckert, 2018-07-24:
What do you want to achive with mwan3 in this setup?
Failover?
Is wan3 running on the server or the client?

Per9, 2018-07-24:
I use it to be able to connect my mobile phone to the router for failover. Mwan3 is running on the router, which also has the OpenVPN server. I am trying to connect to the OpenVPN server from the outside.

ptpt52, 2018-08-03:
I think this PR is likely to fix your problem
https://github.com/openwrt/packages/pull/6515

Per9, 2018-08-08:
@ptpt52 is right, it will fix it.

closed 2018-08-08 by Per9