openwrt/packages #6559: mwan3 - feature request >showing "online/offline --- uptime downtime"

camelcamro, 2018-07-23:
mwan3 - showing "online/offline --- uptime downtime"
would be nice to  show in mwan3- details of luci or/and via status cmd
current mwan3 version:

opkg list | grep mwan3
luci-app-mwan3 - git-18.203.74221-1961ce6-1
mwan3 - 2.6.18-1

#######
with such kind of change could be visible: like:
**Interface status:
 interface 4G is unknown and tracking is down
 interface 4GA is unknown and tracking is down
 interface VPN_ROUT02 is online since 00h:01m:18s and uptime 00h:34m:58s and tracking is active
 interface VPN_ROUT03 is online since 00h:00m:45s and uptime 00h:34m:58s and tracking is active
 interface VPN_ROUT04 is online since 00h:00m:14s and uptime 00h:34m:58s and tracking is active**

therefore would be needed such kind of change in /lib/mwan3/mwan3.sh

line 826
-		result="$(mwan3_get_iface_hotplug_state $1)"
+		ifuptime=`/sbin/ifstatus $1 | grep 'uptime' | grep -o '[0-9]\+'`
+		ifonline=$(($(date +%s) - $(date +%s -r /var/run/mwan3/iface_state/$1)))
+		result="$(mwan3_get_iface_hotplug_state $1) since $(printf '%02dh:%02dm:%02ds\n' $(($ifonline/3600)) $(($ifonline%3600/60)) $(($ifonline%60))) and uptime $(printf '%02dh:%02dm:%02ds\n' $(($ifuptime/3600)) $(($ifuptime%3600/60)) $(($ifuptime%60)))"

but maybe there is a better coding for this to show .... I'm not an expert ..

camelcamro, 2018-07-23:
additional, there are 2 cut cmd, which are giving me always an cut error.
therefore i replaced the ipv4 and ipv6 rules to read ...

in same file: line ~857
`mwan3_report_policies_v4()
{
	local percent policy share total_weight weight iface

	for policy in $($IPT4 -S | awk '{print $2}' | grep mwan3_policy_ | sort -u); do
		echo "$policy:" | sed 's/mwan3_policy_//'

-		[ -n "$total_weight" ] || total_weight=$($IPT4 -S $policy | cut -s -d'"' -f2 | head -1 | awk '{print $3}')
+		[ -n "$total_weight" ] || total_weight=$($IPT4 -S $policy | grep 'comment' | awk -F\" '{print $2}' | head -1 | awk '{print $3}')
`

and also for v6 - line ~880

`mwan3_report_policies_v6()
{
	local percent policy share total_weight weight iface

	for policy in $($IPT6 -S | awk '{print $2}' | grep mwan3_policy_ | sort -u); do
		echo "$policy:" | sed 's/mwan3_policy_//'

-		[ -n "$total_weight" ] || total_weight=$($IPT6 -S $policy | cut -s -d'"' -f2 | head -1 | awk '{print $3}')
+		[ -n "$total_weight" ] || total_weight=$($IPT6 -S $policy | grep 'comment' | awk -F\" '{print $2}' | head -1 | awk '{print $3}')
`

feckert, 2018-07-24:
@camelcamro please open a PR to get this fixed

camelcamro, 2018-07-24:
I'm really sorry, but i can't create a Pull Request :(

feckert, 2018-08-03:
@camelcamro Thanks for your feature request. I have set it to my todo list, but this will take a while.
Please ping me if there is no progress in september :-)

feckert, 2018-10-09:
@camelcamro Now I had time to implement your feature request.
Please test the last two commits from the branch https://github.com/TDT-AG/packages/tree/pr/20181009-net-mwan3. The Information for uptime/downtime are integrated int the ubus call.

> ubus call mwan3 status

If this works for you i will open a PR

camelcamro, 2018-10-13:
@feckert 
ok, tested and working .. sorry for the delay.
would be nice, if you can now add it to the luci-mwan3 package to get the online+downline time + interface uptime 
like as described above:

`Interface status:
interface 4G is unknown and tracking is down
interface 4GA is unknown and tracking is down
interface VPN_ROUT02 is online 00h:01m:18s , uptime 00h:34m:58s and tracking is active
interface VPN_ROUT03 is online 00h:00m:45s , uptime 00h:34m:58s and tracking is active
interface VPN_ROUT04 is online 00h:00m:14s , uptime 00h:34m:58s and tracking is active`

camelcamro, 2018-11-08:
can i help anything to add the "ONLINE/OFFLINE/UPTIME/DOWNTIME to the mwan3 luci package ? (as already added to the mwan3 to get this infos ....) to close the ticket ?

feckert, 2018-11-08:
I will do this maybe next week. The Problem is that I want to use **ubus** to get the status information. Because on this place the information was added. On the luci-app-mwan3  -> status -> detail page we use there the command **mwan3 status**. With this command the information your look for is not present.

The preferred method is to use **ubus**. But there are some information missing which the command **mwan3 status** shows. I have to fix this first to add the missing information to **ubus**.

camelcamro, 2018-11-08:
ok, great .. let me know when it is ready to help testing .. thx cu

feckert, 2018-11-21:
@camelcamro I have added this to the ubus output
https://github.com/openwrt/packages/commit/1ee9fb4aff3ee78ab015a32de0a3ab1e594210bf#diff-34590d11545c84b961d1992ce3aefdeb
https://github.com/openwrt/packages/commit/12360037332176f9a286ec754d340a9721479abf#diff-34590d11545c84b961d1992ce3aefdeb
So please test and close this issue/feature request
If we need this info in the LuCI we have to add another feature request on this Repository
Thanks

camelcamro, 2018-11-22:
ok, closing it ... now we need to add to the luci-mwan3 to see the status.
i will try to open a new ticket for it

closed 2018-11-22 by camelcamro