openwrt/packages #6567: xmlrpc-c fails on buildbots due to missing files

diizzyy, 2018-07-24:
http://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a9_neon/packages/xmlrpc-c/internal/compile.txt

Pasted here in case it gets lost

```

Applying ./patches/001-config.mk.in.patch using plaintext: 
patching file config.mk.in

Applying ./patches/002-automake-compat.patch using plaintext: 
patching file configure.in
autoreconf: Entering directory `.'
autoreconf: configure.in: not using Gettext
autoreconf: running: /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/host/bin/aclocal -I /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/host/share/aclocal -I /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/hostpkg/share/aclocal -I /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/usr/share/aclocal -I m4 -I . --force 
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
autoreconf: configure.in: tracing
autoreconf: configure.in: not using Libtool
autoreconf: running: /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/host/bin/autoconf --include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/host/share/aclocal --include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/hostpkg/share/aclocal --include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/usr/share/aclocal --include=m4 --include=. --prepend-include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/host/share/aclocal --force
autoreconf: running: /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/host/bin/autoheader --include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/host/share/aclocal --include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/hostpkg/share/aclocal --include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/usr/share/aclocal --include=m4 --include=. --prepend-include=/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/host/share/aclocal --force
autoreconf: running: /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/host/bin/automake --add-missing --force-missing
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
configure.in:12: warning: AM_INIT_AUTOMAKE: two- and three-arguments forms are deprecated.  For more info, see:
configure.in:12: http://www.gnu.org/software/automake/manual/automake.html#Modernize-AM_005fINIT_005fAUTOMAKE-invocation
configure.in:94: installing './compile'
automake: error: no 'Makefile.am' found for any configure output
autoreconf: /data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/host/bin/automake failed with exit status: 1
configure: WARNING: unrecognized options: --disable-nls
configure: loading site script /data/bowl-builder/arm_cortex-a9_neon/build/sdk/include/site/arm
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for arm-openwrt-linux-strip... arm-openwrt-linux-muslgnueabi-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... arm-openwrt-linux-gnu
checking whether to build Wininet client XML transport module... no
checking for curl-config... no
configure: You don't appear to have Curl installed (no working curl-config in your command search path), so we will not build the Curl client XML transport
checking whether to build Curl client XML transport module... no
checking whether to build Libwww client XML transport module... no
checking for style of include used by make... GNU
checking for arm-openwrt-linux-gcc... arm-openwrt-linux-muslgnueabi-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether arm-openwrt-linux-muslgnueabi-gcc accepts -g... yes
checking for arm-openwrt-linux-muslgnueabi-gcc option to accept ISO C89... none needed
checking whether arm-openwrt-linux-muslgnueabi-gcc understands -c and -o together... yes
checking dependency style of arm-openwrt-linux-muslgnueabi-gcc... none
checking for main in -lncurses... yes
checking for main in -lreadline... no
checking whether to build tools... no
checking whether to build Abyss server module... no
checking whether to build CGI server module... no
checking whether to build C++ wrappers and tools... no
checking for arm-openwrt-linux-gcc... (cached) arm-openwrt-linux-muslgnueabi-gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether arm-openwrt-linux-muslgnueabi-gcc accepts -g... (cached) yes
checking for arm-openwrt-linux-muslgnueabi-gcc option to accept ISO C89... (cached) none needed
checking whether arm-openwrt-linux-muslgnueabi-gcc understands -c and -o together... (cached) yes
checking dependency style of arm-openwrt-linux-muslgnueabi-gcc... (cached) none
checking whether we are using the GNU C++ compiler... yes
checking whether arm-openwrt-linux-muslgnueabi-g++ accepts -g... yes
checking dependency style of arm-openwrt-linux-muslgnueabi-g++... none
checking for socket... yes
[... 1902 lines cut ...]
/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/toolchain-arm_cortex-a9+neon_gcc-7.3.0_musl_eabi/bin/../lib/gcc/arm-openwrt-linux-muslgnueabi/7.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar: `u' modifier ignored since `D' is the default (see `U')
arm-openwrt-linux-muslgnueabi-gcc-ranlib libxmlrpc.a
arm-openwrt-linux-muslgnueabi-gcc -c -o registry.o -DNDEBUG -Wall -W -Wno-uninitialized -Wundef -Wimplicit -Wno-unknown-pragmas -Wmissing-declarations -Wstrict-prototypes -Wmissing-prototypes -fno-common -g -O3  -Iblddir -Iblddir/include -Isrcdir/include -Isrcdir/lib/util/include   -Os -pipe -mcpu=cortex-a9 -mfpu=neon -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/data/bowl-builder/arm_cortex-a9_neon/build/sdk/build_dir/target-arm_cortex-a9+neon_musl_eabi/xmlrpc-c-internal/xmlrpc-c-1.39.13:xmlrpc-c-1.39.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/usr/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/toolchain-arm_cortex-a9+neon_gcc-7.3.0_musl_eabi/usr/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/toolchain-arm_cortex-a9+neon_gcc-7.3.0_musl_eabi/include/fortify -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/toolchain-arm_cortex-a9+neon_gcc-7.3.0_musl_eabi/include  registry.c
arm-openwrt-linux-muslgnueabi-gcc -c -o method.o -DNDEBUG -Wall -W -Wno-uninitialized -Wundef -Wimplicit -Wno-unknown-pragmas -Wmissing-declarations -Wstrict-prototypes -Wmissing-prototypes -fno-common -g -O3  -Iblddir -Iblddir/include -Isrcdir/include -Isrcdir/lib/util/include   -Os -pipe -mcpu=cortex-a9 -mfpu=neon -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/data/bowl-builder/arm_cortex-a9_neon/build/sdk/build_dir/target-arm_cortex-a9+neon_musl_eabi/xmlrpc-c-internal/xmlrpc-c-1.39.13:xmlrpc-c-1.39.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/usr/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/toolchain-arm_cortex-a9+neon_gcc-7.3.0_musl_eabi/usr/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/toolchain-arm_cortex-a9+neon_gcc-7.3.0_musl_eabi/include/fortify -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/toolchain-arm_cortex-a9+neon_gcc-7.3.0_musl_eabi/include  method.c
[... cut ...] -builder/arm_cortex-a9_neon/build/sdk/build_dir/target-arm_cortex-a9+neon_musl_eabi/xmlrpc-c-internal/xmlrpc-c-1.39.13:xmlrpc-c-1.39.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/usr/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/staging_dir/target-arm_cortex-a9+neon_musl_eabi/include -I/data/bowl-builder/arm_cortex-a9_neon/build/sdk/st

neheb, 2018-07-25:
Hrm rtorrent select everything except the client. Will fix. The actual error seems to be this: 

Package xmlrpc-c-client is missing dependencies for the following libraries:
libmbedcrypto.so.3
libmbedtls.so.11
libmbedx509.so.0

diizzyy, 2018-07-26:
Fixed, https://github.com/openwrt/packages/commit/5399ded5992fb6d8c0b28f6481a95e0493cc16a0

closed 2018-07-26 by diizzyy