openwrt/packages #6571: mpd-full: mpd-full is not a full-blown MPD package compared to mpd-mini since commit 2cedae0aa9befe1135c83f007942d89ec906658e

xabolcs, 2018-07-25:
Maintainer: @thess
### Environment:
 * arch:
 ```
# opkg print-architecture
arch all 1
arch noarch 1
arch mipsel_24kc 10
```
* version and model: 
```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r7512-18b87b1
 -----------------------------------------------------
 | Machine: D-Link DIR-860L B1                       |
```
### Description:
`mpd-full` package doesn't have `flac` encoder, but `mpd-mini` has.

(Continuing from commit 2cedae0aa9befe1135c83f007942d89ec906658e comment [...](https://github.com/openwrt/packages/commit/2cedae0aa9befe1135c83f007942d89ec906658e#commitcomment-29819856) )

I wanted to use the `httpd` output with a compressed format. `flac` was supported by `mpd-full` in older versions of OpenWrt, like 15.05 and 17.01 and it had moderate CPU usage with `compression 0`.

Removing `libffmpeg-mini` and installing `libffmpeg-full` doesn't help.

It would be nice if `mpd-full` would have `flac` encoding feature like before.

### `mpd-mini`
#### `opkg info`
```
# opkg info mpd-mini
Package: mpd-mini
Version: 0.20.9-2
Depends: libc, zlib, libcurl, libpthread, libmpdclient, libstdcpp, alsa-lib, boost, boost-container, libexpat, libflac, libmpg123, libvorbisidec
Provides: mpd
Status: install user installed
Section: sound
Architecture: mipsel_24kc
Size: 251890
Filename: mpd-mini_0.20.9-2_mipsel_24kc.ipk
Conffiles:
 /etc/mpd.conf 5197c906c8fa62cf8daab0a694c09861e8658933010bcb7b607a2cd65e9bd0fa
Description: Music Player Daemon (MPD) is a flexible, powerful, server-side
 application for playing music. It is typically controlled over a
 network using one of it's many clients including mpc (console),
 gmpc (gnome), phpmp (php), etc...
 .
 This package contains a minimal Music Player Daemon, with support for
 only Flac, MP3 & OGG media types & only file: & http: protocols.
Installed-Time: 1532475115

```
#### `mpd --version`
```
# mpd --version
Music Player Daemon 0.20.9

Copyright (C) 2003-2007 Warren Dukes <warren.dukes@gmail.com>
Copyright 2008-2017 Max Kellermann <max.kellermann@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Database plugins:
 simple proxy

Storage plugins:
 local

Decoders plugins:
 [mpg123] mp3
 [vorbis] ogg oga
 [oggflac] ogg oga
 [flac] flac
 [dsdiff] dff
 [dsf] dsf
 [pcm]

Filters:

Tag plugins:

Output plugins:
 null alsa oss httpd

Encoder plugins:
 null wave flac

Input plugins:
 file alsa curl

Playlist plugins:
 extm3u m3u pls xspf asx rss flac cue embcue

Protocols:
 file:// http:// https:// alsa://

Other features:
 epoll iconv inotify ipv6 tcp un
```

### `mpd-full`
#### `opkg info`
Note the missing `libflac` dependency!
```
# opkg info mpd-full
Package: mpd-full
Version: 0.20.9-2
Depends: libc, zlib, libcurl, libpthread, libmpdclient, libstdcpp, alsa-lib, boost, boost-container, libexpat, libffmpeg, libid3tag, libmms, libupnp, libshout
Provides: mpd
Status: unknown ok not-installed
Section: sound
Architecture: mipsel_24kc
Size: 277363
Filename: mpd-full_0.20.9-2_mipsel_24kc.ipk
Description: Music Player Daemon (MPD) is a flexible, powerful, server-side
 application for playing music. It is typically controlled over a
 network using one of it's many clients including mpc (console),
 gmpc (gnome), phpmp (php), etc...
 .
 This package contains a full-blown Music Player Daemon.

```
#### `mpd --version`
Note the missing `flac` encoder plugin!
```
# mpd --version
Music Player Daemon 0.20.9

Copyright (C) 2003-2007 Warren Dukes <warren.dukes@gmail.com>
Copyright 2008-2017 Max Kellermann <max.kellermann@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Database plugins:
 simple proxy upnp

Storage plugins:
 local

Neighbor plugins:
 upnp

Decoders plugins:
 [dsdiff] dff
 [dsf] dsf
 [ffmpeg] 16sv 3g2 3gp 4xm 8svx aa3 aac ac3 adx afc aif aifc aiff al alaw amr anim apc ape asf atrac au aud avi avm2 avs bap bfi c93 cak cin cmv cpk daud dct divx dts dv dvd dxa eac3 film flac flc fli fll flx flv g726 gsm gxf iss m1v m2v m2t m2ts m4a m4b m4v mad mj2 mjpeg mjpg mka mkv mlp mm mmf mov mp+ mp1 mp2 mp3 mp4 mpc mpeg mpg mpga mpp mpu mve mvi mxf nc nsv nut nuv oga ogm ogv ogx oma ogg omg opus psp pva qcp qt r3d ra ram rl2 rm rmvb roq rpl rvc shn smk snd sol son spx str swf tak tgi tgq tgv thp ts tsp tta xa xvid uv uv2 vb vid vob voc vp6 vmd wav webm wma wmv wsaud wsvga wv wve
 [pcm]

Filters:

Tag plugins:
 id3tag

Output plugins:
 shout null fifo pipe alsa oss httpd recorder

Encoder plugins:
 null wave

Input plugins:
 file alsa curl ffmpeg mms

Playlist plugins:
 extm3u m3u pls xspf asx rss cue embcue

Protocols:
 file:// http:// https:// mms:// mmsh:// mmst:// mmsu:// gopher:// rtp:// rtsp:// rtmp:// rtmpt:// rtmps:// alsa://

Other features:
 epoll iconv inotify ipv6 tcp un
```

xabolcs, 2018-07-25:
An example `audio_output` section in `/etc/mpd.conf`"

```
audio_output {
        type            "httpd"
        name            "MPD Stream"
        encoder         "flac"
        port            "8000"
        bind_to_address "0.0.0.0"               # optional, IPv4 or IPv6
        compression     "0"                     # Sets the libFLAC compression level. The levels range from 0 (fastest, least compression) to 8 (slowest, most compression).
#       quality         "5.0"                   # do not define if bitrate is defined
#       bitrate         "128"                   # do not define if quality is defined
        format          "44100:16:2"
#       max_clients     "0"                     # optional 0=no limit
        always_on       "1"                     # prevent MPD from disconnecting all listeners when playback is stopped.
        tags            "yes"                   # httpd supports sending tags to listening streams.
}
```

diizzyy, 2018-07-25:
That's easily fixable however mpd needs a refresh, it's a pity that it doesn't support ffmpeg for encoding which would make it a lot nicer instead of having seprate libraries for everything.
I have a private repo however it's not optimized for small flash device (I'm lazy) so ffmpeg (4.02) is like ~4.5mbyte however it includes what you'd expect for normal usage. However you could probably slim it down quite a bit by dumping the video support, however I haven't managed to get it to compile on MIPS.

Anyhow, please create a pull request (if you're able) for adding flac although as I mentioned before ffmpeg support upstream would be nicer... :/

diizzyy, 2018-07-25:
..as a sidenote, this might be worth a consideration as it would lower the load by not encoding to FLAC however it does depend on what you're using on the other end.
https://undeadly.org/cgi?action=article&sid=20180410063454

xabolcs, 2018-07-25:
> That's easily fixable however ...

Sure, I fixed it by manually compiling it.

> Anyhow, please create a pull request (if you're able to) for adding flac ...

Of course I'm able add a PR with those flac lines in the `Makefile`, but I'm unsure about the intends of the referenced commit.

thess, 2018-07-25:
@xabolics - Thanks for the additional info. I will handle the update shortly with an MPD update too.
As Daniel mentioned, it's unfortunate that MPD does not use ffmpeg for encoding.

xabolcs, 2018-07-25:
>...  however it does depend on what you're using on the other end.

OpenWrt boxes on both end! 😉

thess, 2018-07-25:
MPD updated to 0.20.20 and added back flac encoder here: 91171954ea9552298829e495210a0ea5db9865a0
Closing this as fixed.

@diizzyy - Suggestions for building newer versions of mpc and libmpdclient are welcome. Both now require python3/meson/ninja - sigh...

closed 2018-07-25 by thess

diizzyy, 2018-07-25:
@thess 
It's not pretty but it does work ;-)
Since libmpdclient moves slowly I just ripped out the essential commits, left out the new build stuff and things related to the test directory. The commits should line up just fine using git on upstream repo.
[0001-libs-libmpdclient-Update-to-2.14.zip](https://github.com/openwrt/packages/files/2228688/0001-libs-libmpdclient-Update-to-2.14.zip)

thess, 2018-07-25:
Yes, that is certain gross and not very maintainable. You did give me an idea though... Bring forward the minimum autotools files with a few tweaks (config.h.in, configure.ac, Makefile.am,  libmpdclient.pc.in and m4/ax_check_compiler_flags.m4) -- and, you are good to go! You could probably do without the compiler flags check macro if you trust your environment ;-)
```
$ autoreconf -fi
$ ./configure --disable-documentation
$ make
```

diizzyy, 2018-07-25:
That said, I'm pretty sure there's at least one new source file. Is that included?

thess, 2018-07-25:
Already fully tested - As I said - you need a few tweaks. I might check them in after I get the latest mpc to build too. I take it from your question, you'd like the patch.

diizzyy, 2018-07-25:
Yeah, that sounds great :-)
I have no idea about mpc as I used ncmpcpp instead (that name) on another platform