openwrt/packages #6574: libpsl fails due to undefined dependencies

diizzyy, 2018-07-25:
@MikePetullo 

http://downloads.lede-project.org/snapshots/faillogs/mipsel_24kc/packages/libpsl/compile.txt

Pasted here in case it gets lost
```
Package libpsl is missing dependencies for the following libraries:
libidn2.so.0
libunistring.so.2
Makefile:59: recipe for target '/var/lib/buildbot/slaves/slave-lede-builds4/mipsel_24kc/build/sdk/bin/packages/mipsel_24kc/packages/libpsl_0.20.2-1_mipsel_24kc.ipk' failed
make[3]: *** [/var/lib/buildbot/slaves/slave-lede-builds4/mipsel_24kc/build/sdk/bin/packages/mipsel_24kc/packages/libpsl_0.20.2-1_mipsel_24kc.ipk] Error 1
Command exited with non-zero status 2
time: package/feeds/packages/libpsl/compile#0.08#0.02#0.76
```

diizzyy, 2018-07-25:
You probably also want to add ```--disable-rpath```

MikePetullo, 2018-07-26:
Please see pull request #6582.

diizzyy, 2018-07-26:
Great!

closed 2018-07-26 by diizzyy