openwrt/packages #6576: domoticz - fails to launch. boost library problem

ianthatap, 2018-07-25:
Maintainer: @stintel @dwmw2 
Environment: ( OpenWrt version 17.1)

Description:

 New install of openwrt. Have used the inbuilt GUI throughout to do the install. 

Errors in finding the boost library (several of them). I think this is because of a hard coding of 1.56.1 for the /usr/lib/boost*
This is likley because looking at the package release feeds, boost updated its libraries a few days after the last domoticz updates. These go away with ln - s to the new versions of the library. 

Only errors left appear to be. 
```
Error relocating /usr/bin/domoticz: _ZN5boost6thread26do_try_join_until_noexceptERK8timespecRb: symbol not found
Error relocating /usr/bin/domoticz: _ZN5boost11this_thread6hidden11sleep_untilERK8timespec: symbol not found
Error relocating /usr/bin/domoticz: _ZN5boost11this_thread6hidden9sleep_forERK8timespec: symbol not found
```

I cant crack the problem - but it is likely linked to those boosted library updates. 

best wishes - my first attempt at reporting an issue, so I hope it is actually worth reporting -  go soft, if I missed stating something obvious.

best wishes

Formating code blocks by wrapping them with pairs of ```
```

dwmw2, 2018-08-02:
I'm a little confused. A current build of OpenWRT 18.06 should work fine, building boost 1.67 and then building Domoticz against that.

You said you're using 17.01? Is that LEDE? And boost 1.56?
Precisely what domoticz package, and what boost package(s) do you have installed? And if you run 'ldd' on domoticz, what version of libboost is it linked to?

I assume you made a symlink from whatever domoticz wants, to whatever you have — and the reason that doesn't work is because they are incompatible. Which is why they have a different soname (filename) in the first place.

You need to have the correct version of libboost installed, to match the applications that link against it. I note the dependencies on the domoticz package only require "libboost", while in fact they should require the specific *version* of libboost. I'll look at fixing that at least for 18.06 onwards.
I note

dibdot, 2018-08-17:
no feedback - close this by now.

closed 2018-08-17 by dibdot