openwrt/packages #6591: libwebsockets compile error with "TCP_USER_TIMEOUT" undefined

rhyttr, 2018-07-27:
Maintainer: @karlp 
Environment: 
mipsel, mt7688, openwrt 15.05
libwebsockets 3.0.0 stable

Description:

libwebsockets 3.0.0 compile error with: "TCP_USER_TIMEOUT" undefined
there was not defined "TCP_USER_TIMEOUT" in <netinet/tcp.h> header

here is the patch tested:
```
--- a/lib/plat/lws-plat-unix.c	2018-05-07 13:38:53.000000000 +0800
+++ b/lib/plat/lws-plat-unix.c	2018-07-27 12:01:20.115515000 +0800
@@ -351,6 +351,9 @@
 		/* set the keepalive conditions we want on it too */
 
 #if defined(LWS_HAVE_TCP_USER_TIMEOUT)
+	#ifndef TCP_USER_TIMEOUT
+	#define TCP_USER_TIMEOUT	18	/* How long for loss retry before timeout */
+	#endif
 		optval = 1000 * (vhost->ka_time +
 				 (vhost->ka_interval * vhost->ka_probes));
 		if (setsockopt(fd, IPPROTO_TCP, TCP_USER_TIMEOUT,
```
Is there any other good solution?

karlp, 2018-08-04:
and is it a problem with 17.01, 1806 or master?  I'm not patching 15.05.

diizzyy, 2019-03-28:
@karlp 
Close due to timeout?

karlp, 2019-03-28:
works for me.

closed 2019-03-28 by karlp