openwrt/packages #6593: mtr 0.92 missing mtr-packet

notnyt, 2018-07-27:
Maintainer: @neheb 
Environment: x86  OpenWrt SNAPSHOT, r7545+96-354de22bad

Description:
root@rooter:~# mtr 10.10.10.10
mtr: Failure to start mtr-packet: Invalid argument

````
diff --git a/net/mtr/Makefile b/net/mtr/Makefile
index ac11396c7..82027eb4a 100644
--- a/net/mtr/Makefile
+++ b/net/mtr/Makefile
@@ -66,6 +66,7 @@ endef
 define Package/mtr/install
        $(INSTALL_DIR) $(1)/usr/sbin
        $(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/sbin/mtr $(1)/usr/sbin/
+       $(INSTALL_BIN) $(PKG_INSTALL_DIR)/usr/sbin/mtr-packet $(1)/usr/sbin/
 endef

 $(eval $(call BuildPackage,mtr))
````

diizzyy, 2018-07-28:
You can do a pull request?

notnyt, 2018-07-29:
already is one that contains this and other fixes... 

https://github.com/openwrt/packages/pull/6560

diizzyy, 2018-07-29:
Ahh, I see

neheb, 2018-08-05:
Merged. Please close this.

closed 2018-08-05 by hnyman