openwrt/packages #6656: unbound-control failure

K2DLS, 2018-08-02:
unbound-control status
[1533215996] unbound-control[14993:0] warning: control-enable is 'no' in the config file.
[1533215996] unbound-control[14993:0] fatal error: could not send: Connection reset by peer

However, config file shows:
option unbound_control '2'

Also tried set to '1'.

Components installed:

libunbound 1.7.3-2
luci-app-unbound git-18.213.60274-7d69fa6-1
unbound 1.7.3-2
unbound-control 1.7.3-2
unbound-control-setup 1.7.3-2

Firmware: OpenWrt 18.06.0 r7188-b0b5c64c22 / LuCI openwrt-18.06 branch (git-18.210.69179-6df9a57)

dibdot, 2018-08-02:
If you like to use unbound utilities directly, you have to configure /etc/unbound/unbound.conf - and not the openwrt related uci unbound config in /etc/config/unbound, e.g. set the following options in /etc/unbound/unbound.conf:

```
# Remote control config section.
remote-control:
        # Enable remote control with unbound-control(8) here.
        # set up the keys and certificates with unbound-control-setup.
        control-enable: yes

        # Set to no and use an absolute path as control-interface to use
        # a unix local named pipe for unbound-control.
        control-use-cert: no
```
and `unbound-control status` will work, e.g.:

```
root@blackhole:/etc/unbound# unbound-control status
version: 1.7.3
verbosity: 1
threads: 4
modules: 2 [ validator iterator ]
uptime: 42304 seconds
options: control
unbound (pid 8774) is running...
```

@EricLuehrsen who knows best about. ;-)

K2DLS, 2018-08-02:
Thank you.  My intent was to view the data from luci.  Shouldn't setting the value of unbound-control-app on the Unbound Resource Settings menu populate the config?  If not, what is the purpose of this luci setting?

The reason I ran from the command line is to determine why I was not seeing any data from within luci.  BTW, manually populating the Files/Extended field with your suggested config does allow me to see control data from within luci.

But, my question is more what the luci dropdown should cause to happen...

dibdot, 2018-08-02:
Of course, to re-use the uci generated config try something like that:

`unbound-control status -c /var/lib/unbound/unbound.conf`

K2DLS, 2018-08-02:
> unbound-control status -c /var/lib/unbound/unbound.conf

unbound-control[17358:0] warning: control-enable is 'no' in the config file.
version: 1.7.3
verbosity: 1
threads: 1
modules: 2 [ validator iterator ]
uptime: 1298 seconds
options: control
unbound (pid 16462) is running...

But to reiterate, if I set the Unbound-Control-App dropdown in luci to "local host, no encryption" (for example), shouldn't luci take care of populating the config so that I can see stats on the status tabs without having to manually populate the unbound extended conf?

EricLuehrsen, 2018-08-02:
`LuCI: Admin->Service->Unbound->Status->[4 tabs]` each use unbound-control as @dibdot made example. Considering the chroot protection and flash ROM protection by using /var/lib/unboind/, I should probably set the compiled defaults there and not /etc/. It would be easier then.

EricLuehrsen, 2018-08-03:
#6665 should close this.

dibdot, 2018-08-06:
As Eric says, should be "fixed" now.

closed 2018-08-06 by dibdot

miguelgargallo, 2022-03-13:
> If you like to use unbound utilities directly, you have to configure /etc/unbound/unbound.conf - and not the openwrt related uci unbound config in /etc/config/unbound, e.g. set the following options [...]

Thanks