openwrt/packages #6666: luajit: Missing libraries: libc.so.6, libm.so.6, libdl.so.2

3xtrah3rb, 2018-08-03:
Maintainer: @\milani
Environment: 
    BuildHost: Ubuntu (running in Docker), Kernel -> 4.17.11.a-1-hardened, x86_64
    TargetHost: mips, Easybox 904 xDSL, Openwrt 15.04

Description:
After cloning https://github.com/Quallenauge/Easybox-904-XDSL and setting everything up, I start 
make. After some time it dies with this error:

```
Package luajit is missing dependencies for the following libraries:
libc.so.6
libdl.so.2
libm.so.6
Makefile:92: recipe for target '/mnt/Easybox-904-XDSL/bin/packages/mips_24kc/packages/luajit_2017-01-17-71ff7ef-1_mips_24kc.ipk' failed
make[3]: *** [/mnt/Easybox-904-XDSL/bin/packages/mips_24kc/packages/luajit_2017-01-17-71ff7ef-1_mips_24kc.ipk] Error 1
make[3]: Leaving directory '/mnt/Easybox-904-XDSL/feeds/packages/lang/luajit'
Command exited with non-zero status 2
time: package/feeds/packages/luajit/compile#0.25#0.14#0.39
package/Makefile:107: recipe for target 'package/feeds/packages/luajit/compile' failed
make[2]: *** [package/feeds/packages/luajit/compile] Error 2
make[2]: Leaving directory '/mnt/Easybox-904-XDSL'
package/Makefile:103: recipe for target '/mnt/Easybox-904-XDSL/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/mnt/Easybox-904-XDSL/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/Easybox-904-XDSL'
/mnt/Easybox-904-XDSL/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
```

The libraries missing are installed on my host system:

```
maker@31c31144b45f:/mnt/Easybox-904-XDSL$ find /usr/lib -name "libc.so.6"
/usr/lib/x86_64-linux-gnu/libc.so.6
/usr/lib32/libc.so.6

maker@31c31144b45f:/mnt/Easybox-904-XDSL$ find /usr/lib -name "libdl.so.2"
/usr/lib/x86_64-linux-gnu/libdl.so.2
/usr/lib32/libdl.so.2

maker@31c31144b45f:/mnt/Easybox-904-XDSL$ find /usr/lib -name "libm.so.6"
/usr/lib/x86_64-linux-gnu/libm.so.6
/usr/lib32/libm.so.6

```

Also tried making the package manually, works.
So I think the error only occurs when compiling with the compiler from the tool chain.

I hope that this is not just an stupid issue of mine, thanks.

yousong, 2018-08-03:
Hello @3xtrah3rb , thanks for the report.  Please be aware that OpenWrt 15.04 is essentially EOL now.  Please try 18.06, 17.04.5 or current master branch and see if they work for you.

stintel, 2018-08-03:
Also note that you are looking for the wrong libraries - the libraries in /usr/lib are your host libraries, and they are in no way compatible with the target system. Finally, you seem to be using an OpenWrt clone, which we cannot support. Either use plain OpenWrt, or file a ticket at https://github.com/Quallenauge/Easybox-904-XDSL/issues

closed 2018-08-03 by stintel