openwrt/packages #6678: Perl doesnt find same modules ..

DittelHome, 2018-08-04:
Hi,
i am using Version 18.06.00 Perl ist installed under:
/usr/lib/perl5/5.28
**and** 
/usr/lib/perl5/5.26
@INC points to 5.28.
Some modules are missing, for example USB.pm, because it resides only under 5.26.

I few days ago all were fine, because all perl stuff was installed under 5.26. Now some things resides in 5.26, others in 5.28.
This cannot work ..., i think
 regards Klaus

DittelHome, 2018-08-04:
Sorry, maybe a stupid suggestions. Does it make more sense to leave the Version.Release in the folder name ?

pprindeville, 2018-08-05:
Sounds like some of the packages need to be recompiled for 5.28.0.

@Naoir Shouldn't that happen automatically when `lang/perl/Makefile` gets changed?