openwrt/packages #6694: iptables 1.6.2-1 not load match state (Openwrt 18.06.0)

albertogonb, 2018-08-05:
Maintainer: <jo@mein.io>
Environment: mips_24kc, Openwrt 18.06.0

Description:
After upgrading from Lede 17.01.5 to Openwrt 18.06.0, iptables fails to load "state" match:
```
root@router:~# iptables -A INPUT -i br-lan   -m state --state NEW  -j ACCEPT
iptables v1.6.2: Couldn't load match `state':No such file or directory
```
xt_state module are loaded by kernel, but not used:
```
root@router:~# lsmod |grep state
nf_conntrack           52256 13 xt_state,xt_helper,xt_conntrack,xt_connmark,xt_connlimit,xt_connbytes,xt_CT,nf_nat_masquerade_ipv4,nf_conntrack_ipv4,nf_nat_ipv4,nf_nat,nf_conntrack_rtcache,act_connmark
x_tables               11344 46 ipt_REJECT,ipt_MASQUERADE,xt_time,xt_tcpudp,xt_tcpmss,xt_statistic,xt_state,xt_recent,xt_nat,xt_multiport,xt_mark,xt_mac,xt_limit,xt_length,xt_ipp2p,xt_hl,xt_helper,xt_ecn,xt_dscp,xt_conntrack,xt_connmark,xt_connlimit,xt_connbytes,xt_comment,xt_TCPMSS,xt_REDIRECT,xt_LOG,xt_HL,xt_DSCP,xt_CT,xt_CLASSIFY,iptable_mangle,iptable_filter,ipt_ECN,ip_tables,ebtables,ebt_vlan,ebt_stp,ebt_redirect,ebt_pkttype,ebt_mark_m,ebt_mark,ebt_limit,ebt_among,ebt_802_3,act_ipt
xt_state                 640  0 
```

This causes my Fwbuilder-based firewall to stop working, disabling, among other things, management access to the router

hnyman, 2018-08-05:
iptables is in the main Openwrt, not here in the packages feed.

closed 2018-08-05 by hnyman

jow-, 2018-08-06:
State match compatibility is deprecated and has been removed to save space. Use the canonical `-m ctstate --ctstate XXX` instead.

albertogonb, 2018-08-06:
Thank you very much,```-m conntrack --ctstate``` works perfectly