openwrt/packages #6700: Perl: host build under alpine/docker fails

Andy2244, 2018-08-06:
Maintainer: @pprindeville @Naoir
Environment: snapshot-sdk (arm-cortexa9)

Description:
Trying to create a alpine based docker image for sdk compilation, but `perl/host` always fails.
Any ideas?

```
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/Byte/Byte.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/Byte in any of cpan dist ext at make_ext.pl line 251.
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/CN/CN.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/CN in any of cpan dist ext at make_ext.pl line 251.
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/EBCDIC/EBCDIC.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/EBCDIC in any of cpan dist ext at make_ext.pl line 251.
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/JP/JP.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/JP in any of cpan dist ext at make_ext.pl line 251.
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/KR/KR.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/KR in any of cpan dist ext at make_ext.pl line 251.
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/Symbol/Symbol.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/Symbol in any of cpan dist ext at make_ext.pl line 251.
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/TW/TW.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/TW in any of cpan dist ext at make_ext.pl line 251.
./miniperl -Ilib -I. make_ext.pl lib/auto/Encode/Unicode/Unicode.a  MAKE="make" LIBPERL_A=libperl.a LINKTYPE=static CCCDLFLAGS=
Can't find extension Encode/Unicode in any of cpan dist ext at make_ext.pl line 251.

        Making utilities
make[4]: Entering directory '/workdir/sdk-snapshots-mvebu-cortexa9/build_dir/hostpkg/perl/perl-5.28.0/utils'
../perl -I../lib corelist.PL
make[4]: *** [Makefile:21: corelist] Segmentation fault (core dumped)
make[4]: Leaving directory '/workdir/sdk-snapshots-mvebu-cortexa9/build_dir/hostpkg/perl/perl-5.28.0/utils'
make[3]: *** [makefile:286: utilities] Error 2
make[3]: Leaving directory '/workdir/sdk-snapshots-mvebu-cortexa9/build_dir/hostpkg/perl/perl-5.28.0'
make[2]: *** [Makefile:143: /workdir/sdk-snapshots-mvebu-cortexa9/build_dir/hostpkg/perl/perl-5.28.0/.built] Error 2
make[2]: Leaving directory '/workdir/sdk-snapshots-mvebu-cortexa9/feeds/packages/lang/perl'
time: package/feeds/packages/perl/host-compile#7.50#0.91#8.62
make[1]: *** [package/Makefile:107: package/feeds/packages/perl/host/compile] Error 2
make[1]: Leaving directory '/workdir/sdk-snapshots-mvebu-cortexa9'
make: *** [/workdir/sdk-snapshots-mvebu-cortexa9/include/toplevel.mk:211: package/perl/host/compile] Error 2

```

dockerfile
```
FROM alpine

# install build packages 
RUN \
  apk add --update --no-cache \
  ca-certificates gnupg mc nano ccache \
  asciidoc bash bc binutils bzip2 cdrkit coreutils diffutils findutils flex g++ gawk gcc gettext git grep intltool \
  libxslt linux-headers make ncurses-dev patch perl-dev python2-dev tar xz unzip util-linux wget zlib-dev && \
....
```

pprindeville, 2018-08-06:
Sorry, "alpine" is the target architecture or the build (guest) host OS?  Assume you're explaining this to someone who doesn't use Alpine (I don't).

Also, when you take Docker out of the equation and try to run this natively on whatever OS you're using, what happens then?

Because I reliably build Perl on Ubuntu 16.04-4 and Fedora 28 (updated) with no issues.

Given that both of these work reliably as native hosts, what happens when you try to build Openwrt on either of these two guest operating systems as Docker containers?

Andy2244, 2018-08-06:
Alpine Linux is the Host OS inside Docker, which is one of the main distros used as docker image OS. I just tested Ubuntu and it worked fine, yet the ubuntu image is 5 times bigger, which is why if possible Alpine is preferred. The main difference is that Alpine is based on musl+busybox, thats why the base image is just 5MB.
I was hoping you could maybe give some hints, since the errors seem all inside this Encode/xy module? Maybe that's related to glibc vs musl, but than perl should have the same problem on openwrt, which is also based on musl?
Is there any special package/setting required for the host build?

pprindeville, 2018-08-06:
Sorry, I don't have time to pursue build issues on boutique OSes, but if you root-cause this and come up with a patch I'll look at it.

Andy2244, 2018-08-07:
@pprindeville No problem, was just thinking you might have a quick idea on what host package might be missing. 

PS: The problem with Ubuntu/Fedora or other "full" host system's is that often the host dependencies and stated [openwrt requirements](https://openwrt.org/docs/guide-developer/build-system/install-buildsystem) are not complete.
Alpine is specifically listed so i was hoping someone already tried building packages under it.
I switched to a minimal linux distro for my packages, since i noticed under ubuntu/debian some host dependencies are hidden, because those juggernauts just install so much by default.

Andy2244, 2018-08-07:
@pprindeville ok, the `Can't find extension Encode/...` are normal for a `-Uusedl` build, the actual error was at:
```
Making utilities
make[4]: Entering directory '/workdir/sdk-snapshots-mvebu-cortexa9/build_dir/hostpkg/perl/perl-5.28.0/utils'
../perl -I../lib corelist.PL
make[4]: *** [Makefile:21: corelist] Segmentation fault (core dumped)
make[4]: Leaving directory '/workdir/sdk-snapshots-mvebu-cortexa9/build_dir/hostpkg/perl/perl-5.28.0/utils'
make[3]: *** [makefile:286: utilities] Error 2
```
I can resolve this by removing `-Duserelocatableinc` from the host compile options? Maybe you have a idea whats going on here and is this needed for the host build?

pprindeville, 2018-08-09:
Can you get a stack trace from the core dump?

Andy2244, 2018-08-10:
Can try next week, i'm off for the weekend.

Andy2244, 2018-08-12:
@pprindeville mhh i found the core files on the docker host, yet i cant get any infos out of them.

`gdb --core=core.perl.0.bcd54d63a4544d839fb79c2ebbddcdf4.15653.1534102622000000`
```
Core was generated by `../perl -I../lib corelist.PL'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x000055e613a418b4 in ?? ()
(gdb) bt
#0  0x000055e613a418b4 in ?? ()
#1  0x0000000000000000 in ?? ()
(gdb) bt full
#0  0x000055e613a418b4 in ?? ()
No symbol table info available.
#1  0x0000000000000000 in ?? ()
No symbol table info available.
```

pprindeville, 2018-08-15:
Yeah, not going to be very helpful that way.

Andy2244, 2019-01-08:
Closing this, since its a niche problem and i switched to debian for my docker build environment.

closed 2019-01-08 by Andy2244

egorksv, 2024-12-21:
@pprindeville this affects all alpine builds, including VM on arm64, latest attempt was alpine-3.18 with openwrt-23.05.5

gdb reports segfault in `S_incpush` function.

Bumping perl version from 5.28.1 to 5.28.3 fixed the issue, with the only potential issue around `020-storables-stacksize.patch` - could this be done in 23.05 branch? Happy to assist with the patch fix.

also @Naoir