openwrt/packages #6703: adblock: missing button text

facled, 2018-08-07:
how you can see, sometimes i dont see text from both buttons.

Architecture ARMv7 Processor rev 1 (v7l)
Firmware Version OpenWrt 18.06.0 r7188-b0b5c64c22 / LuCI openwrt-18.06 branch (git-18.216.65174-3d5d9dc)
Kernel Version 4.14.54
adblock 3.5.4-2	
luci-app-adblock git-18.216.65174-3d5d9dc-1

[image]

dibdot, 2018-08-07:
Yep, that's a known bug - after the first run you see the buttons, right?

dibdot, 2018-08-07:
I've looked into it and the bug should already be fixed in 18.06 tree as well. Please reset your luci caches (rm -rf /tmp/luci-*) and check again.

dibdot, 2018-08-08:
@facled does the cache reset help regarding your issues (#6703, #6704)?

dibdot, 2018-08-10:
not reproducible, no more user feedback - close this for now.

closed 2018-08-10 by dibdot