openwrt/packages #6717: unbound: silently skips TLS certificate validation

jorti, 2018-08-08:
Maintainer: @EricLuehrsen 
Environment: x86_64 KVM virtual machine, OpenWrt 18.06.0

Versions:
unbound - 1.7.3-2
libopenssl - 1.0.2o-1

Description:
Unbound silently skips TLS certificate validation. It continues to resolve happily after configuring it like this:

```
server:
    tls-cert-bundle: "/etc/ssl/certs/ca-certificates.crt"

forward-zone:
    name: "."
    forward-ssl-upstream: yes
    forward-addr: 1.1.1.1@853#bad-hostname.com
    forward-addr: 1.0.0.1@853#bad-hostname.com
```

In this upstream bug report can be seen that OpenSSL 1.1.x is needed for this feature to work.

https://www.nlnetlabs.nl/bugs-script/show_bug.cgi?id=658#c15

If upgrading to OpenSSL 1.1.x is not yet possible, please, at least include the patch mentioned in that bug to warn the user that the name verification functionality is not available.

n8v8R, 2018-08-09:
It would be sort of detrimental to the TLS concept if the TLS certificates of the upstream resolver(s) are not being validated.

EricLuehrsen, 2018-08-09:
This is all dependent on openssl version. Unbound has all the support, but will set validation TRUE as a stub if API is missing.

EricLuehrsen, 2018-08-09:
This same situation existed when HTTPS was rolling out and it wasn't completely anchored everywhere. One argument was half-security was better than none, or the proverbial, locks help keep honest people honest. Pirates may MiM. ISP could be a higher risk for privacy leaks. An ISP could argue they just so happened to observe plain text communications, and they just so happened to share it with marketing partners. However encrypted communications have a presumption of privacy, and spoofing a domain and certificate pair is using a trademark without license... something like that.

dibdot, 2018-08-10:
fixed in trunk

closed 2018-08-10 by dibdot