openwrt/packages #6719: Adblock: not found in default repos

kriskate, 2018-08-08:
My router can't seem to find the Adblock package in any of the default package lists (https://wiki.openwrt.org/doc/packages).

I've tried both ssh and the Luci interface, none showing this package

```
OpenWRT version: OpenWrt Chaos Calmer 15.05.1 / LuCI 15.05-149-g0d8bbd2 Release (git-15.363.78009-956be55)

Router: TP-LINK 1043ND v1
```

Any suggestions?

dibdot, 2018-08-08:
That's expected, adblock has never been released for ancient CC.

> Any suggestions?

Yes, update to latest stable OpenWrt release 18.06. If that not works for you, try the adblock packages from the old stable release 17.01 with CC - but no promises at all (completely unsupported!)

**Direct download links**
core package: https://downloads.openwrt.org/releases/17.01.5/packages/x86_64/packages/adblock_3.4.3-1_all.ipk
luci addon: https://downloads.openwrt.org/releases/17.01.5/packages/x86_64/luci/luci-app-adblock_git-18.201.27126-7bf0367-1_all.ipk

closed 2018-08-08 by dibdot

kriskate, 2018-08-09:
Oh, I see.
Noob here - first time trying out OpenWRT, had it for a month now and it works great!

The thing is, when I installed OpenWRT I was actually recovering from a brick via the serial port and as you can imagine it consumed a lot of time.
... so I opened the [wiki](https://wiki.openwrt.org/about/latest), for a fast install; this states that CC is the latest, so I went for that 😄.

Will be trying out [18.06.0](https://downloads.openwrt.org/releases/18.06.0/targets/) later.

Thanks for the help!

kriskate, 2018-08-24:
Tried on OpenWrt 18.06.0 r7188-b0b5c64c22 / LuCI openwrt-18.06 branch (git-18.210.69179-6df9a57)

The adblock package does not show in any of the package lists