openwrt/packages #6723: [unbound] set separate tls_port per dns upstream server

n8v8R, 2018-08-09:
DoT port 853 is not set in stone and some DNS upstream have started to DoT on port 443 (mind you not DoH).

Unless specified in the upstream format `forward-addr: ip@port#tls_name` the downstream implementation:

```
list server 
option tls_port
```

does not appear to cater for ports to be set per server but only allow for one tls_port in the governing zone.

It would be rather expected however to be able to diversify ports per server as it is possible in upstream.

EricLuehrsen, 2018-08-09:
UCI and Unbound forward-zones are used in order. The UCI zones compact around a provider (cloudflare, quad9, ...). You can merge disparate zones like [example.net->dns.my-isp-provider.net], or like [{example.com, example.org, example.org.de}->dns.it-assets.example.org] in a corporate infrastructure with telecomuting and vpns. On the public dns side, each provider has different terms of sevice, censor/filter/spam settings, and so on. Mixing them in the same zone has strange results the end user may not be prepared to understand. You should put zones in order of provider preference (maybe cloudflare, quad9, quad9-unfiltered). Set "fallback" on them to permit try next. Set the last fall back based on if you want recursion or dead DNS without TLS.

n8v8R, 2018-08-09:
I am familiar with zone et all. 

I am running various linux boxes with `unbound ` and up to 10 different resolvers from 6 different providers and different DoT ports in the global `.` zone. All providers are carfully selected and there is no detrimental impact on doing so.

Well, I suppose I do not have to use the OpenWRT implementation for the router and rather utilize the upstream format instead.

No worries, just been my 2 cents on the subject.

closed 2018-08-09 by n8v8R

EricLuehrsen, 2018-08-09:
I have tried to model the zones after firewall UCI and LuCI both. I have tried to be careful to support novice users in a similar fashion. The structure is mostly a fall out of how to get LuCI to be able to be well presented. which is not polished yet. It seems like you have a enhanced setup, more like firewall extra rules.

Second thought ... The server field is not validated. If you leave port and tls_index empty, I think maybe you can put Unbound native `ip@port#cert.tld` in there. If not  then I will fix it.

n8v8R, 2018-08-09:
I had thought sort of this way

```
config zone '.'
  option zone_type 'forward_zone'
  option tls_upstream '1'
  option enabled '1'
  list server_name 'quad9'
    ipv4 '9.9.9.9'
    ipv6 'whatever.ip
    port '443'
    hostname (tls_index) 'dns.quad9.net'
  list server_name 'cf'
    ipv4 '1.1.1.1'
    ipv6 '2606:4700:4700::1111'
    port '853'
    hostname (tls_index) 'cloudflare-dns.com'
```

EricLuehrsen, 2018-08-09:
UCI is only two tier: config and option (or list). I see what you are asking. That would require a "config server" section that could be symbolically linked to the zone. Its possible, but it could be ugly. I have no idea how to patch that together and make pretty in LuCI.

n8v8R, 2018-08-09:
I find UCI irritating and trying at times for various reasons, one being this 2 tier inhibition for sure.

Never mind. Thank you for having given it some thought.