openwrt/packages #6744: ddns-scripts_route53-v1: incorrectly complains about curl not being SSL-capable

mark0n, 2018-08-13:
### Maintainer
@chris5560, @maxberger 
### Environment
OpenWRT 18.06.0

### Description
DDNS script for AWS Route 53 aborts with the following error:
```
 202653       : parsing script '/usr/lib/ddns/update_route53_v1.sh'
 202653  WARN : Amazon AWS Route53 communication require cURL with SSL support. Please install - TERMINATE
 202654  WARN : PID '2638' exit WITH ERROR '1' at 2018-08-12 20:26
```
Root cause: Variable $CURL_SSL [is checked](https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/update_route53_v1.sh#L5) in `/usr/lib/ddns/update_route53_v1.sh` but not initialized anywhere. The script aborts if this variable is empty.

### Package versions
```
root@LEDE:/~# opkg list-installed | egrep "ddns|curl|ssl"
curl - 7.60.0-3
ddns-scripts - 2.7.8-1
ddns-scripts_route53-v1 - 2.7.8-1
libcurl - 7.60.0-3
libopenssl - 1.0.2o-1
libustream-openssl - 2018-04-30-527e7002-3
luci-app-ddns - 2.4.9-3
openssl-util - 1.0.2o-1
```

### Suggested solution
Commenting out the check fixes the problem for me (note that the variable doesn't seem to be used). Shouldn't the package dependencies ensure that cURL is SSL-capable?

mark0n, 2018-08-13:
Similar variables like `CURL`, `WGET`, and `WGET_SSL` are being set in `dynamic_dns_functions.sh`. Seems like there is no separate `curl-ssl` executable but only `curl` (which is already SSL-enabled). So we need to find another way to detect if curl has SSL support.

mark0n, 2018-08-13:
https://github.com/openwrt/packages/pull/6747

chris5560, 2018-08-19:
Found the problem.
Setting CURL_SSL was moved from global section to function do_transfer unknown time ago.
Will prepare fix

cshoredaniel, 2018-08-19:
@chris5560 Is this https://github.com/openwrt/packages/commit/8bb49eb5edb9aa455816324b1fe69cb188c9f592#diff-636b35c9cb832b1c8ca09254fbf759c9 the offending comimt?

mark0n, 2018-08-20:
I updated my PR to undo the part of [the commit that introduced this problem](https://github.com/openwrt/packages/commit/8bb49eb5edb9aa455816324b1fe69cb188c9f592#diff-636b35c9cb832b1c8ca09254fbf759c9). This fixes the problem for me without adding any noticeable delay. It's unclear to me what the root cause of the 10 s (!) delay is that @Ansuel claimed to fix with https://github.com/openwrt/packages/pull/6103. I would expect this check to be very fast and in fact on my hardware it is:
```
root@LEDE:~# time /bin/sh -c '`which curl` -V 2>/dev/null | grep "Protocols:" | grep -F "https"'
Protocols: file ftp ftps http https 
real    0m 0.02s
user    0m 0.00s
sys     0m 0.00s
```
Also note that @Ansuel did not move the `CURL_PROXY` line. Apparently that one didn't slow things down for him. The only reason I can think of would be that @Ansuel has a huge number of directories on his search path which slows down the `which curl` calls. Is there a reason why we are calling `which curl` before running the executable? Wouldn't it be better to say `/usr/bin/curl` instead (or just `curl` in case we really want to search the path)?

dibdot, 2018-08-20:
fixed in master.

closed 2018-08-20 by dibdot

Ansuel, 2018-08-20:
@mark0n execute that command using console is ok
Execute that command with os.execute in lua over and over again is not ok...

Lua should never execute external command as it's slow in every way...

patrakov, 2018-08-25:
The fix is wrong. The right fix is to disable the check, and here is why.

Let's start with a fresh LEDE 17.01.4.

```
[aep@aep-haswell ~]$ ssh root@192.168.7.1

BusyBox v1.25.1 () built-in shell (ash)

     _________
    /        /\      _    ___ ___  ___
   /  LE    /  \    | |  | __|   \| __|
  /    DE  /    \   | |__| _|| |) | _|
 /________/  LE  \  |____|___|___/|___|                      lede-project.org
 \        \   DE /
  \    LE  \    /  -----------------------------------------------------------
   \  DE    \  /    Reboot (17.01.4, r3560-79f57e422d)
    \________\/    -----------------------------------------------------------

root@LEDE:~# curl https://www.google.com/
-ash: curl: not found
root@LEDE:~# opkg update
Downloading http://downloads.lede-project.org/releases/17.01.4/targets/ar71xx/generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_core
Downloading http://downloads.lede-project.org/releases/17.01.4/targets/ar71xx/generic/packages/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_base
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/base/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_luci
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/luci/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_packages
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_routing
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_telephony
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
root@LEDE:~# opkg install curl
Installing curl (7.52.1-10) to root...
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/base/curl_7.52.1-10_mips_24kc.ipk
Installing libmbedtls (2.7.5-1) to root...
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/base/libmbedtls_2.7.5-1_mips_24kc.ipk
Installing libcurl (7.52.1-10) to root...
Downloading http://downloads.lede-project.org/releases/17.01.4/packages/mips_24kc/base/libcurl_7.52.1-10_mips_24kc.ipk
Configuring libmbedtls.
Configuring libcurl.
Configuring curl.
```

You see - there is no way to install curl without SSL support.

Let's test it in the same way as ddns-scripts do:

```
root@LEDE:~# $(which curl) -V 2>/dev/null | grep "Protocols:"
Protocols: file ftp ftps http https 
```

So - fresh curl does claim to support https. Of course it does not work:

```
root@LEDE:~# curl https://www.google.com/
curl: (77) Error reading ca cert file /etc/ssl/certs/ca-certificates.crt - mbedTLS: (-0x3E00) PK - Read/write of file failed
```

So, please: remove the pointless $CURL_SSL check, and add the dependency on ca-bundle.

cshoredaniel, 2018-08-25:
On 2018-08-24 10:45 PM, Alexander E. Patrakov wrote:
> The fix is wrong. The right fix is to disable the check, and here is why.

Actually it's right because curl can be built without SSL support (it's
not the default build option, but it is a supported self-build option).

> You see - there is no way to install curl without SSL support.
>
> Let's test it in the same way as ddns-scripts do:
>
> |root@LEDE:~# $(which curl) -V 2>/dev/null | grep "Protocols:" Protocols:
> file ftp [...]

We don't do that in ddns-script because SSL support is optional.  For
curl the issue is that curl users may want to use only their own CA and
not ca-bundle.  So NAK from me.

Regards,

Daniel

ethanbergstrom, 2018-11-20:
The latest version available out on the default repo shipped with OpenWRT (http://downloads.openwrt.org/releases/18.06.1/packages/mips_24kc/packages/) is 2.7.8-1. Manually splicing in the changes did fix it, but it doesnt look like revision 5 has been published.

dibdot, 2018-11-20:
It's currently only available in snapshot repo, see here: 
https://downloads.openwrt.org/snapshots/packages/x86_64/packages/ddns-scripts_2.7.8-5_all.ipk

seth586, 2019-02-11:
Installed 2.7.8-1, got the `WARN : Amazon AWS Route53 communication require cURL with SSL support. Please install - TERMINATE` error.

Installed 2.7.8-6, can confirm it is fixed!

supersebbo, 2019-06-24:
This is still not in the release packages.  Installing from the snapshot package repo fixed this.

I am not familiar with the process, does someone need to make a pull-request for this?

Another thing to watch out for, when I installed the snapshot ddns-scripts base package, it 'lost' the route53 option in the DDNS provider selection.  So I installed the latetest snapshot ddns-scripts_route53-v1 package 2.7.8-9 but this now seems to be broken, as it's missing the username/password fields in the LUCI interface.  

Got it working eventually by keeping the base scripts package from the snapshot repo and removing then re-installing the release package of the route53 scripts.