openwrt/packages #6745: tree-wide: [RFC] Flattening menus (i.e. eliminating second-level submenus)

cshoredaniel, 2018-08-13:
Especially when getting started with OpenWrt finding things in menuconfig is complicated by the second level menus that are sometimes used and sometimes not, even when the category exists.

Further not everything fits neatly in a category.

Finally when, years ago, I tried to improve the situation the above resulted in titles that I think make it harder to find things (in retrospect).

Therefore I would like to do a mass removal of the second-level menus, leaving only the broad top-level categories like 'net', and 'utlitiies'.

Thoughts?

diizzyy, 2018-08-13:
NACK, it works well in a directory sense but if you need to navigate it's more of a hassle.

cshoredaniel, 2018-08-13:
Please note it is preferred if replies could be made as followups to http://lists.infradead.org/pipermail/openwrt-devel/2018-August/013568.html on [OpenWrt-devel Mailing List](https://lists.openwrt.org/mailman/listinfo/openwrt-devel)

diizzyy, 2018-08-13:
Preferred by who, you?

cshoredaniel, 2018-08-13:
@diizzyy I'm under the impression that for matters which also affect core (such as submenus, since core packages also use them, and it's probably best to keep core and package in sync in that regard) that discussions should occur on the project ML.

cshoredaniel, 2018-08-19:
Sorry hadn't noticed #3371.  I've decided based on lack of response and #3371 that it's more effort than it's worth, and I've noticed that it's mostly Utilities that still the most painful, and that's due to all in one list.  I'd rather improve the titles and/or choice of Categories in Network, and also get Utilities in better categorized state.

cshoredaniel, 2018-08-28:
Ok, so flattening doesn't look like worth the effort, and I've realized that retagging for the sake of doing so isn't really worth the effort of contacting maintainers, pushing PR's, etc, etc especially since it'd be a lot of them.

closed 2018-08-28 by cshoredaniel