openwrt/packages #6748: [RFC] Decision on preference (or not) for the use of codeload

cshoredaniel, 2018-08-13:
@neheb Has submitted a number of PR's (referenced in https://github.com/openwrt/packages/issues/6584), full list of his/her PR's (which probably include the codeload thing for any github pulls): https://github.com/openwrt/packages/pulls/neheb

@diizzyy has also mentioned codeload.  @karlp appears to feel the change is for marginal gain (based on https://github.com/openwrt/packages/issues/6584#issuecomment-410452508).

I personally think that tree-wide changes, especially for something @yousong has been trying solve another way ought t be clearly discussed and not just brought in through the back door, hence this ticket and I'll be following up on openwrt-devel as well.

neheb, 2018-08-13:
Just to clarify, I do it for GitHub projects that lack a release tarball.

Andy2244, 2018-08-13:
I also only use it for medium, large sized github packages, without release tarballs. see #6577

PS: I'm fine either way, the git clone + local tarball creation also worked fine for me in the past.

karlp, 2018-08-13:
Daniel Dickinson <notifications@github.com> wrote:
> @neheb Has submitted a number of PR's (referenced in
> https://github.com/openwrt/packages/issues/6584), full list of
> his/her PR's (which probably include the codeload thing for any
> github pulls): [...]

That's out of context.  I have no objection to the use of a documented github apis.  

Cheers,
Karl P

cshoredaniel, 2018-08-13:
@karlp apologies then, I misunderstood (the possibility I had is why I tagged you so you could comment)

cshoredaniel, 2018-08-13:
Please note it is preferred if the responses could be made as followups to http://lists.infradead.org/pipermail/openwrt-devel/2018-August/013570.html (followups to those followups) on [OpenWrt-devel Mailing List](https://lists.openwrt.org/mailman/listinfo/openwrt-devel)

diizzyy, 2018-08-13:
@cshoredaniel 
I'm not sure why you insist on having it on the mailing list since it's not related to OpenWrt distribution itself. GitHub has always been where package related discussions occur so it's the preferred forum.

Anyhow, a few advantages

* Reduces data usage considerably in a lot of cases compared to git clone
* Faster
* Better availability as other distributions also use codeload
* Probably not an issue itself but I would imagine that codeload is less troublesome then git over https in some corp environments

cshoredaniel, 2018-08-13:
@diizzyy The infrastracture and tooling of packages, which is shared amongst core and packages is also a core matter in my view.  In particular with codeload and the like the packages feed shouldn't be doing things substantially different from core, in my view.  Also I feel like the mailing list gets more project eyeballs than a random issue on GitHub, and this is an issue that I think should be discussed project-wide before implementing.

diizzyy, 2018-08-13:
I honestly think that's up for the people actually doing work in the repo but that's just me....

cshoredaniel, 2018-08-13:
Also my single biggest issue that it's a tree-wide change without prior discussion.  I don't really have deep feelings about the actual decision, I just think a discussion should actually happen.

wongsyrone, 2018-08-14:
I would like to follow the reproducible way.

diizzyy, 2018-08-14:
@cshoredaniel 
Several committers have accepted this procedure by committing several PRs and using it in their own commits. I have no idea why you're trying to make a discussion out of something that's already accepted.

dibdot, 2018-08-15:
@cshoredaniel 
Personally I would prefer to wait with codeload migrations until github will support reproducible release downloads that are automatically generated. On the other hand I have no codeload objection if the maintainer said it's OK.

EricLuehrsen, 2018-08-16:
FYI as a release option, you can upload any binary as an "asset." A upstream release authority could instead generate the tarball and upload it. That would be static and not subject to cloud based cache churn and tools changes. It would be less expensive to download than a git clone. There is [no limit]( https://help.github.com/articles/distributing-large-binaries/) to storage or bandwidth; see also [Rest API ](https://developer.github.com/v3/repos/releases/) release documentation.

cshoredaniel, 2018-08-16:
@EricLuehrsen Agreed.  I think the issue comes when upstream isn't doing proper releases (with uploaded assets) and GitHub's auto-release based on tags is wanted to be used as a formal release (rather than a hash and pull).

cshoredaniel, 2018-08-19:
Since it's holiday season I will wait until after first full week of September to close this issue.

I would like to say that impatience (and I'm badly guilty of this too), is a big part of the problem here and with the perceived
maintainer problem with the packages feed, and/or the pushing of commits by non-maintainers.

pprindeville, 2018-08-20:
> FYI as a release option, you can upload any binary as an "asset." 

@EricLuehrsen Maybe I'm misunderstanding something, but my hair stood up as soon as I read that...

cshoredaniel, 2018-09-11:
As no one seems to actually care all that seems to have really happened is pissing of some folks unnecessarily.   Therefore closing.

yousong, 2018-09-12:
@cshoredaniel  Thank you for the heads up.  Appreciate that.

jefferyto, 2019-03-21:
FWIW codeload hashes/checksums **do** change. I've been test compiling prometheus lately and this is what I see (paths changed to protect the guilty):

```
SHELL= flock /openwrt/testing/armvirt-32/tmp/.prometheus-2.7.1.tar.gz.flock -c '        /openwrt/testing/armvirt-32/scripts/download.pl "/openwrt/dl" "prometheus-2.7.1.tar.gz" "bfbeb434342a03b5849e2ec7a0cbe573067299cf59ccf59db0cacd8db8800bb0" "" "https://codeload.github.com/prometheus/prometheus/tar.gz/v2.7.1?"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://codeload.github.com/prometheus/prometheus/tar.gz/v2.7.1?/prometheus-2.7.1.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 7106k    0 7106k    0     0  2053k      0 --:--:--  0:00:03 --:--:-- 2053k
Hash of the downloaded file does not match (file: fe239e23379ce6d50c5577f515d7f576f1c6f9396ea47fd87c08fbf8b5e163fb, requested: bfbeb434342a03b5849e2ec7a0cbe573067299cf59ccf59db0cacd8db8800bb0) - deleting download.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.lede-project.org/prometheus-2.7.1.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 7106k  100 7106k    0     0  2159k      0  0:00:03  0:00:03 --:--:-- 2159k
```

If I wasn't caching downloads, I would be downloading the "same" file twice every time.

neheb, 2019-03-21:
Now we have a proper case of that :).

I ran gunzip on both tar archives, which resulted in .tar files of the same size but different hashes.

I extracted the archives and hashed every file. This is the difference:

```
mangix@mangix-pc:~$ colordiff 1f 2F 
2599c2599
< 67225f38b1b851c3f34d97f5b3eb80c081b3cc935c3b669e1fceaeb352b79357  ./vendor/k8s.io/client-go/pkg/version/base.go
---
> 8466cdeac38ae3017f04b607bb3404a02e8719c9a5d1511d883772724538d87b  ./vendor/k8s.io/client-go/pkg/version/base.go
```

So the content is actually different. My guess is upstream actually modified the tag.

diizzyy, 2019-03-21:
So codeload isn't to blame in this case :-)

neheb, 2019-03-21:
Hrm maybe it is.

https://github.com/prometheus/prometheus/blob/master/vendor/k8s.io/client-go/pkg/version/base.go

is a template that I'm guessing codeload generates.

.gitattribute has

```
base.go export-subst
```

I would guess GitHub updated codeload's Go version? No idea.

diizzyy, 2019-03-21:
There are reports of occational breakage when GitHub updates their backend but it's still very rare and by far provides more pros than cons compared to fetching source code and packaging it.

yousong, 2019-03-22:
> Hrm maybe it is.
> 
> https://github.com/prometheus/prometheus/blob/master/vendor/k8s.io/client-go/pkg/version/base.go
> 
> is a template that I'm guessing codeload generates.
> 
> .gitattribute has
> 
> ```
> base.go export-subst
> ```
> I would guess GitHub updated codeload's Go version? No idea.

This is worse than what I originally expected.  Next time other packages may accumulate enough commits to add another byte to the `%h`...

```
	// NOTE: The $Format strings are replaced during 'git archive' thanks to the
	// companion .gitattributes file containing 'export-subst' in this same
	// directory.  See also https://git-scm.com/docs/gitattributes
	gitVersion   string = "v0.0.0-master+$Format:%h$"
	gitCommit    string = "$Format:%H$" // sha1 from git, output of $(git rev-parse HEAD)
	gitTreeState string = ""            // state of git tree, either "clean" or "dirty"
```

yousong, 2019-03-22:
> There are reports of occational breakage when GitHub updates their backend but it's still very rare and by far the pros outweights the cons compared to fetching source code and packaging it.

Maybe it's rare [1], but when it happens, it affects a lot [2]

 [1]  https://www.google.com/search?q=github+archive+file+checksum
 [2] https://github.com/Homebrew/homebrew-core/issues/18044

diizzyy, 2019-03-22:
...and yet they're sticking to it just like everyone else.
https://github.com/Homebrew/homebrew-core/blob/master/Formula/airspy.rb (as an example)

jow-, 2019-03-22:
I'd prefer to not hand-wave away these issues just because of some Github fanboyism. From a reproducibility standpoint it sucks that these codeload artifacts are not stable and actually might retroactively change due to some future changes to the repository.

It is not even possible to properly archive these tarballs in the long term when packages are forced to adapt their checksums to match the updated github artifacts as this renders the archived tarballs useless.

Can we please opt for a solution that involves *stable*tarballs which are reproducible *locally*? I don't care if everyone else uses codeload. Everyone else also uses `curl random-git-hub-gists | sh` constructs or compiles code straight off random git repositories.

diizzyy, 2019-03-22:
So other projects are plain stupid and don't care? They've obviously evaluated this solution and ended up with the conclusion that it's worth the "occational extra" effort. So, opting for a solution which severly lowers performance and availability is a "much" better solution just because OpenWrt is "different"?

jow-, 2019-03-22:
> So, opting for a solution which severly lowers performance and availability is a "much" better solution just because OpenWrt is "different"?

If this is the only way to ensure bit-identical, reproducible, long-term stable tarballs, then yes.

diizzyy, 2019-03-22:
That's arguable as OpenWrt as well as with other projects archive/cache tarballs (again, availability).

jow-, 2019-03-22:
Caching tarballs is of no use when the same tarball with the same name overwrites an existing one with a different checksum. It is also not acceptable that the Github tarballs are not reproducible. We're solely dependent on whatever proprietary means Github uses to produce its files.

If Github were down one day or decides to discontinue its Codeload mechanism in favor to something else, and all I have is the Git repository, I will not be able to produce tarballs exactly resembling what was previously produced by Codeload.

IMHO it also is a very dangerous precedence to instruct packagers to simply update their checksums whenever supposedly stable tarballs suddenly change. This completely undermines the idea of file verification and makes ensuring long term availability (and more important, buildability) of the entire distro completely impossible.

As it stands right now, a naive file name based long-term caching of Codeload artifacts is not possible due to the non-guaranteed stability of files. This could be solved by caching the files not only by name, but also by their checksum at that moment in time, but that requires adjusting workflows, just to accommodate for GH's deficiencies - again.

diizzyy, 2019-03-22:
You can also look at the other way so either way has it's flaws and I honestly feel it's being blown out of proportion by quite a bit.

jow-, 2019-03-22:
I don't think that questioning a file hosting facility which randomly changes its contents is "blowing things out of proportion". I have no problem with relying on Github facilities as long as the data we acquire from there is *long term stable* and *reproducible*.

Right now it is neither, which also makes mirroring it for future use impossible. So a given tarball with a given hash cannot be acquired from anywhere anymore in case GH's mechanism suddenly decide to package it differently.

cshoredaniel, 2019-03-23:
I don't.  This, amongst other issues I have raised, combined with a lack 
of feeling of connection to the project, and frustration with having not 
spare UPS devices to do what I would consider the proper/required 
testing for the NUT package, have lead to the conclusion that I'm going 
to tidy up loose ends (like making sure NUT is working properly before 
19.xx release; which is the case for me, but I want to follow up with a 
user who's  having trouble with 18.02 version (which lacks some fixes in 
the master branch), to hopefully get at least one other verification), 
and any other loose ends (once I have a build server again), and other 
than that, not return after my technical difficultes are resolved, and 
quietly wander off to other projects.  I could go on, but I won't, as I 
don't think it's productive, but I think the silent majority who've 
wandered are likewise frustrated and put off by things like the state of 
the packages 'community'.

Regards,

Daniel

neheb, 2019-03-23:
I am curious.

Would OpenWrt's git solution change the file to its proper format?

diizzyy, 2019-03-23:
IMHO to even consider OpenWrt's solution own solution viable we need at least 3 mirrors with good peering and at least 2 shouldn't use Hetzner (blocked for people in .ru for instance) or any other blocked DC.

neheb, 2019-03-23:
I'm just wondering if it generates a proper tarball.

champtar, 2019-03-23:
I'm 100% for reproducible builds, so as long as Github don't provides reproducible downloads from codeload, we should stick to OpenWrt magic IMO
Just downloaded all sources for all packages (ar71xx), it takes 2.7G [dl.txt](https://github.com/openwrt/packages/files/2999876/dl.txt)
This would require some scripting, but OpenWrt could have a mirror with all sources, and the 11 mirrors that we have could also mirror that, what do you think ?
https://openwrt.org/downloads#mirrors

diizzyy, 2019-03-24:
I'm don't like the idea of mirroring all sources as it'll both waste space and bandwidth especially for large packages that are rather frequently update such as the kernel and we already have redudancy for most packages. It's also been brought up a few times that there's much waste of bandwidth as mirrors pulls a lot more data than what they serve.

I'm fine with only mirroring GitHub packages but in order for that to happen we should have some kind of geo/round-robin facility and some kind of mirror status facility.

https://www.archlinux.org/ (Under tools)
http://spacehopper.org/mirmon/

Another option would to mirror those tarballs on github and use CDNs such as
https://gitcdn.xyz/ , https://raw.githack.com/ , https://www.staticaly.com/
That way we wouldn't need to involve mirrors at all

poranje, 2019-03-26:
> 
> You can also look at the other way so either way has it's flaws and I honestly feel it's being blown out of proportion by quite a bit.

Security sometimes is difficult to balance against other interests.
In this case trust that the distributed deliverables are built from checked and committed code is severely impeded and that is not acceptable.

Regard,
Paul

neheb, 2020-01-12:
I will close this as there doesn't seem to be a consensus. The summary is:

codeload mostly works
local tarballs mostly work.

Both break occasionally. There are still several locally generated tarballs with wrong hashes on https://downloads.openwrt.org (seen when compiling with make V=s). That should probably be fixed but is beyond the scope of this thread.

closed 2020-01-12 by neheb

cshoredaniel, 2020-01-12:
@neheb @hnyman @jow- I think this should be reopened.  I see two people strongly arguing against and at least nine people who have expressed concerned.  I think think there is a pretty strong majority against codeload.  This really should be put to a vote since it seems that's the only way there is going to be acknowledgement what the majority think.