openwrt/packages #6753: libcurl library linking issue with C++ component in openwrt

newUsertoday, 2018-08-14:
libcurl.so.4 linking issue: `
Description:
I have created a new component in openwrt. I could build the component along with openwrt Make. 
When linking with libcurl.so.4, the build fails.
"#include <curl/curl.h> 
#include <iostream.h> 
are included. 

CURL *c1;
c1=curl_easy_init(); 

When i build this code, the build fails with Undefined Reference : curl_easy_init()".

i have included  LIBS+=-L~/target..../curl/lib path for linking

Can someone please suggest the solution on accessing the base libraries and linking them with the user defined components in openwrt. 
It would be very greatful.

newUsertoday, 2018-08-16:
Can someone please help

cshoredaniel, 2018-08-16:
GitHub issues is not a particularly good place to ask support questions.  You might want to try https://forum.openwrt.org/ or StackExchange-esque forum.

newUsertoday, 2018-08-16:
Not getting proper help. Awaiting for a person who can help!!

jow-, 2018-08-16:
Not a packages issue.

closed 2018-08-16 by jow-

newUsertoday, 2018-08-16:
if i use -lcurl, my component is getting compiled but not generating .ipk file. it says missing dependencies with libcurl.so.4. 

If i dont use -lcurl, then it says undefined references to curl_easy_init(). 

any settings to be done to link my code with the libcurl library from openwrt. Please help.