openwrt/packages #6754: mpd: enable webdav storage back-end

pacien, 2018-08-14:
Maintainer: @thess
Environment: OpenWrt 18.06.0, r7188-b0b5c64c22

Description:
Enabling the WebDAV storage plugin would be really useful in satellite setups. This would not require any additional dependency, `libcurl` and `libexpat` already being used for the curl input plugin.

https://github.com/openwrt/packages/blob/91171954ea9552298829e495210a0ea5db9865a0/sound/mpd/Makefile#L147

---

On a related note, `--disable-wildmidi` is repeated two times in the configure args:
https://github.com/openwrt/packages/blob/91171954ea9552298829e495210a0ea5db9865a0/sound/mpd/Makefile#L117
https://github.com/openwrt/packages/blob/91171954ea9552298829e495210a0ea5db9865a0/sound/mpd/Makefile#L148

thess, 2018-08-14:
I'll take a look at it. Depends on how much it adds to package size whether ```-mini```, ```-full``` or both.

wrt  ```--disable-wildmidi```
I guess I wanted to really turn it off :wink:

thess, 2018-08-28:
Added WebDAV storage to both configs - increases package size by 5K.
Ref: https://github.com/openwrt/packages/commit/864fd2bfee14d4decf8c1db03380a3d7aa080f15

closed 2018-08-28 by thess