openwrt/packages #6757: nfs-kernel-server: nfs4/41 support

Andy2244, 2018-08-15:
Maintainer: @tripolar 

Description:
I checked what needs to-be done to enable nfs4 support and i'm not comfortable adding the kmod changes or maintain those changes, since i don't use nfs atm. So i just document whats missing here, maybe this might help as a starting point, if someone wants to pick this up.

Thanks to @tobiaswaldvogel most could be copy&pasted over.

Missing:
keyutils/libkeyutils for <keyutils.h> is here: [openwrt-addpack/keyutils](https://github.com/tobiaswaldvogel/openwrt-addpack/tree/master/keyutils )
libnfsidmap is here: [openwrt-addpack/libnfsidmap](https://github.com/tobiaswaldvogel/openwrt-addpack/tree/master/libnfsidmap)

Than there are several nfs4 related kmod changes:
[openwrt-addpack/kmod-nfsv4](https://github.com/tobiaswaldvogel/openwrt-addpack/tree/master/kmod-nfsv4)

finally the modified server looks like this:
[openwrt-addpack/nfsv4-server](https://github.com/tobiaswaldvogel/openwrt-addpack/tree/master/nfsv4-server)

Andy2244, 2018-09-30:
Was just added thanks to Michael Petullo !

closed 2018-09-30 by Andy2244

salim-b, 2019-05-20:
> Was just added thanks to Michael Petullo !

Just for reference / future visitors: The mentioned PR is here: https://github.com/openwrt/packages/pull/6878