openwrt/packages #6765: mosquitto-client-nossl Unable to connect error in OpenWRT 18.06

Arduous, 2018-08-16:
Maintainer: @karlp 
Environment: OpenWRT version: 18.06
Issue noticed on GL.inet AR-150 that as some mileage in terms of upgrades and installed packages
BUILD_ID="r7188-b0b5c64c22"
LEDE_BOARD="ar71xx/generic"
LEDE_ARCH="mips_24kc"

Issue is confirmed on a clean and freshly flashed Linksys E3000
BUILD_ID="r7188-b0b5c64c22"
LEDE_BOARD="brcm47xx/generic"
LEDE_ARCH="mipsel_mips32"

Description: mosquitto_pub from package mosquitto-client-nossl is unable to communicate with the server. 
It was working with LEDE 17.01. 
It works with mosquitto_pub from package mosquitto-client-ssl

```
root@E3000-test03:/# opkg update
...
root@E3000-test03:/# opkg install mosquitto-client-nossl
...
Configuring librt.
Configuring libuuid.
Configuring libcares.
Configuring libmosquitto-nossl.
Configuring mosquitto-client-nossl.
root@E3000-test03:/# mosquitto_pub -h 192.168.64.3 -t "foo" -m "bar"
Unable to connect (Invalid function arguments provided.).
root@E3000-test03:/# opkg remove mosquitto-client-nossl libmosquitto-nossl libcares libuuid
...
root@E3000-test03:/# opkg install mosquitto-client-ssl
...
Configuring libuuid.
Configuring libcares.
Configuring zlib.
Configuring libopenssl.
Configuring libmosquitto-ssl.
Configuring mosquitto-client-ssl.
root@E3000-test03:/# mosquitto_pub -h 192.168.64.3 -t "foo" -m "bar"
root@E3000-test03:/# echo $?
0
```

karlp, 2018-08-16:
Traced to a bug upstream.  For now, just add "-p 1883" or similar to work around it.  (Or use the ssl versions)

There's a new bug fix of mosquitto "real soon now" so this may or may not roll out with that.

karlp, 2018-08-16:
I'm going to close this, there's nothing further do from OpenWrt, and my regular package maintennance will patch this as it's fixed. Thanks for the report!

closed 2018-08-16 by karlp