openwrt/packages #6769: unbound: lack of input validation for odhcpd leases file

danielg4, 2018-08-16:
Maintainer: @EricLuehrsen 
Environment: x86/64, 18.06

Description:
Unlike dnsmasq, odhcpd blindly accepts invalid hostnames from broken (e.g. IoT) DHCP clients, since it does not actually use that field. This can break DNS resolution of local hosts, and it breaks the awk script's parsing if there are spaces in the name. Attached is a patch to introduce rudimentary input validation. Please apply it to both the 18.06 branch and master, because this is a security hole, regardless of whether odhcpd gets separately patched.
[odhcpd.patch.gz](https://github.com/openwrt/packages/files/2294698/odhcpd.patch.gz)

EricLuehrsen, 2018-08-17:
That part of the scripting was a bit clumsy to start with, so I just reworked it. Non-conforming lease records should be dropped, and there is an faster way. PR #6774 on master. I will cherry pick 18.06 after merge to ensure the correct hash is xref in the commit message.

danielg4, 2018-08-17:
Discussion to continue via the PR. Closing.

danielg4, 2018-09-05:
Apparently, the RE that would've matched dnsmasq's behavior would really be
```/^[[:alnum:]]([-_[:alnum:]]*[[:alnum:]])?$/```, but that's not so important now that the code doesn't cause actual breakage. Going forward, this is being added to odhcpd, too: https://git.openwrt.org/?p=project/odhcpd.git;a=commitdiff;h=881f66b0c2526bff6a2a51b30d314c512df813b4

EricLuehrsen, 2018-09-05:
Underscore is not a valid A, AAAA, or PTR hostname in DNS. It is used in SRV to indicate sevices rather than real targets. dnsmasq may take some liberty for DHCP lease record, but it shouldn't be in DNS that way.

danielg4, 2018-09-05:
Agreed. Just wanted to add it to the record.

closed 2018-09-05 by danielg4