openwrt/packages #6771: miniupnpd: Incomplete support of public IP detection via stun

JFtico, 2018-08-16:
Maintainer: @neheb  
Original author of commits to miniupnp project: https://github.com/miniupnp/miniupnp/commits?author=pali

Environment: ar71xxx on Archer C7v2 , 8/9/2018 trunk

Description:

It looks like the current implementation of the original miniupnpd added support for resolving public IP via  an option stun server configuration. And while the implementation does support it, there is no UCI mapping for the new options.

If one manually pushes a correctly configured /var/etc/miniupnpd.conf  and restarts the service, it can work. Here is an example of one:
```
root@OpenWrt:/etc/config# cat /var/etc/miniupnpd.conf
ext_ifname=eth0.2
listening_ip=br-lan
port=5000
enable_natpmp=yes
enable_upnp=yes
secure_mode=yes
ext_perform_stun=yes
ext_stun_host=stun.onesuite.com
pcp_allow_thirdparty=no
system_uptime=yes
force_igd_desc_v1=yes
lease_file=/var/run/miniupnpd.leases
bitrate_down=8388608
bitrate_up=4194304
uuid=34d83956-276a-466c-beaf-d2af9d9e7bec
allow 1024-65535 0.0.0.0/0 1024-65535 #Allow high ports
deny 0-65535 0.0.0.0/0 0-65535 #Default deny
```

The other problem is the firewall commands are not effective, and the test fail. If the firewall is manually set to allow UDP inbound indiscriminately, the test passes. So a bug in how the fw is toggled.

Further details in this thread: https://forum.openwrt.org/t/miniupnpd-in-trunk-stun-setup/18688

neheb, 2018-08-17:
I don't maintain miniupnpd. Nevertheless, this seems easy to add at first glance.

Forceflow, 2019-11-16:
Bumping this - won't be fixed for the 19.x release, I presume?

neheb, 2020-02-09:
@JFtico @Forceflow specify a custom miniupnp.conf in /etc/config/upnp as option config_file.

neheb, 2020-02-14:
I added support for this. Add in /etc/config/upnpd

use_stun = 0
stun_host = x
stun_port = x

The first is mandatory. The next two are optional.

closed 2020-02-14 by neheb

Menion2k, 2020-04-18:
Hi all

UCI support is only part of the issue here. There is also the problem that without unrestrict all the UDP inbound the STUN will detect a restricted NAT and won't work as well

neheb, 2020-04-18:
I would like a patch if applicable.

firewire10000, 2020-05-10:
What's the verdict with UPnP and is there a fix for it? 

I would like to move off the 18.06 minupnpd package (version 2.1-1) that I'm having to manually compile into v19.07.2 firmware. I have a public IP address at my WAN/PPPoE interface but my games consoles require igdv1. Unfortunately, non of the versions of miniupnpd included in versions of OpenWrt beyond 18.06 work for me.

neheb, 2020-05-10:
Why not set

```
option igdv1            0
```

in /etc/config/upnpd ?

firewire10000, 2020-05-11:
I do believe I've tried that already and it didn't make any difference. I'm recompiling firmware today I can test it out and post back.