openwrt/packages #6773: openldap: builds ucgendat for target but tries to run on host

nwf, 2018-08-17:
Maintainer: @MikePetullo 
Environment: OpenWRT HEAD compiling for ipq806x (TP-Link C2600)

Description:

Attempting to build `openldap` results in

```
make[6]: Entering directory '/tank/openwrt/scratch/builder-dev-c2600/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openldap-2.4.46/libraries/liblunicode'
rm -f version.c
../../build/mkversion -v "2.4.46" liblunicode.a > version.c
make ucgendat
make[7]: Entering directory '/tank/openwrt/scratch/builder-dev-c2600/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openldap-2.4.46/libraries/liblunicode'
/bin/bash ../../libtool --mode=link ccache_cc  -Os -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/tank/openwrt/scratch/builder-dev-c2600/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openldap-2.4.46:openldap-2.4.46 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -lpthread -DURANDOM_DEVICE=\"/dev/urandom\"   -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/lib -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/lib -fpic -specs=/tank/openwrt/scratch/builder-dev-c2600/include/hardened-ld-pie.specs -znow -zrelro    -o ucgendat ucgendat.o ../../libraries/liblutil/liblutil.a ../../libraries/liblber/liblber.la
OpenWrt-libtool: link: ccache_cc -Os -pipe -mcpu=cortex-a15 -mfpu=neon-vfpv4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/tank/openwrt/scratch/builder-dev-c2600/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openldap-2.4.46:openldap-2.4.46 -Wformat -Werror=format-security -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -fpic -DURANDOM_DEVICE=\"/dev/urandom\" -fpic -specs=/tank/openwrt/scratch/builder-dev-c2600/include/hardened-ld-pie.specs -znow -zrelro -o .libs/ucgendat ucgendat.o  -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/lib -L/tank/openwrt/scratch/builder-dev-c2600/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/lib ../../libraries/liblutil/liblutil.a ../../libraries/liblber/.libs/liblber.so -lpthread
make[7]: Leaving directory '/tank/openwrt/scratch/builder-dev-c2600/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/openldap-2.4.46/libraries/liblunicode'
./ucgendat ./UnicodeData.txt -x ./CompositionExclusions.txt
/lib/ld-musl-armhf.so.1: No such file or directory
Makefile:290: recipe for target 'ucdata//uctable.h' failed
make[6]: *** [ucdata//uctable.h] Error 255
```

This `ucgendat` tool should be compiled for the host or otherwise avoided.

neheb, 2020-01-17:
ping @flyn-org

neheb, 2020-01-28:
Closing. No way this is valid anymore.

That Makefile codepath is commented out by default.

closed 2020-01-28 by neheb