openwrt/packages #6787: stubby: integrate with uci

jonathanunderwood, 2018-08-18:
Maintainer: @iamperson347
Environment: OpenWRT 18.06.0

Description:
At present, stubby has no uci integration, which makes it difficult to integrate with Luci. It would be great if the stubby package gained uci integration.

EricLuehrsen, 2018-09-05:
That would not be much to add but not advised. Again Stubby is a reference example of GetDNS and experimental. It has no persistent cache, so resolving each full DNSSEC chain is a continuous burden. If down stream clients are DNSSEC aware or validating, then the load can balloon. If DNSSEC is desired, then consider alternate package `dnsmasq-full` with switch enabled, and use Stubby merely as TLS tunnel. (If more work goes into Stubby, then maybe some day it could replace the DNS half of dnsmasq.)

jonathanunderwood, 2018-09-05:
I think with stubby doing dnssec, and dnsmasq option dnssec-proxy set, you get the caching by dnsmasq, and Stubby validating. Empirically, this seems to perform better than having dnsmasq do the dnssec validation.

jonathanunderwood, 2018-09-30:
PR #7122  submitted implementing UCI integration.

jonathanunderwood, 2018-10-14:
Closing now that #7122 has been merged.

closed 2018-10-14 by jonathanunderwood

spookyUnknownUser, 2019-07-21:
This might be quite stupid on my part so maybe I missed something, but when I try and run stubby without the `yml` config, so with manual `0` - stubby doesn't seem to see the UCI configuration file and prints `WARNING: No Stubby config file found... using minimal default config (Opportunistic Usage)` when I start it for the first time.

If I see this message does it mean stubby is actually using the UCI configuration file in `/etc/config/stubby`?

jonathanunderwood, 2019-07-21:
> This might be quite stupid on my part so maybe I missed something, but when I try and run stubby without the `yml` config, so `with manual '0'` - stubby doesn't seem to see the UCI configuration [...]

I can't reproduce this.

Please see the README.md for more details.

When `manual` is 0, the UCI config file will be used at startup to create a `stubby.yml` file under `/var/etc/stubby`.

When  `manual` is 1, the UCI config is ignored, and `/etc/stubby/stubby.yml` is used for con figuration.

> If I see this message does it mean stubby is actually using the UCI configuration file in `/etc/config/stubby`?
> 
> This is probably a different issue but I noticed stubby overwrote my `etc/stubby/` directory and `/etc/config/stubby` files on install. Wasn't expecting this so luckily I had backups.

If you created those, and then installed stubby, I am not surprised they were overwritten - I don't think opkg has the smarts to deal with that situation, which is a weird corner case.

jonathanunderwood, 2019-07-21:
Also: please don't continue discussion on closed Issues. Instead, if you have a genuine issue (rather than a support question), please open a new issue.