openwrt/packages #6804: isc-dhcp-server-ipv4: stops working after failover state

sRlinux, 2018-08-19:
Hello,

If i use failover, my dhcpd stops working after a

`failover peer dhcp-failover: I move from communications-interrupted to normal`

or 

`failover peer dhcp-failover: I move from startup to communications-interrupted`

in 18.06 ipv4_4.4.1-2_mips_mips32.ipk
Config is:

```
failover peer "dhcp-failover" {
        primary; # declare this to be the primary server
        address 192.168.10.254;
        port 647;
        peer address 192.168.10.253;
        peer port 647;
        max-response-delay 30;
        max-unacked-updates 10;
        load balance max seconds 3;
        mclt 1800;
        split 128;
}

subnet 192.168.10.0 netmask 255.255.255.0 {
        option routers 192.168.10.254;
        option subnet-mask 255.255.255.0;
        option domain-name-servers 192.168.10.254, 192.168.10.253;
        option domain-name "sr.local";
        default-lease-time 3600;
        max-lease-time 7200;
        pool {
                range 192.168.10.100 192.168.10.200;
                failover peer "dhcp-failover";
        }
}
```

```
failover peer "dhcp-failover" {
        secondary; # declare this to be the secondary server
        address 192.168.10.253;
        port 647;
        peer address 192.168.10.254;
        peer port 647;
        max-response-delay 30;
        max-unacked-updates 10;
        load balance max seconds 3;
}

subnet 192.168.10.0 netmask 255.255.255.0 {
        option routers 192.168.10.254;
        option subnet-mask 255.255.255.0;
        option domain-name-servers 192.168.10.254, 192.168.10.253;
        option domain-name "sr.local";
        default-lease-time 3600;
        max-lease-time 7200;
        pool {
                range 192.168.10.100 192.168.10.200;
                failover peer "dhcp-failover";
        }
}
```

I hope someone can help me!

sRlinux, 2018-08-19:
fyi, it does not look like a release issue. So please prove before closing it. thx!

dibdot, 2018-08-20:
to prevent the next stale ticket without any feedback in > 6 months, you should at least inform the maintainer (as mentioned in the ticket template) and/or ask in the forum ... maybe just a configuration issue.

sRlinux, 2018-08-20:
a configuration issue that will stop the whole daemon?

closed 2018-08-20 by sRlinux