openwrt/packages #6832: openssh: compile fails on pam variant when using glibc

mattreeve, 2018-08-21:
Maintainer: @\tripolar
Environment: ar71xx Netgear WNDR3700v4 Openwrt 18.06

Description:

When building openssh with pam and using the glibc toolchain, openssh-server-pam fails to build because of type redeclarations of pthread_exit, pthread_cancel, pthread_create and pthread_join.

For some reason, the first time you do a clean build of the system, it DOES compile, but anytime you try and recompile it again, it fails (I don't know why this is).

I've found that if I use the patch from lede  - 140-pam_uclibc_pthreads_fix.patch  - then this fixes the problems. It looks as though this patch was designed to exactly target the problem I am having, but it was removed from the package in commit 9be7653a3a1a8a2865fa40aaef6cbf8b3258be8f

It seems as though reinstating this patch might be the right solution.

uxgood, 2018-09-08:
Hi, @cshoredaniel @tripolar
openssh with pam almost breaks on all targets.
[http://downloads.lede-project.org/snapshots/faillogs/x86_64/packages/openssh/with-pam/compile.txt](url)
[http://downloads.lede-project.org/snapshots/faillogs/mips_24kc/packages/openssh/with-pam/compile.txt](url)
[http://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a9/packages/openssh/with-pam/compile.txt](url)
...
this patch will work.
```
diff --git a/net/openssh/Makefile b/net/openssh/Makefile
index 60f97f3..370ed6a 100644
--- a/net/openssh/Makefile
+++ b/net/openssh/Makefile
@@ -193,6 +193,7 @@ CONFIGURE_VARS += LD="$(TARGET_CC)"
 
 ifeq ($(BUILD_VARIANT),with-pam)
 TARGET_LDFLAGS += -lpthread
+TARGET_CFLAGS += -DUNSUPPORTED_POSIX_THREADS_HACK
 endif
 
 define Build/Compile
```

uxgood, 2018-09-08:
#6968

plntyk, 2018-09-08:
There is a proposed patch for this problem afaics that should remove pthread stuff
Upstream Bugtracker: https://bugzilla.mindrot.org/show_bug.cgi?id=2548

However it seems to be lost and has to be resubmitted to apply to current OpenSSH

neheb, 2020-01-12:
This was fixed. I don't remember when but the buildbots no longer fail on it.

closed 2020-01-12 by neheb