openwrt/packages #6833: xtables-extensions cannot build with CONFIG_PKG_ASLR_PIE

ZeroChaos-, 2018-08-21:
The actual error is "mips-openwrt-linux-musl-ld: unrecognized option '-specs=/home/zero/development/openwrt/include/hardened-ld-pie.specs'" but here is context:

```
 CC [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/ltablib.o
/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/ltablib.c: In function 'addfield':
/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/ltablib.c:137:3: warning: this 'if' clause does not guard... [-Wmisleading-indentation]
   if (!lua_isstring(L, -1))
   ^~
/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/ltablib.c:140:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the 'if'
     luaL_addvalue(b);
     ^~~~~~~~~~~~~
  CC [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/ltm.o
  CC [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/lundump.o
  CC [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/lvm.o
  CC [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/lzio.o
  CC [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/lua/lauxlib.o
  LD [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/xt_LUA.o
mips-openwrt-linux-musl-ld: unrecognized option '-specs=/home/zero/development/openwrt/include/hardened-ld-pie.specs'
mips-openwrt-linux-musl-ld: use the --help option for usage information
make[9]: *** [scripts/Makefile.build:534: /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/xt_LUA.o] Error 1
make[8]: *** [scripts/Makefile.build:547: /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA] Error 2
make[7]: *** [Makefile:1508: _module_/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions] Error 2
make[7]: Leaving directory '/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.9.122'
make[6]: *** [Makefile:461: modules] Error 2
make[6]: Leaving directory '/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions'
make[5]: *** [Makefile:495: all-recursive] Error 1
make[5]: Leaving directory '/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14'
make[4]: *** [Makefile:380: all] Error 2
make[4]: Leaving directory '/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14'
make[3]: *** [Makefile:148: /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/.built] Error 2
make[3]: Leaving directory '/home/zero/development/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#16.21#2.80#19.17
make[2]: *** [package/Makefile:109: package/feeds/packages/xtables-addons/compile] Error 2
make[2]: Leaving directory '/home/zero/development/openwrt'
make[1]: *** [package/Makefile:103: /home/zero/development/openwrt/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/zero/development/openwrt'
make: *** [/home/zero/development/openwrt/include/toplevel.mk:218: world] Error 2
```

ZeroChaos-, 2018-08-21:
just a guess, but is it possible -specs should only be passed to CC and not LD?

ryao, 2018-08-21:
I think you are correct. The spec file is used by GCC and nothing else. Here is the description of the option:

"Process file after the compiler reads in the standard specs file, in order to override the defaults which the gcc driver program uses when determining what switches to pass to cc1, cc1plus, as, ld, etc."

https://gcc.gnu.org/onlinedocs/gcc/Overall-Options.html#Overall-Options

Having it in LDFLAGS should be invalid.

ryao, 2018-08-21:
Just a FYI, here are the relevant bits:

https://github.com/openwrt/openwrt/blob/master/include/hardening.mk#L22
https://github.com/openwrt/openwrt/blob/master/include/hardened-ld-pie.specs

The reason this works elsewhere, but not here should be because the build system is invoking ld directly rather than abusing GCC for it. What the specfile does is pass "-pie" to ld unless `-no-pie`, `-static`, `-r` or `-shared` is being passed to ld. It also bypasses any build system interference with `LDFLAGS`, but only when GCC is touching them.

After thinking about it, the right thing to do here should be two fold. The first is to apply this patch to stop passing `-specs` to LD when it should go to GCC:

diff --git a/include/hardening.mk b/include/hardening.mk
index 60f39428e8..95d6b59dca 100644
--- a/include/hardening.mk
+++ b/include/hardening.mk
@@ -18,8 +18,8 @@ ifdef CONFIG_PKG_CHECK_FORMAT_SECURITY
 endif
 ifdef CONFIG_PKG_ASLR_PIE
   ifeq ($(strip $(PKG_ASLR_PIE)),1)
-    TARGET_CFLAGS += $(FPIC)
-    TARGET_LDFLAGS += $(FPIC) -specs=$(INCLUDE_DIR)/hardened-ld-pie.specs
+    TARGET_CFLAGS += $(FPIC) -specs=$(INCLUDE_DIR)/hardened-ld-pie.specs
+    TARGET_LDFLAGS += $(FPIC)
   endif
 endif
 ifdef CONFIG_PKG_CC_STACKPROTECTOR_REGULAR

The second is to write some sort of macro that can be called on packages that invoke ld directly to add `-pie` to LDFLAGS when `CONFIG_PKG_ASLR_PIE` is set. Hopefully, this will be sufficient to make things work. If not, the build system needs to be patched to conform the the hack of (ab)using the compiler to invoke ld.

neheb, 2020-01-12:
Is this still an issue?

ZeroChaos-, 2020-01-13:
sure is

neheb, 2020-01-13:
There's already PKG_ASLR_PIE:=0 because of this.

neheb, 2020-01-28:
I can't reproduce. Is this on 18.06?

ZeroChaos-, 2020-02-18:
I found this on master, but it's also broken in openwrt-19.07
```
Xtables-addons 2.14 - Linux 4.14.169
if [ -n "/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.169" ]; then make -C /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.169 M=/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions modules; fi;
make[7]: Entering directory '/home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.169'
  LD [M]  /home/zero/development/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/xtables-addons-2.14/extensions/LUA/xt_LUA.o
mips-openwrt-linux-musl-ld: unrecognized option '-specs=/home/zero/development/openwrt/include/hardened-ld-pie.specs'
mips-openwrt-linux-musl-ld: use the --help option for usage information
```

pprindeville, 2020-05-25:
Is this MIPS-only?  Can we leave ASLR turned on for everything else?

neheb, 2024-07-04:
18.06 is EOL.

closed 2024-07-04 by neheb