openwrt/packages #6844: Perl package breaks compile

symgryph, 2018-08-22:
Compiling openwrt 18.06.01 with updated package feeds causes BAIL on perl package Can't load module List::Util, dynamic loading not available in this perl.
  (You may need to build a new perl executable which either supports
  dynamic loading or has the List::Util module statically linked into it.)
 at files/perlconfig.pl line 229.
Compilation failed in require at files/perlconfig.pl line 229.
BEGIN failed--compilation aborted at files/perlconfig.pl line 229.

Performing a 'perlectomy' in the packages directory fixes the problem, but I rather suspect that this isn't the solution most people want.

pprindeville, 2018-08-22:
I'm going to need more information than this to reproduce.

symgryph, 2018-08-23:
What info would you like? Happy to send.

symgryph, 2018-08-23:
Here is os info:
Darwin BattleStar.munnster.com 17.7.0 Darwin Kernel Version 17.7.0: Thu Jun 21 22:53:14 PDT 2018; root:xnu-4570.71.2~1/RELEASE_X86_64 x86_64

Using openwrt 18.06.01 (git checked out this version). I am attaching my config for your reference.

[config.txt](https://github.com/openwrt/packages/files/2312723/config.txt)

ian700518, 2018-08-24:
@symgryph Hi~
I got the same problem, may you tell me how do you type 'perlectomy' to fixes this problem?

emotile, 2018-08-24:
And /Volumes/OpenWrt/openwrt/build_dir/hostpkg/perl/perl-5.28.0/config.sh contains an instruction that compiles perl statically.

pprindeville, 2018-08-25:
@symgryph I just built `openwrt-18.06` (also for `x86_64/generic`) and didn't have any problems.

I also note that the `.config` file that you've linked ends with:

```
# CONFIG_PACKAGE_perl is not set
```

so I'm confused that you're building it at all.

Also, `packages` is not tagged, so I'm not sure how you're matching that to base.

What happens when you build `openwrt-18.06/MASTER` against `packages`?

avrhack, 2018-08-26:
I have exactly this issue too on macOS and CONFIG_PACKAGE_perl is not set but it still comes up with this error. Clearly something is broken in the build settings.

cshoredaniel, 2018-08-26:
@pprindeville This can happen if there is a host build dependency on Perl but not a DEPENDS.

pprindeville, 2018-08-27:
What would that look like?

cshoredaniel, 2018-08-27:
I think either HOST_BUILD_DEPENDS:= or maybe HOST_DEPENDS:= ... not sure if BUILD_DEPENDS:= could be a factor too

cshoredaniel, 2018-08-27:
````
$ grep -r HOST_BUILD_DEPENDS ../..
../../ola/Makefile:HOST_BUILD_DEPENDS:=protobuf/host
../../nfs-kernel-server/Makefile:HOST_BUILD_DEPENDS:=libtirpc/host
````

cshoredaniel, 2018-08-27:
````
../../openvswitch/Makefile:PKG_BUILD_DEPENDS:=python/host python-six/host
../../ola/Makefile:PKG_BUILD_DEPENDS:=ola/host
../../seafile-ccnet/Makefile:PKG_BUILD_DEPENDS:=vala/host
../../lcdringer/Makefile:PKG_BUILD_DEPENDS:=vala
../../xtables-addons/Makefile:PKG_BUILD_DEPENDS:=iptables
../../apcupsd/Makefile:PKG_BUILD_DEPENDS:=libgd
../../linknx/Makefile:PKG_BUILD_DEPENDS:=argp-standalone
../../softethervpn/Makefile:PKG_BUILD_DEPENDS:=softethervpn/host
../../subversion/Makefile:PKG_BUILD_DEPENDS:=apr-util
../../jool/Makefile:PKG_BUILD_DEPENDS:=USE_UCLIBC:argp-standalone USE_MUSL:argp-standalone
../../seafile-server/Makefile:PKG_BUILD_DEPENDS:=vala/host libevhtp
../../addrwatch/Makefile:PKG_BUILD_DEPENDS:=USE_UCLIBC:argp-standalone USE_MUSL:argp-standalone
../../geth/Makefile:PKG_BUILD_DEPENDS:=golang/host
../../mdnsresponder/Makefile:PKG_BUILD_DEPENDS:=libcap
../../nbd/Makefile:PKG_BUILD_DEPENDS:=glib2 gettext
../../nfs-kernel-server/Makefile:PKG_BUILD_DEPENDS:=$(PKG_NAME)/host
../../sysrepo/Makefile:PKG_BUILD_DEPENDS:=+SYSREPO_BINDINGS:swig/host
../../fastd/Makefile:PKG_BUILD_DEPENDS:=nacl
../../xl2tpd/Makefile:PKG_BUILD_DEPENDS:=libpcap
../../vnstat/Makefile:PKG_BUILD_DEPENDS:=libgd
../../ulogd/Makefile:PKG_BUILD_DEPENDS:=libnetfilter-acct libnetfilter-conntrack libnetfilter-log
../../bind/Makefile:  PKG_BUILD_DEPENDS += libxml2
../../bind/Makefile:  PKG_BUILD_DEPENDS += libjson-c
../../snort/Makefile:PKG_BUILD_DEPENDS:=libtirpc
../../chrony/Makefile:PKG_BUILD_DEPENDS:=pps-tools
../../rp-pppoe/Makefile:PKG_BUILD_DEPENDS:=ppp
../../iotivity/Makefile:PKG_BUILD_DEPENDS:=boost
../../ncp/Makefile:PKG_BUILD_DEPENDS:=libowfat
../../opentracker/Makefile:PKG_BUILD_DEPENDS:=libowfat
../../samba4/Makefile:PKG_BUILD_DEPENDS:=SAMBA4_SERVER_AD_DC:python-crypto nfs-kernel-server/host
../../knxd/Makefile:PKG_BUILD_DEPENDS:=argp-standalone
````

That's just the 'net' dir

cshoredaniel, 2018-08-27:
so net doesn't seem to have that problem for perl....let's see what does...

Hmmm.... so far only things in perl dir (unless you count stuff in base that has conditional dependencies on Perl for if you build with extra features).

Doesn't look like there's anything in the packages repo that fits the bill....wonder if Freeswitch in telephony is one of them.

cshoredaniel, 2018-08-27:
xtables-addons has some kind of perl dependency if you look at ``tmp/.packagedeps``

cshoredaniel, 2018-08-27:
Upshot; you can't depend on CONFIG_perl and friends to know if you need to do something or not.

symgryph, 2018-09-01:
Dumb question: How do I fix? I am trying 'master' now to see. I am relatively new to openwrt. I would even make a 'faq' on how to 'fix' perl problems!

symgryph, 2018-09-01:
I did find a less permanaent option that seems to make openwrt compile, just go in to package/feeds/packages/xtables-addons edit the makefile and remove the -depends perl stuff. Seems to compile, still doesn't fix the perl things though!

pprindeville, 2018-09-09:
> Upshot; you can't depend on CONFIG_perl and friends to know if you need to do something or not.

@cshoredaniel So what's the fix?

pprindeville, 2018-09-10:
Try this https://github.com/openwrt/packages/pull/6984 and let me know if that fixes your problem.

judepereira, 2018-09-14:
@pprindeville No, that doesn't fix it. I applied the patch locally to test.

jpc, 2018-09-15:
The update to perl 5.28 (commit c875de62b36cb0ae6458ff1b2f094a869170d12c) broke building on mac OS.

pprindeville, 2018-09-19:
I’ll need someone to come up with a fix. I don’t have OS X set up to build OpenWrt.

judepereira, 2018-09-19:
@pprindeville Could you describe the problem? I can try to fix it.

pprindeville, 2018-09-23:
@judepereira I'm not experiencing the problem, as I don't build under OS X.  I'd ask @symgryph, @emotile, or @ian700518.

ian700518, 2018-09-24:
BAIL on perl package Can't load module List::Util, dynamic loading not available in this perl.
  (You may need to build a new perl executable which either supports
  dynamic loading or has the List::Util module statically linked into it.)
 at files/perlconfig.pl line 229.
Compilation failed in require at files/perlconfig.pl line 229.
BEGIN failed--compilation aborted at files/perlconfig.pl line 229.
-----------------------------------------------------------------------------------------
my problem like this, I compiled on openwrt lede and openwrt 18.06, and got the same problem.
my Mac OS X version is 10.13.6 and Xcode-select version is 2349

locojohn, 2018-10-31:
I have exactly the same issue as the above poster.  And I am pretty certain this error is related with collectd stuff.   Please could anyone look into this and provide a fix?

`/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/hostpkg/usr/bin/perl5.28.0 files/perlconfig.pl -Dowrt:target_cc='arm-openwrt-linux-muslgnueabi-gcc' -Dowrt:gccversion="7.3.0" -Dowrt:target_cross='arm-openwrt-linux-muslgnueabi-' -Dowrt:cflags='-O2 -pipe -mcpu=cortex-a9 -mfpu=vfp3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -I/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/include -I/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/include -I/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include/fortify -I/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/include' -Dowrt:ldflags='-L/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/usr/lib -L/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-7.3.0_musl_eabi/lib -znow -zrelro' -Dowrt:libc="musl" -Dowrt:ipv6=undef -Dowrt:threads=no -Dowrt:staging_dir='/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi' -Dowrt:host_perl_prefix='/Volumes/OpenWrt/NETGEAR_R7800/staging_dir/hostpkg/usr' files/version.config files/base.config files/arm.config files/architecture.config files/signal.config files/threads.config files/libc.config files/misc.config > /Volumes/OpenWrt/NETGEAR_R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/perl/perl-5.28.0/config.sh
Can't load module List::Util, dynamic loading not available in this perl.
  (You may need to build a new perl executable which either supports
  dynamic loading or has the List::Util module statically linked into it.)
 at files/perlconfig.pl line 229.
Compilation failed in require at files/perlconfig.pl line 229.
BEGIN failed--compilation aborted at files/perlconfig.pl line 229.
make[3]: *** [/Volumes/OpenWrt/NETGEAR_R7800/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/perl/perl-5.28.0/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 255
make[3]: Leaving directory `/Volumes/OpenWrt/NETGEAR_R7800/feeds/packages/lang/perl'
`

tony-gutierrez, 2019-04-20:
same prob.

tony-gutierrez, 2019-04-20:
https://metacpan.org/pod/distribution/perl/pod/perlembed.pod#Using-Perl-modules,-which-themselves-use-C-libraries,-from-your-C-program

jpc, 2019-08-22:
I managed to isolate the bug and fix it. It was an upstream regression in Perl 5.28 so I hope to get it merged upstream as well (see #9791 in the mean time).

closed 2020-01-12 by neheb

royujjal, 2022-06-30:
Sub-package compiling dependent package can be stopped by using this change in net/xtables-addons/Makefile under sub-package DEPENDS:
define Package/iptgeoip:
-              +perl
+PACKAGE_iptgeoip:perl