openwrt/packages #6857: wifidog-ng: failed to build

KazukiShimada, 2018-08-24:
Maintainer: @zhaojh329
Environment: x86-64, OpenWrt version latest

Description:
When wifidog-ng is enabled, it fails to build as below.
It seems dnamasq-full which is depended by wifidog-ng conflicts with dnsmasq.
Thanks in advance.

-----------------------------------------------------------------------------------------------------------------
Configuring kmod-via-rhine.
Configuring rtty-mbedtls.
Configuring kmod-button-hotplug.
Configuring kmod-ipt-offload.
Configuring ppp-mod-pppoe.
Collected errors:
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/hotplug.d/ntp/25-dnsmasqsec
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/dnsmasq
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/dnsmasq/dhcp-script.sh
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/sbin/dnsmasq
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/share/acl.d/dnsmasq_acl.json
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/share/dnsmasq/rfc6761.conf
	But that file is already provided by package  * dnsmasq
 * opkg_install_cmd: Cannot install package dnsmasq-full.
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/hotplug.d/ntp/25-dnsmasqsec
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/etc/init.d/dnsmasq
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/lib/dnsmasq/dhcp-script.sh
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/sbin/dnsmasq
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/share/acl.d/dnsmasq_acl.json
	But that file is already provided by package  * dnsmasq
 * check_data_file_clashes: Package dnsmasq-full wants to install file /home/zukky/openwrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/share/dnsmasq/rfc6761.conf
	But that file is already provided by package  * dnsmasq
 * opkg_install_cmd: Cannot install package wifidog-ng-mbedtls.
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/zukky/openwrt/openwrt'
package/Makefile:104: recipe for target '/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install' failed
make[1]: *** [/home/zukky/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/zukky/openwrt/openwrt'
/home/zukky/openwrt/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2
zukky@ubuntu:~/openwrt/openwrt$

zhaojh329, 2018-08-26:
You need to deselect dnsmasq.

KazukiShimada, 2018-08-27:
OK

closed 2018-08-27 by KazukiShimada