openwrt/packages #6860: emailrelay: Main::Run::run: local domainname/hostname error: no such host

mariarti, 2018-08-24:
emailrelay 	1.9-4

Maintainer: https://github.com/openwrt/packages/tree/master/mail/emailrelay
Environment:
 mipsel_74kc
asus rt-n16
LEDE Reboot 17.01.4 r3560-79f57e422d

Description:
```
Fri Aug 24 15:30:08 2018 user.notice emailrelay: Starting emailrelay service.
Fri Aug 24 15:30:12 2018 mail.err [1099]: emailrelay: error: Main::Run::run: local domainname/hostname error: no such host: "rt-n16"
```
After some attempts to run in manual mode, I figured out that emailrelay starts to work only after starting/ed dnsmasq daemon. 
I tried to set lowest priority START=99 in /etc/init.d/emailrelay conf file, but no luck, dnsmasq finish to setup settings almost at the end of booting process (so, hostname appears almost last in router boot process).

dibdot, 2018-08-24:
Please always inform the maintainer: @fededim

neheb, 2020-01-12:
Is this still an issue?

neheb, 2020-01-28:
LEDE 17.01 is EOL. Closing.

closed 2020-01-28 by neheb