openwrt/packages #6861: ddns-scripts: ip_source "network" and "interface" don't work anymore

jpolnetpl, 2018-08-24:
Maintainer: @chris5560 
Environment: ramips, mir3g, OpenWrt SNAPSHOT r7894-6c227e4, ddns-scripts 2.7.8-2

Description:
After OpenWrt snapshot upgrade from r7493-b123921 to r7894-6c227e4 and installation of current ddns-scripts (2.7.8-2) I had to switch 'option.ip_source' from "network" (which worked for me previously) to "web" to be able to use dynamic DNS.

I believe the error was introduced in commit 65d9b3f92bc96d5101830893388ae0a957412666.

Example of a failure to obtain current IP from "network" (which shows clearly, that IP was really obtained, only not processed correctly; non-public information in the report was anonymysed in a consistent manner):

```
# /usr/lib/ddns/dynamic_dns_updater.sh -S duckdns -v 1 start
 154835       : ************ ************** ************** **************
 154836  note : PID '8892' started at 2018-08-23 15:48
 154836       : ddns version  : 2.7.8-2
 154836       : uci configuration:
ddns.duckdns.check_interval='10'
ddns.duckdns.check_unit='minutes'
ddns.duckdns.domain='debugexample.duckdns.org'
ddns.duckdns.enabled='1'
ddns.duckdns.force_interval='72'
ddns.duckdns.force_unit='hours'
ddns.duckdns.ip_network='wan'
ddns.duckdns.ip_source='network'
ddns.duckdns.lookup_host='debugexample.duckdns.org'
ddns.duckdns.password='xxxx'
ddns.duckdns.update_url='http://www.duckdns.org/update?domains=[USERNAME]&token=[PASSWORD]&ip=[IP]'                                                                                            
ddns.duckdns.use_https='1'
ddns.duckdns.username='debugexample'
ddns.duckdns=service
 154839       : verbose mode  : 1 - run normal, console mode
 154840       : check interval: 600 seconds
 154840       : force interval: 259200 seconds
 154840       : retry interval: 60 seconds
 154840       : retry counter : 0 times
 154841       : No old process
 154841       : last update: 2018-08-22 16:03
 154841       : Detect registered/public IP
 154841       : #> /usr/bin/hostip debugexample.duckdns.org >/var/run/ddns/duckdns.dat 2>/var/run/ddns/duckdns.err                                                                             
 154842       : Registered IP '1.2.3.4' detected
 154842  info : Starting main loop at 2018-08-23 15:48
 154843       : Detect local IP on 'network'
 154843       : Data detected:
 154843       : 1.2.3.4
 154843  WARN : Get local IP via 'network' failed - retry 1/0 in 60 seconds
^C 154853  note : PID '8892' terminated by 'SIGINT' at 2018-08-23 15:48
```

And yet:
```
# jsonfilter -s "$(ubus call network.interface dump)" -e "$.interface[@.interface=\"wan\"]['ipv4-address'][0].address"                                                                         
1.2.3.4
```
```
# hostip debugexample.duckdns.org
1.2.3.4
```

I believe trouble is caused by 65d9b3f92bc96d5101830893388ae0a957412666, line 949 in 'dynamic_dns_functions.sh':
```
( set -o noglob ; eval "$__RUNPROG __DATA $ip_network" ) || \
                write_log 13 "Can not detect local IP using $__RUNPROG '$ip_network' - Error: '$?'"                                                                                            
```

Subshell does not modify parent's environemt. And i think section in parentheses is executed in a subshell, so "network_get_ipaddr" function does not modify $__DATA and therefore it is empty in test at line 951.

Similar issue occurs with IP detection via $ip_interface.

I had to use $ip_url with "web" method for ddns-scripts to work for me.

b-h-l-c, 2018-08-25:
I'm having this same issue as well

dibdot, 2018-08-25:
@jpolnetpl @b-h-l-c Could you please test the referenced pull request? Personally I don't use ddns-scripts, therefore I need your test support/efforts. Thanks!

b-h-l-c, 2018-08-25:
@dibdot Other than some difficulty reverting the "IP address source" back to "Network" from "URL" in the UI that required changing through the CLI to actually save, the errors have gone away and IP updates are going through as expected using those edits. Thanks!

PR #6868 (2018-08-27): ddns-scripts: revert latest noglob/subshell changes
Maintainer: none
Compile tested: -
Run tested: -

Description:
* partly revert 9e79e1b to fix #6861
* simplify CURL_SSL detection
* bump package version in Makefile & script library
commit 2e13d92b (2018-08-27): ddns-scripts: revert latest noglob/subshell changes

* partly revert 9e79e1b to fix #6861
* simplify CURL_SSL detection
* bump package version in Makefile & script library
closed 2018-08-27 by dibdot